realsenseai/librealsense #6510: D435i not detected built pyrealsense

pwoolvett, 2020-06-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 / D435i } | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |  Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                       | 
| Platform                                 | NVIDIA Jetson Xavier  |
| Language                            |  python                          | 
| Segment			|  others                          | 

### Issue Description
`pyrealsense2.context.devices` does not contain an entry for a D435i form a Jetson Xavier.

1. install librealsense2: `librealsense2-utils librealsense2-dev`
1. cloned librealsense repo
2. built with build python binding flag only
3. connect a D435 () and D435i (5.12.5.00)
3. `[pyrealsense2.context().devices]` contains a D435, but not the 435i

Additional info:
* A Laptop with `pip install pyrealsense2` does detect both cameras

MartyG-RealSense, 2020-06-04:
Hi @pwoolvett   It sounds as though you are using multiple cameras attached to the computer at the same time.  Is that correct, please?

If so, the RealSense developer team have created improvements to multicam that should appear in an upcoming SDK version.  It should improve multicam problems that are related to rs2::pipeline but may not correct issues caused by specific USB hub hardware models.

https://github.com/IntelRealSense/librealsense/pull/6467

There have been issues encountered by other Xavier users recently with using more than one camera at the same time.  Until the multicam improvements are available, a RealSense user has offered a workaround for multi-camera detection problems in the link below:

https://github.com/IntelRealSense/librealsense/issues/5828#issuecomment-622993107

pwoolvett, 2020-06-04:
I do not think this issue is related to multicamera setup. 

> It sounds as though you are using multiple cameras attached to the computer at the same time. Is that correct, please?

Indeed, but the problem also appeared when connecting only the D435i.

> If so, the RealSense developer team have created improvements to multicam that should appear in an upcoming SDK version. It should improve multicam problems that are related to rs2::pipeline but may not correct issues caused by specific USB hub hardware models.

Good to know, but the problem showed up before pipeline creation earlier, as the D435i wouldn't show up using the library (although both `realsense-viewer` and gstreamer were able to launch videos - `realsense-viewer` could even stream depth and imu data)

> There have been issues encountered by other Xavier users recently with using more than one camera at the same time. Until the multicam improvements are available, a RealSense user has offered a workaround for multi-camera detection problems in the link below in the meantime:
#5828 (comment)

Thanks, I'll keep it on my toolbox if such a problem appears.

I managed to solve the issue by re-building but now with these flags:
`-DBUILD_PYTHON_BINDINGS:bool=true -DFORCE_RSUSB_BACKEND=ON -DPYTHON_EXECUTABLE=...`

NOTE:
* `/dev/video` still only shows three devs for D435i

closed 2020-06-04 by pwoolvett

MartyG-RealSense, 2020-06-04:
Great news that you found a solution - thanks for the update!

pwoolvett, 2020-06-04:
> Great news that you found a solution - thanks for the update!

Maybe its useful for someone checking compatibility with tegra devices to know the `D435i` reports different number of devices in `/dev/video` ?