realsenseai/librealsense #6512: jetson nano cross compile(yocto) err

carlo0120, 2020-06-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (05.12.05.00) | 
| Operating System & Version |   {yocto linux (zeus)} | 
| Kernel Version (Linux Only)    |  (4.9)                                         | 
| Platform                                 | NVIDIA Jetson nano  |
| SDK Version                            |  { v2.35.0 }                          | 
| Language                            |  {C }                          | 
| Segment			|  {Robot }                          | 

### Issue Description

| [2/249] /home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++  -DBUILD_EASYLOGGINGPP -DBUILD_SHARED_LIBS -DELPP_NO_DEFAULT_LOG_FILE -DELPP_THREAD_SAFE -DENABLE_L500_DEPTH_INVALIDATION -DHWM_OVER_XU -DRS2_USE_CUDA -DRS2_USE_V4L2_BACKEND -DUNICODE -DWITH_TRACKING=1 -Drealsense2_EXPORTS -I. -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/realsense2 -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/src -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/console_bridge/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/cpp_common/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/rosbag_storage/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/roscpp_serialization/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/rostime/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/roscpp_traits/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/roslz4/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/msgs -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/boost -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/lz4/lib -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/recipe-sysroot/usr/include/libusb-1.0 -Icommon/fw -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0=/usr/src/debug/librealsense2/2.35.0-r0                           -fdebug-prefix-map=/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/recipe-sysroot=                           -fdebug-prefix-map=/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/recipe-sysroot -I=/usr/local/cuda-10.0/include -pedantic -g -Wno-missing-field-initializers -Wno-switch -Wno-multichar -Wsequence-point -Wformat -Wformat-security -mstrict-align -ftree-vectorize -pthread -fPIC   -std=gnu++11 -MD -MT CMakeFiles/realsense2.dir/src/proc/cuda/cuda-pointcloud.cpp.o -MF CMakeFiles/realsense2.dir/src/proc/cuda/cuda-pointcloud.cpp.o.d -o CMakeFiles/realsense2.dir/src/proc/cuda/cuda-pointcloud.cpp.o -c /home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/src/proc/cuda/cuda-pointcloud.cpp
| FAILED: CMakeFiles/realsense2.dir/src/proc/cuda/cuda-pointcloud.cpp.o
| /home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++  -DBUILD_EASYLOGGINGPP -DBUILD_SHARED_LIBS -DELPP_NO_DEFAULT_LOG_FILE -DELPP_THREAD_SAFE -DENABLE_L500_DEPTH_INVALIDATION -DHWM_OVER_XU -DRS2_USE_CUDA -DRS2_USE_V4L2_BACKEND -DUNICODE -DWITH_TRACKING=1 -Drealsense2_EXPORTS -I. -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/realsense2 -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/src -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/console_bridge/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/cpp_common/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/rosbag_storage/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/roscpp_serialization/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/rostime/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/roscpp_traits/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/roslz4/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/rosbag/msgs -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/boost -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/third-party/realsense-file/lz4/lib -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/include -I/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/recipe-sysroot/usr/include/libusb-1.0 -Icommon/fw -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0=/usr/src/debug/librealsense2/2.35.0-r0                           -fdebug-prefix-map=/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/recipe-sysroot=                           -fdebug-prefix-map=/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/recipe-sysroot -I=/usr/local/cuda-10.0/include -pedantic -g -Wno-missing-field-initializers -Wno-switch -Wno-multichar -Wsequence-point -Wformat -Wformat-security -mstrict-align -ftree-vectorize -pthread -fPIC   -std=gnu++11 -MD -MT CMakeFiles/realsense2.dir/src/proc/cuda/cuda-pointcloud.cpp.o -MF CMakeFiles/realsense2.dir/src/proc/cuda/cuda-pointcloud.cpp.o.d -o CMakeFiles/realsense2.dir/src/proc/cuda/cuda-pointcloud.cpp.o -c /home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/src/proc/cuda/cuda-pointcloud.cpp
[... cut ...]
| /home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/src/proc/cuda/cuda-pointcloud.cpp: At global scope:
| /home/xxxxx/work/nano/build/tmp/work/armv8a_tegra210-poky-linux/librealsense2/2.35.0-r0/git/src/proc/cuda/cuda-pointcloud.cpp:17:26: error: no 'const float3* librealsense::pointcloud_cuda::depth_to_points(rs2::points, const rs2_intrinsics&, const rs2::depth_frame&, float)' member function declared in class 'librealsense::pointcloud_cuda'
|          float depth_scale)

MartyG-RealSense, 2020-06-04:
As you are using Yocto Linux instead of Ubuntu, the best method of installing Librealsense on your Nano may be to use the "backend" method of installation.  It requires an internet connection but is not reliant on Linux versions, kernel versions or patches.  I have listed installation steps below.

1.  Obtain the Librealsense source code for SDK 2.35.0.  If you do not have it already, you can visit the link below, go to the 'Assets' list at the bottom of its page and download the source-code zip file. Then un-zip it.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.35.0

[image]

2.  In the unzipped Librealsense folder, go to the **Build** directory (Librealsense > Build).
If you do not have a Build folder, use the command below whilst in the Librealsense root folder to create one and navigate to it:

**mkdir build && cd build**

3.  Run the CMake command below whilst in the Build directory to install Librealsense:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

MartyG-RealSense, 2020-06-18:
Hi @carlo0120  Do you require further assistance with this case please?  Thanks!

closed 2020-06-25 by MartyG-RealSense