realsenseai/librealsense #6514: Update Firmware of D415

raghav1906, 2020-06-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
import pyrealsense2 as rs
ctx = rs.context()

dev = ctx.query_devices()[0]
dev.as_updatable().enter_update_state()

import time
time.sleep(1)

dev = ctx.query_devices()[0]
dev.as_update_device().update(open("Signed_Image_UVC_5_11_6_250.bin", "rb").read())

I used this code to update the firmware of Intel Realsense D415 but when i executed it, the camera got disconnected and I got an error message saying:
Runtime Error: access denied for 8086:adb uid: 2-3-5.
Now I am also unable to access the camera using pyrealsense2.
Can someone help me fix this??

MartyG-RealSense, 2020-06-04:
Hi @raghav1906   It looks as though you are using this script exactly: https://github.com/IntelRealSense/librealsense/issues/4416#issuecomment-511481279

To use this script, you should download the firmware file version referenced in the script and place the file in the same folder location that your script is located in.

https://dev.intelrealsense.com/docs/firmware-releases

The firmware file version used in the script is quite old, so I would recommend a more recent version such as **5.12.3.0**.  In the script, the change would look like this:

dev.as_update_device().update(open("**Signed_Image_UVC_5_12_3_0.bin**", "rb").read())

raghav1906, 2020-06-04:
> Hi @raghav1906 It looks as though you are using this script exactly: [#4416 (comment)](https://github.com/IntelRealSense/librealsense/issues/4416#issuecomment-511481279)
> 
> To use this script, you should download the firmware file version referenced in the script and place the file in the same folder location that your script is located in.
> 
> https://dev.intelrealsense.com/docs/firmware-releases
> 
> The firmware file version used in the script is quite old, so I would recommend downloading to your script folder a more recent version such as **5.12.3.0**. In the script, the change would look like this:
> 
> dev.as_update_device().update(open("**Signed_Image_UVC_5_12_3_0.bin**", "rb").read())
> 
> Please try running your script again after making this change and see if the camera can be detected in Pyrealsense2 and the firmware updated.
> 
> As @dorodnic mentions in #4416 though, the recommended way to update the firmware is through the RealSense Viewer or the **rs-fw-update** tool.

I have placed the file in the same folder location that my script is located in.I am unable to run this new script as camera is still not detected using pyrealsense2.
Can you please explain the reason why running that script has disconnected the camera and not accessible throught pyrealsense2 after executing it.
Because now the camera can only be accesses using sudo permissions if I use commands like sudo rs-enumerate-devices.

MartyG-RealSense, 2020-06-04:
Updating of firmware through Pyrealsense2 scripting and the consequences of doing so is outside of my experience, unfortunately.  I will ask within Intel for assistance with this case.  I apologise for the delay to receiving a further response in the meantime.

MartyG-RealSense, 2020-06-04:
If **rs-enumerate-devices** can detect the camera then that sounds more hopeful than no detection at all.

As a next step, could you try using the recovery mode of the **rs-fw-update** firmware updater tool with the recent firmware file that you downloaded.

https://dev.intelrealsense.com/docs/firmware-update-tool

I believe that the recovery would be activated by running the rs-fw-update program using this launch command in the command line:

**rs-fw-update -r -f Signed_Image_UVC_5_12_3_0.bin**

raghav1906, 2020-06-05:
Before running the python script for firmware update, rs-enumerate-devices and python script was showing the camera is connected but after executing it, rs-enumerate-devices and python script to check number of devices gives the output that no Intel devices connected.
But if I use sudo rs-enumerate-devices then the camera is shown as connected.
So nothing is working without this sudo command now.
How to resolve this issue?

MartyG-RealSense, 2020-06-05:
I should make clear that I am not currently a Python expert.  Having said that, the purpose of using **sudo** in a terminal is apparently to run a program with "super user" admin level permissions.  So it is possible that access to the camera from the script is being blocked because it does not have access permissions without sudo.

raghav1906, 2020-06-05:
But before running this script, I was able to access the camera without sudo permissions.
Yes, I am able to run non-RealSense Python scripts successfully.

MartyG-RealSense, 2020-06-05:
If you are able to use the camera successfully in non-Python RealSense applications in Ubuntu such as the RealSense Viewer but you are not able to access the camera in Pyrealsense2, that would suggest that the script might have altered something in your Pyrealsense2  installation in a way that broke camera access permissions.

raghav1906, 2020-06-05:
realsense-viewer is also not working now.
But as you said that the script might have altered something in your Pyrealsense2 installation,
I will try installing it again.

raghav1906, 2020-06-09:
I have to update the firmware using python script. So if anyone who can help me solve this issue using python script, please help.

raghav1906, 2020-06-09:
I uninstalled both librealsense and pyrealsense2 and installed them again.
When I executed the following script for Intel Realsense D435:
{
import pyrealsense2 as rs
ctx = rs.context()

dev = ctx.query_devices()[0]
dev.as_updatable().enter_update_state()

import time
time.sleep(1)

dev = ctx.query_devices()[0]
dev.as_update_device().update(open("Signed_Image_UVC_5_11_6_250.bin", "rb").read())
}
 After its execution, the camera got disconnected and I got an error message saying:
Runtime Error: access denied for 8086:adb uid: 2-3-5.
I am using a ubuntu machine.
Now I am also unable to access the camera using pyrealsense2.
rs-enumerate devices is also not working.
But when I used the command sudo rs-enumerate-devices, I got the following output:
Device info:
Name   :   Intel Realsense D4xx Recovery
The camera got stuck in recovery mode. How to resolve this issue?

MartyG-RealSense, 2020-06-09:
After you reinstalled Librealsense and Pyrealsense2, were you able to access the camera before you ran your firmware update script?  If so, it suggests that the script is the cause of the camera detection problem and should not be used again.

If the camera goes into recovery mode, please try repeating the firmware recovery instruction in **rs-fw-update** that I mentioned earlier in this discussion.

raghav1906, 2020-06-09:
Yes after the reinstallation , I was able to access the camera before I ran that firmware update script.
Yes ,  it suggests that the script is the cause of the camera detection problem.
Can you help me with the python script for firmware update because I need a python script only in my module for my project to update the firmware.
For now, I have used the recovery instructions for updating firmware in recovery mode.

MartyG-RealSense, 2020-06-09:
The Pyrealsense2 documentation has a section on code for updating with a provided firmware.  Please visit the link below and scroll down to the **Attributes** section at the bottom of the page.

raghav1906, 2020-06-11:
@radfordi or @dorodnic can you help how to fix this script and how to use ctx.set_devices_changed_callback()
as you mentioned in Issue#4416

MartyG-RealSense, 2020-06-25:
Hi @raghav1906   Do you require further assistance with this case please, or can it be closed?  Thanks!

raghav1906, 2020-06-29:
Yes one more help. Actually, I have to run the script through docker. So can you help me with how to give camera(connected through USB) access in docker.Thanks for all the help till now!

MartyG-RealSense, 2020-06-29:
You are very welcome.  Docker is not a subject that I have personal experience with, but others have used Docker with librealsense.

The discussion comment linked to below, and the subsequent comments below it, seem to provide useful instructions for setting up Docker with librealsense.

https://github.com/IntelRealSense/librealsense/issues/1684#issuecomment-395130874

A recent example of a user-created Dockerfile for librealsense is here:

https://github.com/advian-oss/librealsense-docker

dorodnic, 2020-07-09:
Hi @raghav1906 
I hope the information from @MartyG-RealSense was helpful. If you have more questions, please open a separate issue, since this one somewhat diverged from its original subject.

closed 2020-07-09 by dorodnic