realsenseai/librealsense #6515: Error during time_diff_keeper polling: usb device disconnected！！

WaWoWaWa, 2020-06-04:
$:roslaunch realsense2_camera demo_pointcloud.launch 

... logging to /home/wang/.ros/log/5dd829a8-a5f7-11ea-90d9-00d861d2211d/roslaunch-wang-MS-7B98-20609.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://wang-MS-7B98:46269/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 0
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 720
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 1280
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 720
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 1280
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 0
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 0
 * /camera/realsense2_camera/gyro_fps: 0
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
[... 139 lines cut ...]
 04/06 13:31:25,043 WARNING [140199250085632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No such device, number: 19
 04/06 13:31:25,053 ERROR [140199266871040] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: usb device disconnected
 04/06 13:31:25,153 WARNING [140199250085632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No such device, number: 19
 04/06 13:31:25,163 ERROR [140199266871040] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: usb device disconnected
 04/06 13:31:25,264 WARNING [140199250085632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No such device, number: 19
 04/06 13:31:25,274 ERROR [140199266871040] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: usb device disconnected
 04/06 13:31:25,374 WARNING [140199250085632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No such device, number: 19
 04/06 13:31:25,384 ERROR [140199266871040] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: usb device disconnected
 04/06 13:31:25,419 WARNING [140199250085632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No such device, number: 19
 04/06 13:31:25,429 ERROR [140199224907520] (error-handling.cpp:68) Error during polling error handler: usb device disconnected
 04/06 13:31:25,484 WARNING [140199250085632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No such device, number: 19
 04/06 13:31:25,494 ERROR [140199266871040] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: usb device disconnected
 04/06 13:31:25,594 WARNING [140199250085632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No such device, number: 19
 04/06 13:31:25,605 ERROR [140199266871040] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: usb device disconnected
 04/06 13:31:25,705 WARNING [140199250085632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No such device, number: 19
 04/06 13:31:25,715 ERROR [140199266871040] (global_timestamp_reader.cpp:188) Error during time_diff_keeper polling: usb device disconnected
 04/06 13:31:25,718 WARNING [140198443935488] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 130 error: No such device, number: 19
 04/06 13:31:25,718 WARNING [140198443935488] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 130 error: No such device, number: 19
 04/06 13:31:25,718 WARNING [140198443935488] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 130, error: No such device, number: 19
 04/06 13:31:25,728 WARNING [140199250085632] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 135 error: No such device, number: 19
 04/06 13:31:25,749 WARNING [140198964864768] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 132 error: No such device, number: 19
 04/06 13:31:25,749 WARNING [140198964864768] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 132 error: No such device, number: 19
 04/06 13:31:25,749 WARNING [140198964864768] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 132, error: No such device, number: 19
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/wang/.ros/log/5dd829a8-a5f7-11ea-90d9-00d861d2211d/camera-realsense2_camera-2*.log

MartyG-RealSense, 2020-06-04:
In your log, there is repeated occurrence of the **control_transfer returned** error.  When these errors occur continuously (because the launch program is re-trying the connection until it works), it can suggest that there is a serious problem with communication between the code and the device.

Sometimes topics are able to be correctly published in RViz despite the continuous errors.  Is there any camera imagery viewable in RViz whilst the errors are occurring, please?

MartyG-RealSense, 2020-06-18:
Hi @WaWoWaWa  Do you require further assistance with this case please?   Thanks!

MartyG-RealSense, 2020-06-25:
Case closed due to no further comments received.

closed 2020-06-25 by MartyG-RealSense

haoyuanz13, 2020-08-28:
why close this issue? 
Is there any update? Still meet the exact same problem and need help!

MartyG-RealSense, 2020-08-28:
@haoyuanz13  If there is no further comments received on a case after a certain period of time after we have responded then the case is closed.  If you have a similar issue, please create a new case on the front page of the GitHub by clicking the **New Issue** button.  Thanks!

https://github.com/IntelRealSense/librealsense/issues