realsenseai/librealsense #6516: No device connected

mydlgit, 2020-06-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435I | 
| Operating System & Version |  WIN8,1| 
| Platform                                 | PC |
| SDK Version                            | 2.0|                          | 
| Language                            |  python | 

# question
import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
config = rs.config()
print(config)
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)
config.enable_stream(rs.stream.infrared, 2, 640, 480, rs.format.y8, 30)
pipeline.start(config)

when i run the demo above, i have met this question, 'Traceback (most recent call last):
  File "E:/Machine_Learning/pycharm_project/anti_spoof/src/realsense/readVideo.py", line 16, in <module>
    pipeline.start(config)
RuntimeError: No device connected'
But in Device Manager, I can find d435i's driver. This is my first use, I don't know how to solve this bug.

MartyG-RealSense, 2020-06-04:
Hi @mydlgit  If you are using Windows 8.1, support for D435i is only listed for **Windows 10**, and there is not Windows 7 support yet for D435i either.

[image]

mydlgit, 2020-06-04:
> Hi @mydlgit If you are using Windows 8.1, support for D435i is only listed for **Windows 10**, and there is not Windows 7 support yet for D435i either.
> 
> [image]

ok, i know. thx!

closed 2020-06-04 by mydlgit