realsenseai/librealsense #6518: RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x7f8401b180):     Frame didn't arrived within 15000  in jeston TX2

shiguangzqz, 2020-06-04:
hello ,I install the librealsense2 by the doc/installation_jetson.md successfully in JestonTx2 ,and I successfully run my program to get photo data and depth data by this realsense2 in JestonTx2. 
But now I found when I open the device in my program and after get some framse data ,if I close my program suddenly (ctrl + c) whitout use the  pipeline.stop(),   it will failed to open device in my another program,it return a rs2::error:RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x7f8401b180):     Frame didn't arrived within 15000
But when I remove and reinsert the device in jestonTx2,it can done normally
my realsense so version : librealsense2.so.2.28
my realsense firmware version :05.12.01.00
this is my code :
`
rs2::config cfg;
rs2::pipeline pipe;
cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
pipe.start(cfg);
auto frames=pipe.wait_for_frames()
//do something 
...

pipe.stop()

`

shiguangzqz, 2020-06-04:
I found that this error is generated when executing the pipe.wait_for_frames() function in another program

MartyG-RealSense, 2020-06-04:
It sounds as though you want the pipeline to close cleanly when the camera is unplugged, and that it closes successfully on Jetson TX2 when unplugging but not in another program.  Is that correct, please?

Also, is your other program on the Jetson TX2 or a diferent computer?

The link below may be a useful C++ reference about creating code to close the pipeline when the camera is detected to be unplugged.

https://github.com/IntelRealSense/librealsense/issues/931

shiguangzqz, 2020-06-05:
when I use this test code ,it will not get the rs2::error when I stop the program by Ctrl+C andthen rerun the same program in the same jestonTX2.
`
#include <iostream>
#include <librealsense2/rs.hpp>
using namespace std;   
class RealSenseCamera
{
   public:
    RealSenseCamera();

    void testGetData()
    {
       auto frame=  _pipe.wait_for_frames();
      std::cout<<"get one frames"<<std::endl;
    }
    ~RealSenseCamera();

    rs2::pipeline _pipe;
    rs2::config _cfg;
 static   bool flag;

};
bool RealSenseCamera::flag =0;
 RealSenseCamera:: RealSenseCamera()
 {
   try
        {
            rs2::context ctx;
            auto devs = ctx.query_devices();
            int devs_n = devs.size();
            if (devs_n < 1)
                return;
            
             _cfg.enable_stream(RS2_STREAM_COLOR,640,480, RS2_FORMAT_BGR8, 0);

            _cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 0);
            
            flag=1;
             _pipe.start(_cfg);
             std::cout<<"start XXXXXXXX"<<std::endl;
        }
        catch (const rs2::error &e)
        {

            std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
            return ;
        }
        catch (...)
        {
          std::cout<<"catch other exception"<<std::endl;
        }
 }
 
  RealSenseCamera::~RealSenseCamera()
  {
    if(flag)
            {
              _pipe.stop();
              cout<<"stop"<<std::endl;
            }
  }

int main()
{
    try
    {
        RealSenseCamera camera;
        camera.testGetData();

        while(1)
        {
            int num;
            cin>>num;
            if(num)
                camera.testGetData();
            else 
                break;
                
        }
    }
    catch (const rs2::error &e)
    {

        std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    }
    catch (...)
    {
        std::cout<<"catch other exception"<<std::endl;
    }
    return 1;
}
`
I will try to modify my code to solve this problem in this weekend,after all , if I have other problems ,I will ask for help herer.
Thanks.

MartyG-RealSense, 2020-06-05:
@shiguangzqz   Thanks very much for sharing your code with the RealSense community.  Good luck!

shiguangzqz, 2020-06-06:
Hello. I think i find something about how to better use realsense  in JestonTx2
now i talk about my underestand about the realsense.

1.rs2::pipeline.start() /rs2::pipeline.stop() /rs2::pipeline.wait_for_frames will throw exception,so when  we use some code about realsense ,we have to try{} catch() them to avoid something error happen.

2.before you use the rs2::pipeline.start(), I suggest you make sure to enum the devices.This will make you program stronger.If you not enum the devices and there is not device in your tx2, start function will wait a few seconds and then throw exception.
`auto devs = ctx.query_devices();
 int devs_n = devs.size();
 if (devs_n < 1)
     return ErrorCode::DEVICE_ENUM_ERR;`

3.start() and stop() must use symmetricly.Never stop before start .
As for me ,I find it I use start(),and get data successfully.I stop the program by Ctrl +C suddenly,I do not know why the program did not call stop() in the destructor.But if the test demo it will call the stop().
So,I add some code in my open function :
`
try//you must have two try catch code 
{
 rs2::pipeline _pipe;
try
{
//init something ...
_pipe.start();
_pipe.wait_for_frames();
return true;
}
catch(...)
{
//catch something error 
if(enum())//there is device in tx2
{
 _pipe.stop();
_pipe.start();
_pipe.wait_for_frames();
return true;
}
}
}catch(...){return false;}
`
By this way sometime will case you 15 seconds to throw exception in the first wait_for_frames function if you have the same encounter like me.I guess this is the device config, can i modify adjust this time ?@somebody who can ask me 

Thank you for seeing the last

MartyG-RealSense, 2020-06-06:
@shiguangzqz   I went over your kindly provided script very carefully.  It looks to me as though the way that you have designed it is good.

There is a script in the link below provided by a RealSense team member that does the same set of functions (enum device, stream color and depth, catch errors, start and stop streams, etc).  It may be worth trying to see if it performs better than your own script.

https://github.com/IntelRealSense/librealsense/issues/1984#issuecomment-402649770

MartyG-RealSense, 2020-06-18:
Hi @shiguangzqz  Do you still require assistance with this case please?  Thanks!

shiguangzqz, 2020-06-18:
hello，I install the realsense-viewer by
sudo apt-get install librealsense2-utils
sudo apt-get install librealsense2-dev

But,now I find the version of realSense-viewer in JestonTx2 is 2.28 .Howerver the newest version of realSense-viewer is 2.33 in Windows and X86-linux .
And Seriesly , I find the point cloud data captured by the camera is abnormal unless the camera is more than 30cm away from the object in Jeston. However ,in the Windows or X86-linux ,point cloud data captured by the camera is abnormal unless the camera is more than 20cm.
So how can I install the 2.33 realsense-viewer in the JestonTx2 ,I can install it by compiling the source code?Can you recommend some information to me ？ Thanks！

MartyG-RealSense, 2020-06-19:
Hi @shiguangzqz   Apologies for the delay in responding further, as I was talking with Intel about the recommended method for installing Librealsense packages on Jetson.  They recommended the command below for installing / upgrading Librealsense on Jetson:

**sudo apt upgrade**

If you want to check what the current version of the package is, you can use a two step process.  The Debian packages are named after the Librealsense SDK version that they were built from.

1.  Do **sudo apt update** to sync your computer with the package server.  This command does not install the package.

2.  Do **apt-cache showpkg librealsense2** to generate a list of package information.  There is a lot of information in the list but it will contain a list of available packages.

MartyG-RealSense, 2020-07-05:
Hi @shiguangzqz   Do you reuire further assistance with this case please, or can it be closed?   Thanks!

closed 2020-07-18 by MartyG-RealSense