realsenseai/librealsense #6519: Windows 7 32bit Python dll

wolfysan, 2020-06-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   |  | 
| Operating System & Version |  Win 7  | 
| Platform                                 | PC |
| SDK Version                            |  2}                          | 
| Language                            |  python                         | 
| Segment			|  others                          | 

### Issue Description
I am trying to build to code for a 32 bit python wrapper. I generated the Visual Studio 2017 solution files, but when I try to build it, it tells me "No source directory set for -S".
Does anyone know where I could get a 32bit version of python wrapper?

MartyG-RealSense, 2020-06-04:
Hi @wolfysan  First, may I confirm please that your RealSense camera is a D410 / D415 or D430 / D435.  The D435i camera does not yet have Windows 7 support.

My recollection is that pyrealsense can be built for 32-bit by building the wrapper from source and using the 32-bit version of the Librealsense DLL files.  On Windows, the 32-bit **pyrealsense2.pyd** and **realsense2.dll** files can be found in **C: > Program files (x86) > Intel RealSense SDK 2.0 > bin > x86**

Ryderxz4224, 2020-06-04:
Hello @MartyG-RealSense !

I am actually experiencing the same issue with a D415. I tried two different installations of the Win7 SDK 2.0. Installing the pyrealsense2 library for python the way I normally do, via `pip install pyrealsense2`, leads to python crashing when I try to import the library. 

I tried your suggestion and copied all the files from **C: > Program Files > Intel RealSense SDK 2.0 > bin > x86** next to my test script for both installations. The result was the same. Python no longer crashes when I try to import the library but instead I get the 'DLL load' error you can see in the attached image. 

Any ideas as to why this is happening? 
[image]
[image]

MartyG-RealSense, 2020-06-04:
@wolfysan   I am not currently a Python expert myself at the time of writing this.  My understanding though is that you should put the two files **pyrealsense2.pyd** and **realsense2.dll** in the same folder as the one that your Python project script is located.

MartyG-RealSense, 2020-06-04:
@Ryderxz4224   The pip archive has a 32 bit Pyrealsense2 wheel file called **pyrealsense2-2.35.0.1791-cp27-cp27m-win32.whl** so I am surprised that pip does not allow it to install on Windows 7, since pip in general can apparently be used with Win7.

Are you using Python 2.7 or Python 3, please?

Ryderxz4224, 2020-06-05:
@MartyG-RealSense I am using Python 3.7.7. I have also tried 3.6.9 but same issues occurred. I have attached a photo of the files that are installed in **Lib/site-packages** via pip, maybe this will help you track down the issue. Also, the info of the crash when I try to import the library:

> Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	python.exe
  Application Version:	3.7.7150.1013
  Application Timestamp:	5e6761c5
  Fault Module Name:	StackHash_55fd
  Fault Module Version:	6.1.7601.18247
  Fault Module Timestamp:	521ea91c
  Exception Code:	c0000374
  Exception Offset:	000c3873
  OS Version:	6.1.7601.2.1.0.320.65
  Locale ID:	1033
  Additional Information 1:	55fd
  Additional Information 2:	55fdff221c5ce4acb343524663c5a05b
  Additional Information 3:	6e79
  Additional Information 4:	6e79283ac7a3df13b2773b23be5b08f6
[image]
[image]

MartyG-RealSense, 2020-06-05:
I will seek developer advice to get confirmation on whether it is possible to use pyrealsense2 in Windows 7, whether by source code compiling or pip install.

Ryderxz4224, 2020-06-05:
That's alright. In the meanwhile I managed to build LRS from source for python 3.7. The DLL error is gone now (I assume it was there in the first place as the pre-compiled files were meant for 2.7) but the output is the same: Python crashes when trying to import pyrealsense2. 

Looking forward to your answer and thank you for your time! :)

MartyG-RealSense, 2020-06-17:
Hi @Ryderxz4224   I was reminded to check your case because another RealSense user has experienced crash problems with pyrealsense2 on Windows 7.  I will follow up with Intel again about whether it is possible to use pyrealsense2 on a 32-bit setup in Windows 7.  I apologise for the delay in the meantime.

closed 2020-07-05 by MartyG-RealSense