realsenseai/librealsense #6526: Unity application on android not working

castanor, 2020-06-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | Android 10 |
| SDK Version                            |  2.35                     | 
| Language                            |  unity                          | 

### Issue Description
Hello everybody,

I'm trying to build a unity application for android that uses the depth camera. To do so I downloaded the unity package and I also built the native Android library (librealsense.aar) and placed it in the Plugins folder. As seen in another issue (#4155) I also add the request for the camera permission and the RsContext init but I'm still facing the same error:

    ExternalException: rs2_pipeline_start_with_config(pipe:0xb988c8b8, config:0xb988c940)
    
    Rethrow as Exception: No device connected
      at Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) [0x000a7] in <8ba4a07bda0d406b93ef994ea9f0e3de>:0 
        at (wrapper managed-to-native) Intel.RealSense.NativeMethods.rs2_pipeline_start_with_config(intptr,intptr,object&)
      at Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg) [0x0000d] in <8ba4a07bda0d406b93ef994ea9f0e3de>:0 
      at RsDevice.OnEnable () [0x000cf] in D:\Unity\trunk\SenseApp\Assets\RealSenseSDK2.0\Scripts\RsDevice.cs:95 

I successfully deployed from android studio the example "RS Camera". 

Does the unitypackage work with android? If so, what I'm missing? 

Thank you!

MartyG-RealSense, 2020-06-05:
Hi @castanor   Android 10 devices have been known to have conflicts with the RealSense Android wrapper.   In non-Unity projects, this is usually resolved by setting **targetSdkVersion** to an earlier version.  '27' is a targetSdkVersion value that has been proven to solve the conflict.

castanor, 2020-06-08:
Thanks, I was able to target version 27 of the sdk from unity and now it works.

closed 2020-06-08 by castanor

MartyG-RealSense, 2020-06-08:
Great news!   Thanks so much for the update.

qnach10, 2020-06-10:
@castanor I am also struggling to make unity realsense android, can I ask more details how you did that? Thanks

castanor, 2020-06-10:
@qnach10 Hi, can you give me more info? Where is it exactly that you are having problems?

qnach10, 2020-06-11:
@castanor I made interactive installation, which using unity and Realsense (playing .bag file in Unity). and I am trying to make Oculus Quest version for it. 
I succeeded to put bag files to android but I had an error when I debug with adb catlog "Unable to find realsense2 at(wrapper managed to native) Intel.Realsense.NativeMethods.rs2_get_api_version(object&). 
I changed to target version 27. 
I read another post from you and I tried to do "Build RealSense SDK for Android OS" but I am having trouble to do that (I tried both build with gradle and build with android studio)
If you have any suggestions, it would be really helpful for me!
I appreciate for your time.

Thanks

castanor, 2020-06-11:
@qnach10 I remember having the same problem when I started so I suppose the problem is indeed the missing .aar file. I'm not an expert of Android Studio but I simply did "Build->Make Module 'librealsense'" and then I found the .aar inside ..build/outputs/aar. Try and see if you manage to build this way.