realsenseai/librealsense #6528: Empty Depth frame

malapatiravi, 2020-06-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.12.03 | 
| Operating System & Version |  Linux Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-101-generic  | 
| Platform                                 | PC |
| SDK Version                            | 2.34.1      | 
| Language                            |  Python  | 
| Segment			|  Robot                          | 

### Issue Description
I compiled the SDK using the following flags
cmake ../ -DBUILD_EXAMPLES=true -DENFORCE_METADATA=false -DBUILD_PYTHON_BINDINGS=true -DFORCE_LIBUVC=false -DBUILD_WITH_CUDA=true -DIMPORT_DEPTH_CAM_FW=false -DBUILD_WITH_TM2=false -DCMAKE_BUILD_TYPE=Release

At sometimes i see after running cameras for 5 to 6 hours , i am observing an empty depth frame? 
Is this observed anytime?

MartyG-RealSense, 2020-06-05:
Does the long-run test always take place roughly during the same time of the day each time, please?  If the depth image no longer appears correctly and stays that way (at least until the camera is reset), I wonder if there is an environmental lighting event occurring, such as a flash of sunlight, that is observed by the IR imager and causes it to become saturated.

Such an event can occur even when indoors.  There was a case where a camera seemed to be getting frozen when a glass door was opened by the building's occupants and the angle of the opening door would reflect a bright ray of sunlight onto the camera lens.

**Edit:**  I am off-shift no.  If you require further assistance, I will be happy to continue this support conversation when I return if you leave a comment below.   Thanks!

malapatiravi, 2020-06-09:
For the long run test the cameras were facing a white wall. It happens randomly. It recovers and works fine after one or two frames. We have observed few times now very frequent.

MartyG-RealSense, 2020-06-09:
When the camera experiences a bad frame, it recovers by going back to the last good frame and continuing onward from there.  A sign that this has occurred is if the same frame number repeats (e.g 1, 2, 3, 3, 4).  This may not be what is happening to you in this case though, as I would therefore expect to see duplicate frames instead of an empty frame.

I note that you mention use of "cameras" (suggesting more than one).  Are you using a multi camera setup please, and if so are they linked together with hardware sync cabling?

malapatiravi, 2020-06-11:
I am using multi camera setup and they are not linked together.

MartyG-RealSense, 2020-06-12:
Apologies for the delay in responding further, as I was doing further  research into your case.

I considered whether camera auto-reset might be occurring, causing a blank frame.  But as it only lasts for a frame or two, a camera reset might take longer than that to complete and resume streaming.

Such a short period of time before recovery sounds more like a change in the camera's conditions to me, such as an auto-exposure adjustment.  Are you using auto-exposure, please?

malapatiravi, 2020-06-17:
We are using auto exposure only for depth stream.

MartyG-RealSense, 2020-06-17:
Are the white walls at a far distance from the camera?

malapatiravi, 2020-07-03:
@MartyG-RealSense I think its issue with environment. We had white walls and no proper environment. We are not observing currently. Thanks.

closed 2020-07-03 by malapatiravi