realsenseai/librealsense #6530: Makefile:129: recipe for target 'all' failed

AdamMiltonBarker, 2020-06-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   |  | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    | 5.3.0-1017-raspi2                                        | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  2 }                          | 
| Language                            |  {NA }                          | 
| Segment			|                          | 

### Issue Description
Will not install on RPI3b 

```
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
CMakeFiles/realsense2.dir/build.make:782: recipe for target 'CMakeFiles/realsense2.dir/src/media/ros/ros_reader.cpp.o' failed
make[2]: *** [CMakeFiles/realsense2.dir/src/media/ros/ros_reader.cpp.o] Error 4
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/realsense2.dir/all' failed
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

AdamMiltonBarker, 2020-06-06:
As an update I found existing librealsense.so's using the following command I removed those and so far it has got past the stage where it normally crashes. This may have been my fault, if so will close the ticket.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge `

MartyG-RealSense, 2020-06-06:
Hi @AdamMiltonBarker   If you are installing with Debian packages, this will not work on Pi because it is an Arm-based device and the Debian packages are designed for x86 and x64 processors.  

https://github.com/IntelRealSense/librealsense/issues/3931#issuecomment-490155952

If you wish to build Librealsense for Ubuntu 18.04, you should compile the SDK from source code.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

Alternatively, Intel provide a Librealsense installation guide for **Raspbian** on Pi 3.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md

There is also a guide to compile the **Ubuntu MATE** OS from source code on Pi 3b.

https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md

AdamMiltonBarker, 2020-06-06:
Thanks I am following this guide.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md 

 Failed many times, but since removing the so's it has not yet crashed, as I said I think this may be my fault and will update once/if installed. 

Thank you for the links the following I had not seen, if this build fails I will update here and try this guide. 

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md

MartyG-RealSense, 2020-06-06:
Thanks very much for the details.   Good luck with the installation(s)   :)

AdamMiltonBarker, 2020-06-06:
Thank you and thank you for the quick reply :)

AdamMiltonBarker, 2020-06-06:
OK it has frozen here:  Building CXX object CMakeFiles/realsense2.dir/src/media/ros/ros_reader.cpp.o
 it will crash again, this is following the exact steps in the tutorial I linked above, many times failed at this point.

AdamMiltonBarker, 2020-06-06:
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
CMakeFiles/realsense2.dir/build.make:782: recipe for target 'CMakeFiles/realsense2.dir/src/media/ros/ros_reader.cpp.o' failed
make[2]: *** [CMakeFiles/realsense2.dir/src/media/ros/ros_reader.cpp.o] Error 4
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/realsense2.dir/all' failed
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

MartyG-RealSense, 2020-06-06:
The installation guide is assuming Ubuntu LTS.  I do not know what differences there might be between the general Ubuntu 18.04 LTS and the Raspberry Pi site's version of Ubuntu 18.04.

AdamMiltonBarker, 2020-06-06:
It was linked from Intel's documentation, however it did say Ubuntu Mate which I am not using as we use Ubuntu Server. This is the page I downloaded 18.04 from https://ubuntu.com/download/raspberry-pi

I will download and install Raspian and use the link you gave me. I had hoped not to use it (Raspian) as all our tutorials use Ubuntu Server, but will try now. Thanks again for quick replies.

MartyG-RealSense, 2020-06-06:
There is a "backend" installation process that requires an internet connection for the build process but is not dependent on Linux versions or kernel versions and does not require patching.  The steps for installation with this method are below:

1.  Go to the Librealsense root directory and create a Build folder, and then change to that created folder using the command below:

mkdir build && cd build

2.  Whilst in the Build folder, run the CMake command below to install Librealsense without dependency on Linux version, kernel version or patches:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

AdamMiltonBarker, 2020-06-06:
Cool ok thanks I will try this, thank you!

AdamMiltonBarker, 2020-06-06:
Your suggestion above has frozen at this point for the last 10 minutes.

Building CXX object CMakeFiles/realsense2.dir/src/ds5/ds5-device.cpp.o

MartyG-RealSense, 2020-06-06:
The guide for installing Ubuntu MATE on Pi 3b says " In some cases the RAM capacity is not sufficient to compile the SDK, so if the compilation process crashes or exits with an error-code try to create a swap file and then recompile the SDK".

I would guess that limited memory capacity on Pi might also affect an Ubuntu 18.04 installation (perhaps even more so that MATE), requiring a swap file to be set up.

https://www.howtoforge.com/ubuntu-swap-file

AdamMiltonBarker, 2020-06-06:
Thanks already did that in the very first install and each new OS install since. I have followed the guide to the dot. I am installing now with your suggestion above, I thought it was crashing again but it just REALLLLLY slow lol. I am at 19% right now, will keep you informed. And again thanks for your time in helping mate.

MartyG-RealSense, 2020-06-06:
You're very welcome.  If all else fails, you could perhaps consider upgrading to a Raspberry Pi 4, which is more powerful (performance equal to a business desktop PC) and more RAM.  Good luck!

AdamMiltonBarker, 2020-06-06:
I have one in the mail :) Still on 19% is it meant to be that slow ?

MartyG-RealSense, 2020-06-06:
Once memory gets consumed, processing performance for computing in general can slow to a near infinite crawl.  I experience this with my own personal PC with non-RealSense applications.  And the longer the install speed, the higher the likelihood that it will go wrong even if it gets past the point that it seems stuck on.

AdamMiltonBarker, 2020-06-06:
I have given up with this OS now it has wasted too much time that I don't have. Installing Raspberry Pi OS (32-bit) Lite https://www.raspberrypi.org/downloads/raspberry-pi-os/

AdamMiltonBarker, 2020-06-06:
This is insane:

 sudo apt-get install -y libdrm-amdgpu1 libdrm-amdgpu1-dbg libdrm-dev libdrm-exynos1 libdrm-exynos1-dbg libdrm-freedreno1 libdrm-freedreno1-dbg libdrm-nouveau2 libdrm-nouveau2-dbg libdrm-omap1 libdrm-omap1-dbg libdrm-radeon1 libdrm-radeon1-dbg libdrm-tegra0 libdrm-tegra0-dbg libdrm2 libdrm2-dbg
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libdrm-amdgpu1-dbg
E: Unable to locate package libdrm-exynos1-dbg
E: Unable to locate package libdrm-freedreno1-dbg
E: Unable to locate package libdrm-nouveau2-dbg
E: Unable to locate package libdrm-omap1-dbg
E: Unable to locate package libdrm-radeon1-dbg
E: Unable to locate package libdrm-tegra0-dbg
E: Unable to locate package libdrm2-dbg

MartyG-RealSense, 2020-06-06:
In a case where someone else had these missing files on a Pi 4 installation, this method for fixing it was suggested:

https://github.com/IntelRealSense/librealsense/issues/5092#issuecomment-568311798

AdamMiltonBarker, 2020-06-06:
Yes found that, basically ignore the errors, there are also other errors in the documentation. 

`sudo udevadm control --reload-rules && udevadm trigger ` 

needs to be

`sudo udevadm control --reload-rules && sudo udevadm trigger `

AdamMiltonBarker, 2020-06-06:
Sorry you linked to another issue where a user suggested just ignore it https://github.com/IntelRealSense/librealsense/issues/5092#issuecomment-554190266 

I have to say this is one reason I rarely use Intel tech anymore, it is a very frustrating and time consuming process.

MartyG-RealSense, 2020-06-06:
I genuinely understand your frustration.  There are some combinations of hardware and OS that can be very problematic at first but recommended practices and guide documents may emerge over time to ease the process.  If the computing hardware is low specification though (e.g limited memory of 1 GB) then it may always be challenging.

AdamMiltonBarker, 2020-06-06:
I get that, I shouldn't of said it really I am an Intel Software Innovator, just very tired and been working on this since last night. I hadn't got to the comment you shared yet I was still following the previous one, do you know if ignoring the error is not a correct way to get past this? I am up to cmake in the tutorial at the moment.

AdamMiltonBarker, 2020-06-06:
Just posting errors as I find them, please get someone to update all your tutorials this is really  crazy.

```
Google Mock not present.  Fetching gmock-1.7.0 from the web...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   129  100   129    0     0    381      0 --:--:-- --:--:-- --:--:--   380
100  362k    0  362k    0     0   275k      0 --:--:--  0:00:01 --:--:-- 1861k
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   129  100   129    0     0    365      0 --:--:-- --:--:-- --:--:--   365
100  618k    0  618k    0     0   436k      0 --:--:--  0:00:01 --:--:--  892k
+ autoreconf -f -i -Wall,no-obsolete
./autogen.sh: 50: ./autogen.sh: autoreconf: not found
```

MartyG-RealSense, 2020-06-06:
My understanding is that Intel Innovators are not banned from making constructive criticism.  **smiles**. When tired, it is usually best to take a break and come back later with a fresh mind if your project schedule allows.

Typos do sometimes occur in the documentation, so please do keep reporting it if you believe that there are errors and you have an alternative that seems to work.

AdamMiltonBarker, 2020-06-06:
Thanks, I wish I could take a break. More issues after installing autoconf autogen:

```
pi@RPI3-1:~/protobuf $ ./autogen.sh
+ autoreconf -f -i -Wall,no-obsolete
configure.ac:30: error: possibly undefined macro: AC_PROG_LIBTOOL
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1
```

MartyG-RealSense, 2020-06-06:
Thanks very much, keep it up please.  To quote Pokémon lyrics: Each time you try, gonna get just a little bit better.  Each step you climb is one more step up the ladder

AdamMiltonBarker, 2020-06-06:
:) That made me smile lol Although I don't remember the lyrics, I will have a listen :)

AdamMiltonBarker, 2020-06-06:
```
pi@RPI3-1: python3 setup.py build --cpp_implementation
Traceback (most recent call last):
  File "setup.py", line 12, in <module>
    from setuptools import setup, Extension, find_packages
ModuleNotFoundError: No module named 'setuptools'
```

Fix: 
pip/pip3 (Need to install pip3 on fresh installation) install setuptools

AdamMiltonBarker, 2020-06-06:
```
 python3 setup.py build --cpp_implementation
running build
running build_py
Generating google/protobuf/descriptor_pb2.py...
Generating google/protobuf/compiler/plugin_pb2.py...
Generating google/protobuf/any_pb2.py...
Generating google/protobuf/api_pb2.py...
Generating google/protobuf/duration_pb2.py...
Generating google/protobuf/empty_pb2.py...
Generating google/protobuf/field_mask_pb2.py...
Generating google/protobuf/source_context_pb2.py...
Generating google/protobuf/struct_pb2.py...
Generating google/protobuf/timestamp_pb2.py...
Generating google/protobuf/type_pb2.py...
Generating google/protobuf/wrappers_pb2.py...
Generating google/protobuf/any_test_pb2.py...
Generating google/protobuf/map_proto2_unittest_pb2.py...
Generating google/protobuf/map_unittest_pb2.py...
Generating google/protobuf/test_messages_proto3_pb2.py...
Generating google/protobuf/test_messages_proto2_pb2.py...
Generating google/protobuf/unittest_arena_pb2.py...
Generating google/protobuf/unittest_no_arena_pb2.py...
Generating google/protobuf/unittest_no_arena_import_pb2.py...
Generating google/protobuf/unittest_pb2.py...
Generating google/protobuf/unittest_custom_options_pb2.py...
Generating google/protobuf/unittest_import_pb2.py...
Generating google/protobuf/unittest_import_public_pb2.py...
Generating google/protobuf/unittest_mset_pb2.py...
Generating google/protobuf/unittest_mset_wire_format_pb2.py...
Generating google/protobuf/unittest_no_generic_services_pb2.py...
Generating google/protobuf/unittest_proto3_arena_pb2.py...
Generating google/protobuf/util/json_format_proto3_pb2.py...
Generating google/protobuf/internal/any_test_pb2.py...
Generating google/protobuf/internal/descriptor_pool_test1_pb2.py...
Generating google/protobuf/internal/descriptor_pool_test2_pb2.py...
Generating google/protobuf/internal/factory_test1_pb2.py...
Generating google/protobuf/internal/factory_test2_pb2.py...
Generating google/protobuf/internal/file_options_test_pb2.py...
Generating google/protobuf/internal/import_test_package/inner_pb2.py...
Generating google/protobuf/internal/import_test_package/outer_pb2.py...
Generating google/protobuf/internal/missing_enum_values_pb2.py...
Generating google/protobuf/internal/message_set_extensions_pb2.py...
Generating google/protobuf/internal/more_extensions_pb2.py...
Generating google/protobuf/internal/more_extensions_dynamic_pb2.py...
Generating google/protobuf/internal/more_messages_pb2.py...
Generating google/protobuf/internal/packed_field_test_pb2.py...
Generating google/protobuf/internal/test_bad_identifiers_pb2.py...
Generating google/protobuf/pyext/python_pb2.py...
creating build
creating build/lib.linux-armv7l-3.7
creating build/lib.linux-armv7l-3.7/google
copying google/__init__.py -> build/lib.linux-armv7l-3.7/google
creating build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/text_encoding.py -> build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/descriptor_pb2.py -> build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/descriptor.py -> build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/unittest_pb2.py -> build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/unittest_import_public_pb2.py -> build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/unittest_mset_pb2.py -> build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/text_format.py -> build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/descriptor_database.py -> build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/timestamp_pb2.py -> build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/any_test_pb2.py -> build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/field_mask_pb2.py -> build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/wrappers_pb2.py -> build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/json_format.py -> build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/struct_pb2.py -> build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/unittest_no_generic_services_pb2.py -> build/lib.linux-armv7l-3.7/google/protobuf
copying google/protobuf/service_reflection.py -> build/lib.linux-armv7l-3.7/google/protobuf
[... 78 lines cut ...]
Skipping optional fixer: idioms
Skipping optional fixer: set_literal
Skipping optional fixer: ws_comma
running build_ext
building 'google.protobuf.pyext._message' extension
creating build/temp.linux-armv7l-3.7
creating build/temp.linux-armv7l-3.7/google
creating build/temp.linux-armv7l-3.7/google/protobuf
creating build/temp.linux-armv7l-3.7/google/protobuf/pyext
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I. -I../src -I/usr/include/python3.7m -c google/protobuf/pyext/descriptor_containers.cc -o build/temp.linux-armv7l-3.7/google/protobuf/pyext/descriptor_containers.o -Wno-write-strings -Wno-invalid-offsetof -Wno-sign-compare
google/protobuf/pyext/descriptor_containers.cc: In function ‘bool google::protobuf::python::descriptor::_GetItemByKey(google::protobuf::python::PyContainer*, PyObject*, const void**)’:
google/protobuf/pyext/descriptor_containers.cc:69:45: error: invalid conversion from ‘const char*’ to ‘char*’ [-fpermissive]
        ((*(charpp) = PyUnicode_AsUTF8AndSize(ob, (sizep))) == NULL? -1: 0): \
                      ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
[... cut ...]
google/protobuf/pyext/descriptor_containers.cc:69:45: error: invalid conversion from ‘const char*’ to ‘char*’ [-fpermissive]
        ((*(charpp) = PyUnicode_AsUTF8AndSize(ob, (sizep))) == NULL? -1: 0): \
                      ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
google/protobuf/pyext/descriptor_containers.cc:189:13: note: in expansion of macro ‘PyString_AsStringAndSize’
         if (PyString_AsStringAndSize(key, &camelcase_name, &name_size) < 0) {
             ^~~~~~~~~~~~~~~~~~~~~~~~
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
```

AdamMiltonBarker, 2020-06-06:
Cannot find a way around the above, any suggestions?

AdamMiltonBarker, 2020-06-06:
```
 sudo apt install -y ./libopencv3_3.4.3-20180907.1_armhf.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'libopencv3' instead of './libopencv3_3.4.3-20180907.1_armhf.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libopencv3 : Depends: libavcodec57 but it is not installable
              Depends: libavformat57 but it is not installable
              Depends: libswscale4 but it is not installable
              Depends: libx264-148 but it is not installable
              Depends: libavresample3 but it is not installable
              Depends: libtesseract3 but it is not installable
              Depends: libhdf5-100 but it is not installable
E: Unable to correct problems, you have held broken packages.

```

AdamMiltonBarker, 2020-06-06:
```

       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/8/vector:69,
                 from /home/pi/librealsense/src/usb/usb-types.h:10,
                 from /home/pi/librealsense/src/backend.h:10,
                 from /home/pi/librealsense/src/ds5/ds5-private.h:6,
                 from /home/pi/librealsense/src/core/advanced_mode.h:5,
                 from /home/pi/librealsense/src/ds5/advanced_mode/advanced_mode.cpp:4:
/usr/include/c++/8/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_realloc_insert(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long long int, long long unsigned int, double, std::allocator>}; _Tp = nlohmann::basic_json<>; _Alloc = std::allocator<nlohmann::basic_json<> >]’:
/usr/include/c++/8/bits/vector.tcc:413:7: note: parameter passing for argument of type ‘std::vector<nlohmann::basic_json<>, std::allocator<nlohmann::basic_json<> > >::iterator’ {aka ‘__gnu_cxx::__normal_iterator<nlohmann::basic_json<>*, std::vector<nlohmann::basic_json<>, std::allocator<nlohmann::basic_json<> > > >’} changed in GCC 7.1
       vector<_Tp, _Alloc>::
       ^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/8/map:61,
                 from /home/pi/librealsense/src/usb/usb-types.h:9,
                 from /home/pi/librealsense/src/backend.h:10,
                 from /home/pi/librealsense/src/ds5/ds5-private.h:6,
                 from /home/pi/librealsense/src/core/advanced_mode.h:5,
                 from /home/pi/librealsense/src/ds5/advanced_mode/advanced_mode.cpp:4:
/usr/include/c++/8/bits/stl_map.h: In member function ‘nlohmann::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType> nlohmann::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType>::parser::parse_internal(bool) [with ObjectType = std::map; ArrayType = std::vector; StringType = std::__cxx11::basic_string<char>; BooleanType = bool; NumberIntegerType = long long int; NumberUnsignedType = long long unsigned int; NumberFloatType = double; AllocatorType = std::allocator]’:
/usr/include/c++/8/bits/stl_map.h:499:8: note: parameter passing for argument of type ‘std::_Rb_tree<std::__cxx11::basic_string<char>, std::pair<const std::__cxx11::basic_string<char>, nlohmann::basic_json<> >, std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, nlohmann::basic_json<> > >, std::less<std::__cxx11::basic_string<char> >, std::allocator<std::pair<const std::__cxx11::basic_string<char>, nlohmann::basic_json<> > > >::const_iterator’ {aka ‘std::_Rb_tree_const_iterator<std::pair<const std::__cxx11::basic_string<char>, nlohmann::basic_json<> > >’} changed in GCC 7.1
    __i = _M_t._M_emplace_hint_unique(__i, std::piecewise_construct,
In file included from /usr/include/c++/8/vector:69,
                 from /home/pi/librealsense/src/usb/usb-types.h:10,
                 from /home/pi/librealsense/src/backend.h:10,
                 from /home/pi/librealsense/src/ds5/ds5-private.h:6,
                 from /home/pi/librealsense/src/core/advanced_mode.h:5,
                 from /home/pi/librealsense/src/ds5/advanced_mode/advanced_mode.cpp:4:
/usr/include/c++/8/bits/vector.tcc:109:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<nlohmann::basic_json<>*, std::vector<nlohmann::basic_json<>, std::allocator<nlohmann::basic_json<> > > >’ changed in GCC 7.1
    _M_realloc_insert(end(), std::forward<_Args>(__args)...);
    ^~~~~~~~~~~~~~~~~
/home/pi/librealsense/src/ds5/advanced_mode/advanced_mode.cpp: At global scope:
/home/pi/librealsense/src/ds5/advanced_mode/advanced_mode.cpp:940:1: fatal error: error writing to /tmp/cc5Us16c.s: No space left on device
 }
 ^
compilation terminated.
make[2]: *** [CMakeFiles/realsense2.dir/build.make:219: CMakeFiles/realsense2.dir/src/ds5/advanced_mode/advanced_mode.cpp.o] Error 1
/home/pi/librealsense/src/ds5/ds5-device.cpp:1034:1: fatal error: error writing to /tmp/cc4ziUtS.s: No space left on device
 }
 ^
compilation terminated.
make[2]: *** [CMakeFiles/realsense2.dir/build.make:128: CMakeFiles/realsense2.dir/src/ds5/ds5-device.cpp.o] Error 1
/home/pi/librealsense/src/ds5/ds5-motion.cpp:506:1: fatal error: error writing to /tmp/ccmGMafi.s: No space left on device
 }
 ^
compilation terminated.
make[2]: *** [CMakeFiles/realsense2.dir/build.make:102: CMakeFiles/realsense2.dir/src/ds5/ds5-motion.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:69: CMakeFiles/realsense2.dir/all] Error 2
```

I am going to try Mint now, there are serious issues with the installation guides, this has all be done on a fresh installation of first RPI Ubuntu 18.04 and now Raspberry Pi OS.

AdamMiltonBarker, 2020-06-07:
Over 24 hours working on this now, 3 different OS, three different official tutorials. Using Intel's Ubuntu mate tutorial it crashes out of memory and there is already a swap file.

AdamMiltonBarker, 2020-06-07:
I have tried again and it fails at ros_reader, again. The RPI4 get's here today, I hope I do not face the same with that device. I think you guys need to do some testing and update the information in the tutorials as I have not been able to get a single one working.

MartyG-RealSense, 2020-06-07:
I really appreciate your intense and lengthy efforts!  The Librealsense developers and engineers want it to run on everything, though sometimes this may be made more complex by design decisions that have been made by a computing device's manufacturer.

Ubuntu 18.04 and Raspberry Pi 3b are a rare combination, and RealSense team members tend to just recommend the Raspbian and MATE operating systems for Pi 3.  I will add a Documentation label to this discussion and leave it open, though it may be a low-priority developer task.

AdamMiltonBarker, 2020-06-07:
That's ok. I tested all 3 tutorials and all three OS, they all fail at:

Building CXX object CMakeFiles/realsense2.dir/src/media/ros/ros_reader.cpp.o 

I think that would be a good place to start. From my part if you like you can close, or leave open, I will leave that to you guys. 

I will let you know how it goes with RPI4 (shudders) :) 

Thanks for everything.

MartyG-RealSense, 2020-06-07:
You are very welcome.  Good luck with Pi 4, and I will pass your findings about **ros_reader_cpp.o** onto Support.

AdamMiltonBarker, 2020-06-07:
[image]

Something just arrived :)

MartyG-RealSense, 2020-06-07:
The joy of the smell of freshly unboxed circuit board.  :)

MartyG-RealSense, 2020-06-18:
Hi @AdamMiltonBarker   Do you have an update about your tests with Pi 4 please?   Thanks!

AdamMiltonBarker, 2020-06-18:
@MartyG-RealSense hi mate sorry I haven't got round to it yet been working on other parts of the project, I definitely will when I do mate.

MartyG-RealSense, 2020-06-18:
No problem at all.   Thanks for the update!

csr-kick, 2020-06-18:
If it helps any I did successfully build and run librealsense on Ubuntu 18.04 on the Rpi4. I used the 64-bit link from here: https://ubuntu.com/download/raspberry-pi

I don't recall having to do anything super special, but the normal pi installer never worked for me. I either use Balena Etcher or win32diskimager. Then of course setting up wifi and ssh manually and getting raspi-config and wiringpi from a shady source (https://launchpad.net/~ubuntu-pi-flavour-makers/+archive/ubuntu/ppa). I used the xubuntu desktop and had the pose, depth, and a few other examples running.

AdamMiltonBarker, 2020-06-18:
@csr-kick thank you for the advice and info mate, much appreciated! @MartyG-RealSense I will hopefully be updating you next week mate.

AdamMiltonBarker, 2020-06-20:
@csr-kick I am starting this now, have Ubuntu installed on the RPI4, just as a heads up, are you aware of being able to use wiringpi without sudo ? There is an issue with OpenVino that doesn't allow you to run as root, I tried all sorts of udev rules on UP2 and got close but wasn't able to complete it.

AdamMiltonBarker, 2020-06-20:
"The repository 'http://ppa.launchpad.net/ubuntu-pi-flavour-makers/ppa/ubuntu focal Release' does not have a Release file. "

Failed lol FML

AdamMiltonBarker, 2020-06-20:
@csr-kick first I got:

```
Err:5 http://ppa.launchpad.net/ubuntu-pi-flavour-makers/ppa/ubuntu focal Release
  404  Not Found [IP: 91.189.95.83 80]
Hit:6 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Reading package lists... Done
E: The repository 'http://ppa.launchpad.net/ubuntu-pi-flavour-makers/ppa/ubuntu focal Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

Then using:

`sudo apt-get update --allow-insecure-repositories
`
I got: 

```
E: Failed to fetch http://ppa.launchpad.net/ubuntu-pi-flavour-makers/ppa/ubuntu/dists/focal/main/binary-arm64/Packages  404  Not Found [IP: 91.189.95.83 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

Will update when find solution.

AdamMiltonBarker, 2020-06-20:
Easy fix :D Removed the ppa and then used:

`sudo apt-get install wiringpi`

AdamMiltonBarker, 2020-06-20:
Yer 20.04 not supported. Back to 18.04.

AdamMiltonBarker, 2020-06-21:
@MartyG-RealSense  using  -DFORCE_RSUSB_BACKEND=true

```
In file included from /home/ubuntu/librealsense/examples/software-device/../example.hpp:10:0,
                 from /home/ubuntu/librealsense/examples/software-device/rs-software-device.cpp:6:
/home/ubuntu/librealsense/third-party/glfw/include/GLFW/glfw3.h:207:13: fatal error: GL/glu.h: No such file or directory
    #include <GL/glu.h>
             ^~~~~~~~~~
compilation terminated.
examples/software-device/CMakeFiles/rs-software-device.dir/build.make:62: recipe for target 'examples/software-device/CMakeFiles/rs-software-device.dir/rs-software-device.cpp.o' failed
make[2]: *** [examples/software-device/CMakeFiles/rs-software-device.dir/rs-software-device.cpp.o] Error 1
CMakeFiles/Makefile2:506: recipe for target 'examples/software-device/CMakeFiles/rs-software-device.dir/all' failed
make[1]: *** [examples/software-device/CMakeFiles/rs-software-device.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

Trying the script now LibUVC-backend installation

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

AdamMiltonBarker, 2020-06-21:
This is really getting annoying now, I have successful completed installation using https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md 

No warnings, no errors, completed fine. Device shows using enemurate devices. 

```
import pyrealsense2 as rs
ModuleNotFoundError: No module named 'pyrealsense2'
```

AdamMiltonBarker, 2020-06-21:
I think I am pretty much done with Realsense, this is a joke.

Frame didn't arrive within 5000

AdamMiltonBarker, 2020-06-21:
I got it working making the changes above, then ran into a whole load of issues with NCS2, OpenVINO, but that is another story and the end of my time using Intel tech in my projects. 

To confirm the way I got it working was to add -DBUILD_PYTHON_BINDINGS=bool:true to cmake, maybe would be useful to add this to the docs or as an option in the script.

closed 2020-06-21 by AdamMiltonBarker