realsenseai/librealsense #6531: Unable to load recorded sequence into realsense2 viewer

jamjahal, 2020-06-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515| 
| Firmware Version                   | ? | 
| Operating System & Version |  MacOS  | 
| Kernel Version (Linux Only)    | N/A                                     | 
| Platform                                 | Mac |
| SDK Version                            |  2.34.0                      | 
| Language                            |  Python, shell                       | 
| Segment			|  AR                   | 

### Issue Description
Hi, The issue I'm seeing is not found in the support community. I'm unable to open .bag files in realsense2 sdk viewer. I keep gettinig the following message:

```
Failed to load file 20200602_122353.bag. Reason: Zero order filter requires both IR and Depth streams turned on.
Please rectify the configuration and rerun.
```

My question is how would I change the configuration of this file without being able to open it? is there a configuration file I need to edit? if so where might it be located? I am new to realsense sdk. 

I don't have the camera onsite. Working remotely with another site that has cameras and is sending files for me to build object detection and tracking model. Any help would be appreciated.

It seems that the Zero Order Filter is set to "On" by default.  However with only the .bag file to work with, I don't see a way to enable the zero order mode.  Please let me know if there is a way to rectify this.

MartyG-RealSense, 2020-06-06:
Responses to the question of @jamjahal have been posted on the Intel Support forum's version of this case.   Intel RealSense team members on the GitHub are welcome to contribute their own input though.

https://support.intelrealsense.com/hc/en-us/community/posts/360046922333

MartyG-RealSense, 2020-06-18:
Hi @jamjahal  Do you still require assistance with this case please?   Thanks!

MartyG-RealSense, 2020-06-22:
This discussion is continued onward from the end of https://support.intelrealsense.com/hc/en-us/community/posts/360046922333 on the Intel Support forum.

Hi again @jamjahal   At this point it has been established that you have tried the L515 bag file in the newest **2.35.2** version of the RealSense Viewer and it causes the Viewer to crash.

You considered the theory that as the bag file is several gigabytes in size, this may have caused the crash.  It did not occur on your colleagues' PC that created the file and they were able to run it fine.  It is possible that there is a difference in memory size between your PC and theirs.  I will consult with Intel about this.  Thanks so much for your patience.

MartyG-RealSense, 2020-06-22:
I received feedback from a senior member of the RealSense developer team who very kindly reviewed the whole of your case.  I have transcribed their advice into sections below for you to please work through.

1.  Do you have an Intel Non Disclosure Agreement (NDA)?  If so, you could share the L515 bag file with Intel.

2.  If the Viewer is crashing then it is probably not related to file size, as bag files do not get fully loaded into memory.  So the cause may be some other factor.

there could be an incompatibility of recordings from the pre-release (pre-PRQ) unit of the L515 camera that made the bag recording.  Though it was able to be played successfully on your colleagues' computers.

3.  You are using Mac OSX 10.15.4.  Are your colleagues using Windows or Linux PCs instead of Mac?  RealSense can be less stable on Macs.  It may be worth trying to access the file through RealSense on a Windows or Linux computer if you have such a computer available.  If not, then Windows VM on Parallels on your Mac may be an alternative route.

4.  There could also be a problem related to file access permissions.  An example of something that could cause this would be if the bag file is stored on a network drive.

5.  Another option that was advised that you could consider would be downgrading to an older RealSense SDK version such as **2.29.0** and see if that provides a reasonable workaround.

There may also be several ways to workaround the problem using code, though this approach may not be viable for you.

MartyG-RealSense, 2020-07-16:
Closed due to no further comments received.

closed 2020-07-16 by MartyG-RealSense