realsenseai/librealsense #6532: T265 pose data jumping

dschnabel, 2020-06-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version |   Linux (Raspbian 10.3)  | 
| Kernel Version (Linux Only)    |  4.19.97-v7l+ armv7l  | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                            |  2.0 (build 2.33.1) | 
| Language                            |  C++/ROS  | 
| Segment			|  Robot  | 

### Issue Description
The pose data (odometry) from my T265 continually jumps back and forth. Some locations are worse than others. 

Take a look:
[image]

I have disabled `RS2_OPTION_ENABLE_RELOCALIZATION` and `RS2_OPTION_ENABLE_POSE_JUMPING` but this didn't help:
```
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
auto pipe = make_shared<rs2::pipeline>();

auto sensor = cfg.resolve(*pipe).get_device().query_sensors().front();
sensor.set_option(RS2_OPTION_ENABLE_RELOCALIZATION, 0);
sensor.set_option(RS2_OPTION_ENABLE_POSE_JUMPING, 0);

pipe->start(cfg);
```

Ideas?

dschnabel, 2020-06-09:
This seems to be a problem with rtabmap transform. Closing since not related to the T265.

closed 2020-06-09 by dschnabel

CeccAnd, 2020-10-25:
Hello 

I would try to disable pose data jumping because seems make issue with SLAM.

I am pretty new here. Where are RS2_OPTION_ENABLE_RELOCALIZATION and RS2_OPTION_ENABLE_POSE_JUMPING?

Where are located? Thanks

dschnabel, 2020-10-26:
They are located [here](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/h/rs_option.h#L80-L81). Take a look at my code snippet above how I disable them using `sensor.set_option()`

CeccAnd, 2020-10-26:
Thank you Daniel for your feedback!

Sorry for my poor knowledge, but where I need to place your code?

I have installed Realsense with ROS. I do not see librealsense2.

Do I need to download first another program/wrapper?

Many thanks.

Andrea

dschnabel, 2020-10-27:
Hi @CeccAnd, maybe take a look at the examples that come with librealsense and try to compile and run them, for example https://github.com/IntelRealSense/librealsense/tree/master/examples/pose. This example shows how to create a configuration and how to start a pipe.

Once you understand the code in the example you will know where to place the call to `sensor.set_option()`.