realsenseai/librealsense #6533: Segfault restarting T265 Pipe

csr-kick, 2020-06-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   |  0.2.0.951 | 
| Operating System & Version |  Raspbian / Raspbery Pi OS  | 
| Kernel Version (Linux Only)    |  4.19.97-v7l+ -> 5.4.44-v7l+                                      | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                            | 2.33.1 -> 2.35.1                         | 
| Language                            |  C++                          | 
| Segment			|  Robot                     | 

Having a segfault when stopping a pipe, making a new pipe (as a shared pointer) and starting the new pipe with the existing config.

As documented in #6362 the T265 has a frequent issue with USB communication on the RPI4, where it seems the SDK stops asking for data, and the device the stops sending with the error:

`02/06 01:04:05,676 INFO [2949632528] (tm-device.cpp:1408) T265 FW message: 13042424847758289392: [0x�/2:721] Host not reading - stopping`

In our efforts to find a solution, we tried to find a workaround whereby we set a timeout, and if no message has been received, catch the error and restart the pipe. 

My current code is below:
```
    rs2::log_to_console(RS2_LOG_SEVERITY_INFO);

    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
    auto pipe = std::make_shared<rs2::pipeline>();
    pipe->start(cfg);
    unsigned int to = 20000;
    while (true)
    {
        try {
            
            auto frames = pipe->wait_for_frames(to);
            to = 40;
            auto f = frames.first_or_default(RS2_STREAM_POSE);
            // Cast the frame to pose_frame and get its data
            auto pose_data = f.as<rs2::pose_frame>().get_pose_data();
    
            // Print the x, y, z values of the translation, relative to initial position
            std::cout << "\r" << "Device Position: " << std::setprecision(3) << std::fixed << pose_data.translation.x << " " <<
                pose_data.translation.y << " " << pose_data.translation.z << " (meters)" << std::endl;
        }
        catch (const rs2::error& e) {
            to = 20000;
            std::cout << "T265 crashed. Restarting ..." << std::endl;
            pipe->stop();
            pipe = std::make_shared<rs2::pipeline>();
            pipe->start(cfg);
            std::cout << "... restarted." << std::endl;
        }
    }
```
This will catch and restart the data pretty well, typically taking under a second to recover. A recovery is performed roughly once every 30 seconds - 3 minutes. However, sometimes instead of recovering we get a segfault.

So, in lieu of an actual solution to the rebooting problem, is there a way to safely (and quickly) dispose of whatever handles, resources, and callbacks belong to the pipe, so that when we make and start a new one maybe we avoid the segfault? Ideally we restart the pipe as fast as possible to avoid the robot's position changing too much during the reboot.

RealSenseSupport, 2021-03-29:
Attaching a PR worth reviewing for this issue:  https://github.com/IntelRealSense/librealsense/pull/8561

ankyur, 2021-04-01:
> 
> 
> Attaching a PR worth reviewing for this issue: #8561

No. This is another issue.
But it's also easy to fix. You just need to call cfg.disable_all_streams() before starting pipeline again. And don't forget enable stream one more time:

```
catch (const rs2::error& e)
{
    to = 20000;
    std::cout << "T265 crashed. Restarting ..." << std::endl;
    pipe->stop();
    cfg.disable_all_streams();                               // add these two lines
    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);     //
    pipe = std::make_shared<rs2::pipeline>();
    pipe->start(cfg);
    std::cout << "... restarted." << std::endl;
}
```

hfdlwt, 2023-08-02:
Hi @csr-kick  May I ask you a question, where should the code you mentioned be modified in the camera program？

    rs2::log_to_console(RS2_LOG_SEVERITY_INFO);

    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
    auto pipe = std::make_shared<rs2::pipeline>();
    pipe->start(cfg);
    unsigned int to = 20000;
    while (true)
    {
        try {
            
            auto frames = pipe->wait_for_frames(to);
            to = 40;
            auto f = frames.first_or_default(RS2_STREAM_POSE);
            // Cast the frame to pose_frame and get its data
            auto pose_data = f.as<rs2::pose_frame>().get_pose_data();
    
            // Print the x, y, z values of the translation, relative to initial position
            std::cout << "\r" << "Device Position: " << std::setprecision(3) << std::fixed << pose_data.translation.x << " " <<
                pose_data.translation.y << " " << pose_data.translation.z << " (meters)" << std::endl;
        }
        catch (const rs2::error& e) {
            to = 20000;
            std::cout << "T265 crashed. Restarting ..." << std::endl;
            pipe->stop();
            pipe = std::make_shared<rs2::pipeline>();
            pipe->start(cfg);
            std::cout << "... restarted." << std::endl;
        }
    }

closed 2026-10-23 by MartyG-RealSense