realsenseai/librealsense #6536: Extract metadata (fps/resolution/color,depth,ir formats, etc) from a bag file in python

avasbr, 2020-06-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Debian   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.34                       | 
| Language                            |  python                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

Apologies if this has been asked already, but I couldn't quite find what I was looking for in my search. Is there a simple way to extract all metadata (fps, resolution, color/depth/ir format, etc) given a random bag file, in python? Basically all the things I'd need to feed into `config.enable_stream`, to start playback.

Thanks

MartyG-RealSense, 2020-06-07:
Hi @avasbr  There have been some RealSense users who have extracted bag file data to files such as images and .csv depth matrix files.  It sounds as though you would prefer to read the metadata information directly from the bag file instead of saving it as text into a csv file and then reading the information from the csv file.  Is that a correct interpretation, please?

If you prefer to try to access metadata information directly with Python, the discussion in the link below may provide useful insights.

https://github.com/IntelRealSense/librealsense/issues/3179

avasbr, 2020-06-07:
@MartyG-RealSense thanks for the quick response! yes, ideally i'd want to read that info directly in python, but if push comes to shove, I don't mind going the csv approach either. Are there any links you're able to provide to past issues around this? in the meantime i'll check out the issue you linked above

MartyG-RealSense, 2020-06-07:
@avasbr   The long discussion in the link below looks at approaches to writing camera data to csv file.

https://github.com/IntelRealSense/librealsense/issues/1485

MartyG-RealSense, 2020-06-25:
Case closed due to no further comments received.

closed 2020-06-25 by MartyG-RealSense

avasbr, 2020-07-02:
Sorry, I let this go for a while, but these links aren't providing me the information I need. When I read a bag file and process it, I have to pass in the following settings: 

```bag_settings = {
    "height": 1280,
    "width": 720,
    "color_fps": 30,
    "depth_fps": 30,
    "ir_fps": 30,
    "color_format": "rgb8",
    "depth_format": "z16",
    "ir_format": "y8",
}```

I would like to be able to, in python, do something along the lines of `get_metadata(<path_to_bag_file>)` and it just outputs this information. is there a full example you can point me to which does this?

MartyG-RealSense, 2020-07-02:
The SDK has a C++ tool called **rosbag-inspector** that can read bag file contents.

https://github.com/IntelRealSense/librealsense/tree/master/tools/rosbag-inspector

I re-researched the subject but did not find any new links for Python other than the ones shared earlier, unfortunately.