realsenseai/librealsense #6537: Intel RealSense Viewer “No device to found.”

LYKlyk, 2020-06-07:
OS:window
Realsense SR300
usb3.0

"Intel RealSense Viewer" -> "Add Source"-> "No SR300 Device"
Intel RealSense Viewer can not show data flow.

MartyG-RealSense, 2020-06-07:
Please check in your Windows privacy settings if access to the camera hardware and camera applications are both set to **On**.  Instructions for doing this can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

LYKlyk, 2020-06-07:
@MartyG-RealSense  Sorry, It still can not work.

MartyG-RealSense, 2020-06-07:
Please open your Device Manager in Windows, look under the "Camera" category of it and see if the SR300 camera drivers are there.

You can find the Device Manager by typing 'device' into the text box at the bottom of your Windows screen, next to the button in the corner where you shut the PC down.

LYKlyk, 2020-06-07:
Yes, It contain "SR300 RGB" and "SR300 Depth"

LYKlyk, 2020-06-07:
Yes, It contain "SR300 RGB" and "SR300 Depth". But, It still can not work.

MartyG-RealSense, 2020-06-07:
There is a third driver called the Virtual Driver.  If this is missing then the SR300 may not work.  Could you look under the "Imaging Devices" category of the Device Manager please if you have that category and see if the Virtual Driver is there.

[image]

LYKlyk, 2020-06-07:
Yes, I can not found "Virtual Driver" in "Imaging Devices" category. What should I do? How can I install "Virtual Driver"？ @MartyG-RealSense

MartyG-RealSense, 2020-06-07:
There are a few approaches that could be tried for restoring  a missing Virtual Driver.  Let's start with the easiest: using the **Windows Troubleshooter** tool to see if it can find a hardware problem and auto-correct it.

https://forums.intel.com/s/question/0D50P00004McnzkSAB/i-have-a-intel-sr300-but-i-cannot-install-the-realsense-camera-sr300-virtual-driver

LYKlyk, 2020-06-07:
Sorry, the Windows Troubleshooter tool still cannot install "SR300 Virtual Driver".  Is there another methods? @MartyG-RealSense

MartyG-RealSense, 2020-06-07:
The next method requires an internet connection.

Go back to the Device Manager and right-click on the SR300 Depth and RGB drivers.  On each one, select Uninstall to remove it.  

When both drivers are moved from the Device Manager, unplug the camera, wait a couple of seconds and plug the camera back in.  Windows Update should automatically download and install the SR300 drivers, and hopefully this will result in the Virtual Driver being restored too.

LYKlyk, 2020-06-07:
Sorry, It still can not work. @MartyG-RealSense

MartyG-RealSense, 2020-06-07:
Please visit the link below and find the section titled **Intel® RealSense™ Camera SR300 Firmware**.  Left-click on the 'Drivers' link below it to download a driver installer program called the **Depth Camera Manager (DCM)**.

https://software.intel.com/content/www/us/en/develop/articles/realsense-sdk-windows-eol.html

When the DCM program has finished downloading, please run the file to start the SR300 driver installation process.

LYKlyk, 2020-06-08:
Sorry, When I use "DCM" to install the SR300 driver, it always show **" Setup cannot detect Intel & real  sense in this system ™ 3D camera. Installation terminated."** What should i do? @MartyG-RealSense

MartyG-RealSense, 2020-06-08:
I'm sorry that the DCM did not work for you.  Sometimes it has problems with modern versions of Windows 10.  There is a method for running the DCM file to initiate the driver installation without running the DCM program and therefore potentially avoiding the incompatibility problem.

https://support.intelrealsense.com/hc/en-us/articles/360022951533-Windows-10-Issues-with-Intel-RealSense-Cameras-SR300-and-F200

LYKlyk, 2020-06-12:
I have tried this method, but it still can not work. I still can not find "SR300 Virtual Driver", and "2016 R2 SDK" provide some tools, they still cannot detect the "SR300" device. @MartyG-RealSense

MartyG-RealSense, 2020-06-12:
The next method to try is even more complicated, unfortunately.

https://support.intelrealsense.com/hc/en-us/community/posts/360038275074-Managed-SR300-to-work-on-latest-Windows-10-18363-535-

MartyG-RealSense, 2020-06-25:
Hi @LYKlyk   Do you require further assistance on this case please, or can it be closed?  Thankd!

LYKlyk, 2020-06-25:
Sorry, It is still can not work. "2016 R3 SDK" still show can not find device. @MartyG-RealSense

MartyG-RealSense, 2020-06-25:
The problem with trying to fix an SR300 is not that there aren't many solutions.   The problem is that there are so many possible workarounds that can be tried that you might spend days or weeks trying them all and not succeeding.

An example is the discussion in the link below, which has 91 comments on it.

https://community.intel.com/t5/Items-with-no-label/How-to-get-SR300-working-on-Windows-10-Build-1703-Including-SDK/td-p/498958

I ran some further tests on my own computers.  I tried using the SR300 on my laptop.  The drivers installed over the internet but the Virtual Driver was missing.  I then tried the DCM installer program for the SR300 on the laptop and the SR300 camera was not detected.

I remembered that the DCM program for the F200 (the SR300's predecessor) also installed the Virtual Driver.  Because I had an F200 camera, I was able to successfully run the F200 DCM and it installed the Virtual Driver.

I unplugged the F200 and plugged in the SR300.  The F200 Virtual Driver was still present in the Device Manager under the **Imaging Devices** category, confirming my recollection that both cameras used the same Virtual Driver.  

This was not directly helpful for your situation though, as you need to own an F200 camera for the F200 DCM program to work.  It did seem to establish though that the DCM is helpful for installing the Virtual Driver if it is missing.

I tried extracting the driver from my computer to test whether it could be installed on another computer.  That did not work, unfortunately.

All of this backed up the experiences of SR300 users in the past ... if you have tried this many fixes and it still does not work, it probably is not going to work whatever is tried.

closed 2020-07-18 by MartyG-RealSense