realsenseai/librealsense #6538: Problem when build with libusb

xmmzzz, 2020-06-07:
I want to build pyrealsense2 from source code on Jetson TX2. When I follow the 
Python Wrapper/readme and use make to make files, there is a question that libusb/libusb/core.c need include<config.h>, but there is no such file named 'config.h' in such dir, how can I solve this question?

MartyG-RealSense, 2020-06-07:
Hi @xmmzzz   A solution for this problem that has worked for some RealSense users is to use the command below:

**sudo apt-get install libusb-1.0-0-dev**

xmmzzz, 2020-06-08:
Thanks for your quick response, I have solved the problem following your guidance.

MartyG-RealSense, 2020-06-08:
Great news - thanks for the update!

MartyG-RealSense, 2020-06-18:
I will close the case now as a solution was achieved.   Thanks!

closed 2020-06-18 by MartyG-RealSense