realsenseai/librealsense #6540: realsense-net device in C# wrapper

SyEns, 2020-06-07:
Camera Model: D435i & D415
Language: C#/unity

In SDK version 2.34 realsense-net and rs-server got introduced. I successfully got the cpp examples working with a D435i connected to a Raspberry Pi 4 running rs-server. Now I wonder if it is possible to add a realsense-net device to a context and initialize the pipeline with the context in C# like in cpp, especially for unity. I couldn't find a C# wrapper for realsense-net. How could I get a C# app running with the realsense over ethernet?

RealSenseSupport, 2020-12-05:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-12-15 by RealSenseSupport

AndreV84, 2021-06-08:
are there also other wrappers for that? 
C, python etc? for using realsense-net towards network rs-server device?