realsenseai/librealsense #6544: How can I measure distance between two points in an image

Chanch2000, 2020-06-08:
I have an image taken from a normal camera. How can I measure the distance between two points in the image using realsense api. I detected the point coordinates using openCV. So I have the coordinates of the points. Now I just want to measure the distance between them. 
May I know if that can be done?

MartyG-RealSense, 2020-06-08:
As you mention doing the capture with a 'normal' camera, it sounds as though the image is a non-RealSense image file such as a PNG.  Is that correct, please, or is it a live color video image?

I researched your question very carefully to look at the available options for achieving a solution.  There was a past case where a RealSense user outlined the method that they had used to calculate the height of an object with OpenCV, apparently using live camera data instead of an image file.

https://forums.intel.com/s/question/0D50P0000490TN7SAM/how-to-compute-objects-height-for-d435?language=en_US

If using OpenCV is not an absolute requirement of your project, the RealSense SDK has an example program called **rs-measure** in the C++ language for measuring the distance between two points that also uses live camera data.

https://github.com/IntelRealSense/librealsense/tree/master/examples/measure

In regard to options for analysing pre-made files instead of live camera data:  there is a project for the Python language that uses OpenCV to measure the distance between two points, with the data used in the measuring taken from a **bag** format file.

https://github.com/soarwing52/RealsensePython

The measuring part of this project was originally based on a Python conversion of the C++ rs-measure program, before the developer expanded the scope of the project.  That code is available on its own here:

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

I hope that some of this information will be useful to you.

Chanch2000, 2020-06-08:
Thank you for such a quick and detailed reply 

OpenCV in my case is not a necessary requirement.

I saw the rs-measure.cpp code. I have a few questions. Sorry since I’m a rookie things are not very clear 

1. Will the rs-measure.cpp code run in the same way for a normal camera? 

2. One line in rs-measure.cpp code is 

    rs2_intrinsics intr = frame.get_profile().as<rs2::video_stream_profile>().get_intrinsics(); // Calibration data

Will I get correct calibration data for a normal camera. By normal I mean a webcam connected to my laptop

MartyG-RealSense, 2020-06-08:
This may be a project that does not actually require RealSense SDK or a RealSense camera at all, as OpenCV on its own should be able to do it.  For more details, please google for 'opencv measure between two points' to find a lot of examples in the search results.

Here is an example, that describes measuring between two points with an OpenCV feature called **norm**:

https://www.quora.com/Can-I-get-the-measurement-between-two-points-in-an-image-and-also-the-width-of-the-body-features-using-tensor-flow-or-OpenCV-in-Java

Chanch2000, 2020-06-08:
Thanks for leading me in the right direction. 
I’ll close the issue

closed 2020-06-08 by Chanch2000

MartyG-RealSense, 2020-06-08:
Glad I could help!

I have not to an ordinary RGB USB webcam to test with right now.  My experience with using built-in ordinary webcams on laptops though is that the webcam is selectable in the RealSense SDK's "RealSense Viewer" software as a **Platform Camera** with the Add Source button.  Once detected, you can then access the RGB camera functions only.

[image]

The RealSense SDK can also access streams such as RGB from non-RealSense cameras.  So you could use a RealSense camera for a depth stream and pair it with a non-RealSense high resolution RGB camera to provide the color stream.

rawathemant246, 2024-04-24:
Hii,  i have a question, i want to create an app  basically the functionality is  i want to caculate the area of the garden in the realtime using the camera live feed can you help me how can i do that ?   

can you please help me figuring out the solution how can i do this ?

MartyG-RealSense, 2024-04-24:
Hi @rawathemant246  

You could perhaps use the RealSense SDK's **rs-measure** C++ example as the foundation for your app.  You could stand at the bottom of the garden with the camera and point it towards the end of the garden, and then draw a measuring line with the rs-measure tool from the garden start corner to its end corner.

https://github.com/IntelRealSense/librealsense/tree/master/examples/measure

[image]

The measurement value is displayed in cm.  So 280 cm would represent 2.8 meters.

Most RealSense 400 Series camera models only capture depth up to a distance of 10 meters from the camera though, which would make them suitable only for small gardens.  The D455 class of RealSense camera model can depth-sense up to 20 m though, which may make it the best choice for this garden application.

rawathemant246, 2024-04-24:
sorry for the silly question can this works  for the android mobile devices also ?

also this is my used 

[image]

suppose this is the picture captured by the android iphone can we still use the measure.py to calculate the area of the rough backyard

MartyG-RealSense, 2024-04-24:
It's not a silly question, so don't worry.  :)

Android code would be different from C++ code as it is usually based upon the Java language (though you can access the RealSense SDK's C++ instruction set from Android).  There is a discussion at https://github.com/IntelRealSense/librealsense/issues/9466 about implementing measurement between two points in Android.

In regard to the picture that you captured, if you can achieve an RGB color image from your RealSense camera that can also capture the whole front garden area then you should be able to capture it as a depth image too, since the size of the field of view for depth on RealSense cameras will be the same or larger than the RGB viewpoint size.  Camera models in the D455 class (D455, D455f, D456, D457, etc) will have the largest view size for depth and RGB.