realsenseai/librealsense #6546: Android - Runtime exception when using getStreamProfiles

Murded, 2020-06-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415 | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |   Android 9 | 
| Platform                                 | Samsung Galaxy Tab Active2 |
| Language                            | Java | 

### Issue Description
Hi,

Please let me first start of by saying that I don't think this is an issue with my current code as I have it working fine on other devices, however I have been tasked to investigage why it doesn't working on the Samsung Galaxy Tab Active2.

I have the following simple bit of code that I'm using to query the device stream profiles

```csharp
Device d = mRsContext.queryDevices().createDevice(0);

for (Sensor sensor : d.querySensors()) 
{
    for (StreamProfile sp : sensor.getStreamProfiles())
    {
        //CRASHES AT THE .getStreamProfiles() call above
    }
}
```

However it produces the following error message:

```
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542] JNI DETECTED ERROR IN APPLICATION: JNI GetObjectClass called with pending exception java.lang.RuntimeException: null pointer passed for argument "mode"
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void com.intel.realsense.librealsense.StreamProfile.nGetProfile(long, com.intel.realsense.librealsense.StreamProfile$ProfileParams) (StreamProfile.java:-2)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void com.intel.realsense.librealsense.StreamProfile.<init>(long) (StreamProfile.java:20)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at java.util.List com.intel.realsense.librealsense.Sensor.getStreamProfiles() (Sensor.java:16)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void com.gpcsl.freightmeasure.RealsenseActivity.ConfigAndStart() (RealsenseActivity.java:983)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void com.gpcsl.freightmeasure.RealsenseActivity.Start() (RealsenseActivity.java:1092)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void com.gpcsl.freightmeasure.RealsenseActivity.SetupDevice() (RealsenseActivity.java:879)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void com.gpcsl.freightmeasure.RealsenseActivity.access$3000(com.gpcsl.freightmeasure.RealsenseActivity) (RealsenseActivity.java:86)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void com.gpcsl.freightmeasure.RealsenseActivity$2.onDeviceAttach() (RealsenseActivity.java:435)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void com.intel.realsense.librealsense.DeviceWatcher.addDevice(android.hardware.usb.UsbDevice) (DeviceWatcher.java:113)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void com.intel.realsense.librealsense.DeviceWatcher.invalidateDevices() (DeviceWatcher.java:78)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void com.intel.realsense.librealsense.DeviceWatcher.access$000(com.intel.realsense.librealsense.DeviceWatcher) (DeviceWatcher.java:16)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void com.intel.realsense.librealsense.DeviceWatcher$1.onDeviceAttach() (DeviceWatcher.java:44)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void com.intel.realsense.librealsense.Enumerator.notifyOnAttach() (Enumerator.java:102)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void com.intel.realsense.librealsense.Enumerator.access$200(com.intel.realsense.librealsense.Enumerator) (Enumerator.java:18)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void com.intel.realsense.librealsense.Enumerator$MessagesHandler.handleMessage(android.os.Message) (Enumerator.java:144)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void android.os.Handler.dispatchMessage(android.os.Message) (Handler.java:106)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void android.os.Looper.loop() (Looper.java:214)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   at void android.os.HandlerThread.run() (HandlerThread.java:65)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542] 
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]     in call to GetObjectClass
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]     from void com.intel.realsense.librealsense.StreamProfile.nGetProfile(long, com.intel.realsense.librealsense.StreamProfile$ProfileParams)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542] "DeviceManager device availability message thread" prio=5 tid=17 Runnable
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   | group="main" sCount=0 dsCount=0 flags=0 obj=0x12c6e3a8 self=0xd3fc6600
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   | sysTid=22324 nice=0 cgrp=default sched=0/0 handle=0xd1095970
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   | state=R schedstat=( 0 0 0 ) utm=5 stm=5 core=5 HZ=100
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   | stack=0xd0f92000-0xd0f94000 stackSize=1042KB
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   | held mutexes= "mutator lock"(shared held)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   native: #00 pc 002e5b0f  /system/lib/libart.so (art::DumpNativeStack(std::__1::basic_ostream<char, std::__1::char_traits<char>>&, int, BacktraceMap*, char const*, art::ArtMethod*, void*, bool)+134)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   native: #01 pc 003804ed  /system/lib/libart.so (art::Thread::DumpStack(std::__1::basic_ostream<char, std::__1::char_traits<char>>&, bool, BacktraceMap*, bool) const+212)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   native: #02 pc 0037caab  /system/lib/libart.so (art::Thread::Dump(std::__1::basic_ostream<char, std::__1::char_traits<char>>&, bool, BacktraceMap*, bool) const+34)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   native: #03 pc 00239ce7  /system/lib/libart.so (art::JavaVMExt::JniAbort(char const*, char const*)+722)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   native: #04 pc 0023a04f  /system/lib/libart.so (art::JavaVMExt::JniAbortV(char const*, char const*, std::__va_list)+58)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   native: #05 pc 000c56a7  /system/lib/libart.so (art::(anonymous namespace)::ScopedCheck::AbortF(char const*, ...)+46)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   native: #06 pc 000c4349  /system/lib/libart.so (art::(anonymous namespace)::ScopedCheck::CheckPossibleHeapValue(art::ScopedObjectAccess&, char, art::(anonymous namespace)::JniValueType)+1080)
2020-06-08 15:37:30.350 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: java_vm_ext.cc:542]   native: #07 pc 000c382b  /system/lib/libart.so (art::(anonymous namespace)::ScopedCheck::Check(art::ScopedObjectAccess&, bool, char const*, art::(anonymous na [... cut ...]
2020-06-08 15:37:30.824 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: runtime.cc:585]   native: #111 pc 001ecedb  /system/lib/libart.so (art::interpreter::ArtInterpreterToCompiledCodeBridge(art::Thread*, art::ArtMethod*, art::ShadowFrame*, unsigned short, art::JValue*)+230)
2020-06-08 15:37:30.824 22245-22324/com.gpcsl.freightmeasure A/.freightmeasur: runtime.cc:585]   native: #112 pc 001e7b55  /system/lib/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction co

MartyG-RealSense, 2020-06-08:
If a program works with one Android device but not another, it is worth first checking whether the device that does not work has Android 10 installed, as it is known to have conflicts with the Librealsense Android wrapper.

I know you listed it as being an Android 9 device in the case description, but could you check and confirm please that it is not running Android 10.

If it does turn out to have Android 10, a simple fix for the incompatibility that usually works is to set **targetSdkVersion** to an earlier version.  '**27**' is a targetSdkVersion value that has been proven to solve the conflict.

Murded, 2020-06-08:
> If a program works with one Android device but not another, it is worth first checking whether the device that does not work has Android 10 installed, as it is known to have conflicts with the [...]

Hi @MartyG-RealSense  thanks for the reply, I have confirmed that this is the case.

I'm aware of the issue with android 10 and already have that fix in place as we also have android 10 devices that work fine

MartyG-RealSense, 2020-06-08:
Despite extensive research, I could not find many useful leads.  The Samsung Galaxy Tab Active 2 is apparently aimed at business customers rather than consumers though.  this made me wonder if there is some kind of extra protection installed for enterprise customers that prevents the Android wrapper from accessing it.

Murded, 2020-06-08:
> Despite extensive research, I could not find many useful leads. The Samsung Galaxy Tab Active 2 is apparently aimed at business customers rather than consumers though. This made me wonder if there [...]

Hmm okay, I'll dig around the tablet tomorrow to see if there's anything obvious in the settings I can access.

If you have any other ideas please let me know, I will contact samsung to see if they have any ideas and will update here if I find anything.

MartyG-RealSense, 2020-06-08:
My Android wrapper knowledge is currently limited but I will do an informal tag into this discussion of the wrapper developer @matkatz in the hope that he can kindly offer input.

Murded, 2020-06-08:
> My Android wrapper knowledge is currently limited but I will do an informal tag into this discussion of the wrapper developer @matkatz in the hope that he can kindly offer input.

Brilliant, we may have a contact at samsung that we're going to reach out to as well as contact their support to see if they can offer some insight.

Murded, 2020-06-08:
Small update, we were given permission to reset the android device (So it's no longer encrypted) however the issue still continues.

MartyG-RealSense, 2020-06-08:
Thanks very much for trying!

Murded, 2020-06-12:
After speaking to someone at the Samsung support team, this is the latest response.

Hi Andrew,
We have checked the log and GitHub. The exception is thrown because one internal data structure, rs2_stream_profile, of Intel librealsense contains an invalid null value. Also, Intel SDK does not control their device with android camera API. It seems that the library controls the device with direct USB comm.

Tab A and Active Tab 2 use different AP chipset, so there may be differences in whether USB Video Class devices are supported or not. However, it is difficult to check the association between UVC support and invalid value in the internal data structure of librealsense from the Samsung side.
Further, analyze the issue from the Samsung side seems not feasible until Intel narrows down the reason for the invalid value of internal data in the library after reviewing their code.

Looking forward to your cooperation.

Best Regards,
Md. Iqbal Hossain
Samsung Developer Program

MartyG-RealSense, 2020-06-12:
I will pass Samsung's message on to the RealSense team.  Thanks so much for providing the information.

Murded, 2020-06-12:
Thanks so much, would love to be kept up to date if there is going to be any action made. (we're currently debating suggesting that type of tablet to a customer)

Murded, 2020-07-29:
@MartyG-RealSense Do you have any update on this issue? we're also experiencing it on the new Samsung Tab A 10.1

MartyG-RealSense, 2020-07-29:
@Murded   I have not got new information about it, but I will add your experience of the problem with the new Samsung Tab A 10.1 to the bug report that I already filed.

MartyG-RealSense, 2021-08-24:
Hi @Murded   I have received feedback from Intel regarding the bug report associated with this Android case.  They said that the bug involved in this case should already be fixed in the SDK.  Could you test with a current SDK please to see whether the issue is still present for you?  Thanks!

MartyG-RealSense, 2021-08-29:
Hi @Murded  Have you had the opportunity to test with a current SDK as suggested above, please?  Thanks!

MartyG-RealSense, 2022-02-13:
Case closed due to fix implemented in android/jni/sensor.cpp and no further comments received.

closed 2022-02-13 by MartyG-RealSense