realsenseai/librealsense #6550: ImportError: librealsense2.so.2.24: cannot open shared object file: No such file or directory

wangjunfeng90214, 2020-06-09:
I have installed pyrealsense2, and when I run in the python:
```bash
wangjunfeng@PVG-DSK-009:~$ python
Python 3.6.9 (default, Apr 18 2020, 01:56:04) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2 as rs
>>> pipeline = rs.pipeline()
>>> 
```
It's ok.
But when I built a test.py like this:
```bash
import pyrealsense2 as rs
print('Done')
```
and run the test.py
```bash
python test.py
```
and I got this error
```bash
$ python test/test_realsense2_point.py 
Traceback (most recent call last):
  File "test/test_realsense2_point.py", line 136, in <module>
    import pyrealsense2 as rs
ImportError: librealsense2.so.2.24: cannot open shared object file: No such file or directory
```
How can I fix this issue?

MartyG-RealSense, 2020-06-09:
Hi @wangjunfeng90214   Have you installed pyrealsense2 before?   If so, it is possible that a conflict could arise if there are now multiple versions of it on your computer.

https://github.com/IntelRealSense/librealsense/issues/4519

MartyG-RealSense, 2020-06-18:
@wangjunfeng90214  Do you require further assistance with this case please?   Thanks!

closed 2020-06-25 by MartyG-RealSense