realsenseai/librealsense #6554: XYZ reconstruction on aligned images

EnricoBeltramo, 2020-06-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D235 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win10  | 
| Kernel Version (Linux Only)    |  2.33.1.1360                                       | 
| Platform                                 | PC |
| SDK Version                            |  { 2.33.1.1360  }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I have to create an application similar to box_dimensioner_multicam_demo, but in addition to that I need to align the deepth image with color image. Honever, if I modifiy the example simply adding an alignement before get the frameset:

    def poll_frames(self):
        """
        Poll for frames from the enabled Intel RealSense devices. This will return at least one frame from each device. 
        If temporal post processing is enabled, the depth stream is averaged over a certain amount of frames
        
        Parameters:
        -----------
        """
        frames = {}
        while len(frames) < len(self._enabled_devices.items()) :
            for (serial, device) in self._enabled_devices.items():
                streams = device.pipeline_profile.get_streams()
                frameset = device.pipeline.poll_for_frames() #frameset will be a pyrealsense2.composite_frame object
                if frameset.size() == len(streams):

                    **frameset = self.align.process(frameset)**
                    frames[serial] = {}
                    for stream in streams:
                        if (rs.stream.infrared == stream.stream_type()):
                            frame = frameset.get_infrared_frame(stream.stream_index())
                            key_ = (stream.stream_type(), stream.stream_index())
                        else:
                            frame = frameset.first_or_default(stream.stream_type())
                            key_ = stream.stream_type()
                        frames[serial][key_] = frame

        return frames

but after that, both the calibration and measurement doesn't work correctly. The estimation of xyz should not be same after the alignment? Or is necessary to change something (i.e. different value of intrinsic)?

MartyG-RealSense, 2020-06-09:
Generally speaking, it is recommended that alignment is done **after** processing work has been done on the data (for example, aligning after post-processing filters have been applied and not before).  This can help to avoid distortion such as aliasing (jagged lines).

EnricoBeltramo, 2020-06-09:
Yea, unterstood. But in this case I removed any postprocessing and still the results doesn't match. The question is what about need to change in terms on calculation when I use the aligned frame. I need to apply a different estrinsic / intrinsic matrix for calculation?

MartyG-RealSense, 2020-06-09:
Can you confirm please that your project is an adaptation of the **box_dimensioner_multicam_demo** example program, please?  And if so, are you using a calibration chessboard image like in that project to put objects on?

EnricoBeltramo, 2020-06-11:
Yes, I did. I did any time the calibration with chessboard. Respect original project I just added the alignment. Without it, everything works correctly, if I add it I have the mismatch.

MartyG-RealSense, 2020-06-11:
I asked about the chessboard because I remembered with that project that if the size of the calibration chessboard is changed then the values in the script have to be updated, otherwise the measurements become incorrect.

https://github.com/IntelRealSense/librealsense/issues/3540

MartyG-RealSense, 2020-06-25:
Hi @EnricoBeltramo   Do you still require assistance with this case please, or can it be closed?  Thanks!

EnricoBeltramo, 2020-06-28:
I fixed writing a different calibration strategy and making all the related transformations by myself.

closed 2020-06-28 by EnricoBeltramo

MartyG-RealSense, 2020-06-28:
I'm very pleased that you found a solution - thanks so much for the update!