realsenseai/librealsense #6555: failed to set power state

viisautta, 2020-06-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.12.5 | 
| Operating System & Version |   Fedora32 | 
| Kernel Version (Linux Only)    |  5.6.15   | 
| Platform                                 | DELL Precision 3530 |
| SDK Version                            |  2.35.0   | 
| Language                            | c++    | 
| Segment			|  others                         | 

### Issue Description 
While testing some SDK-provided samples like rs-align, rs-color and etc., there always is output as below:
RealSense error calling rs2_pipeline_start_with_config(pipe:0x135bd90, config:0x13601d0):
    failed to set power state

MartyG-RealSense, 2020-06-09:
Hi @viisautta  Could you run **rs-enumate-devices** on the DELL Precision 3530 computer please and provide the output please?  Thanks!

https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices

viisautta, 2020-06-10:
some captured outputs are as below:
$ ./rs-enumerate-devices 
 10/06 08:37:30,867 ERROR [139713792964352] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
Could not create device - failed to set power state . Check SDK logs for details

closed 2020-06-10 by viisautta

MartyG-RealSense, 2020-06-10:
No problem at all. I'm pleased to hear that you found a solution!

stino78, 2020-10-07:
I had a similar problem on Raspberry Pi OS, using pyrealsense2 and the D435i

I was opening two pipelines, one for the camera and one for the IMU. Starting the camera didn't give any problems, but when starting the IMU, right after, there was an error message "failed to set power state".

After may software and hardware attempts, I solved it in a trivial way... I added 500ms pause between the two starting.

Good luck!

jerry-loungelab, 2022-04-01:
hello, stino78
How did you solve the error message "failed to set power state"?

I will wait for your reply!
Thank you in advance:)

stino78, 2022-04-01:
Hi Jerry,

In my code I am opening two different pipelines, one for the color camera
and ine for the depth camera, for asynchronous reading.

The solution for me was to set a 1sec pause before opening the second
pipeline.

Good luck!

jerry-loungelab, 2022-04-01:
Thank you very much!

AkshajGupta, 2023-07-05:
I am having this error
`UNKNOWN in rs2_get_option(options:0x600003df9400, option:Brightness):
failed to set power state`

using just one d435i camera. I want to perform the firmware update through the realsense-viewer but it won't let me get through the error.

MartyG-RealSense, 2023-07-05:
Hi @AkshajGupta  Are you using the official 1 meter USB cable supplied with the camera or a longer cable of your own choice, please?

littlefred117, 2023-10-02:
Hi,

I am trying to establish a pipeline between my Mac and my LiDAR L515, and when I originally plugged the camera in, the RGB and sensor mode were working. I had to "sudo real sense-viewer" into RealSense, otherwise it would crash, but generally everything else was working. I then ran this code:

import pyrealsense2 as rs
import numpy as np

# Initialize the RealSense pipeline
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)

# Add point cloud stream
config.enable_stream(rs.stream.points, 640, 480, rs.format.xyz32f, 30)

# Start the pipeline with the given configuration
pipeline.start(config)

try:
    while True:
        # Wait for a frame
        frames = pipeline.wait_for_frames()

        # Get depth, color, and point cloud frames
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        points_frame = frames.get_points()

        if not depth_frame or not color_frame or not points_frame:
            continue

        # Get the raw depth and color data as numpy arrays
        depth_data = np.asanyarray(depth_frame.get_data())
        color_data = np.asanyarray(color_frame.get_data())

        # Get the raw point cloud data as a numpy array
        points_data = np.asanyarray(points_frame.get_vertices())

        # Print raw depth, color, and point cloud data
        print("Raw Depth Data:")
        print(depth_data)
        print("Raw Color Data (RGB):")
        print(color_data)
        print("Raw Point Cloud Data (XYZ):")
        print(points_data)

        # Save the point cloud data to a file
        np.savetxt('point_cloud.xyz', points_data, delimiter=' ')

finally:
    # Stop the pipeline
    pipeline.stop()

and from here it stopped working. The code would not compile because it said a device was not connected. Whenever I go to my RealSense, I get these errors: UNKNOWN in rs2_get_option(options:0x600003df9400, option:Brightness): failed to set power state. I was wondering if anyone could help.

MartyG-RealSense, 2023-10-02:
Hi @littlefred117  I have responded at the discussion that you created at https://github.com/IntelRealSense/librealsense/discussions/12244 - thanks!