realsenseai/librealsense #6556: Calculate texture in python

EnricoBeltramo, 2020-06-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D235 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win10  | 
| Kernel Version (Linux Only)    |  (-)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.33.1.1360 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I'm doing an application in python in multiprocessing to manipulate information from point cloud. 
I have a process that acquire frames and other separate that elaborate them in parallel.
Actually I'm using rs.pointcloud() to obtain points coords and texture coords and it works well:

# calculate the cloud points
pc.map_to(mapped_frame)
points = pc.calculate(depth_frame)
				
# Pointcloud data to arrays
v, t = points.get_vertices(), points.get_texture_coordinates()

verts = np.asanyarray(v).view(np.float32).reshape(-1, 3)  # xyz
texcoords = np.asanyarray(t).view(np.float32).reshape(-1, 2)  # 

but I can't use this approach in multiprocessing (is not possible to pass composite frame to different process: only arrays o base types). Is it possible to calculate explicity the uv map without rs.pointcloud() but only using color/depth map arrays and intrinsic? Using the box_dimensioner_multicam_demo I found how calculate the vertex easly, but I don't understand how calculate the uv map/ texture

MartyG-RealSense, 2020-06-09:
Hi @EnricoBeltramo   Before looking at creating a UV map, it may be worth making sure whether the multi-processing approach is going to be possible at all or not.

I had considered the Python multi-processing method called **pickling** but others have found that they were unable to use it for composite frames, as you suggested.

I did find a past case though where a RealSense Python user took the approach of storing intrinsics in a **dictionary**.

 https://forums.intel.com/s/question/0D70P000006Yr5OSAS

EnricoBeltramo, 2020-06-11:
I done the calculation. I convert the deepth point in xyz coords --> convert to RGB camera coords (using extrinsinc transformation) --> convert xyz to pixel coords. In this way I can pass only numpy arrays and all works fine.

MartyG-RealSense, 2020-06-11:
Great news - thanks so much for the update and for sharing your method with the community!

MartyG-RealSense, 2020-06-18:
Case closed due to solution being achieved.

closed 2020-06-18 by MartyG-RealSense