realsenseai/librealsense #6558: Severity: Error Description: USB CAM overflow  With some strange behavior

pjmara, 2020-06-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |       5.3.0-53-generic                               | 
| Platform                                 | PC (XPS 15 9570) |
| SDK Version                            |  2.35.0                          | 
| Language                            |  NA                           | 
| Segment			|  NA                        | 

### Issue Description

I'm a new user to the Intel D435. I plugged the camera in, booted up realsense-viewer for the first time, and it worked amazing! I then tried to update the device firmware, and something broke. I've since rolled back to a firmware where I know it was working at some point (5.11.15.00). I'll describe what I've done to try and remedy but **I don't know what logs would be most helpful to solve, so I will need some guidance for that**

The USB CAM overflow and USB SCP overflow issue seems to be what is happening to me. I have seen this issue produced in many places. When opening realsense viewer, if I try to enable the camera I receive choppy images with lots of errors like the one in the title of this issue.  Here is how my system has responded to some of the suggested troubleshooting:

### Troubleshooting attempts

Choosing a different cable
I've tried a different USB cable than the time I one I used on the first successful attempt. It was a usb 3 cable, I'm sure of it, and I am using the usb 3 ports on my XPS 15. It is not a very long cable either, < 5ft. No change in the issue.

I've also tried increasing the USBFS memory size to 1gb in case that was the issue

I'm currently in Ubuntu, but I am dual booting so I decided to load the device up in Windows with the realsense viewer. Same issue, but it didn't work at all in the windows version.

I cannot try using a different computer as suggested in some other troubleshooting because this is my only one. 

The camera is not overheating. It is cool to the touch in every spot of it.

### Some other odd behaviors

I can get the video to play with no choppy behavior with ffplay /dev/videoX for the rgb module. However, sometimes I must do ffplay/dev/videoY where Y is the default system webcam before I can get the ffplay for the D435 rgb sensor to work. Very odd behavior. 

I suspect the issue might have something to do with loading data from old runs / connections as it just happened to work the first time I ever plugged it in. Not sure how to verify this but its my instinct.
Would appreciate any assistance, @MartyG-RealSense have seen you on many of the other posts about this error so tagging you in hopes of getting it seen.

MartyG-RealSense, 2020-06-09:
Hi @pjmara   Let's start with trying to get the camera working on Windows.

Could you first check if access to the camera hardware and to camera applications is set to On in the Windows privacy settings please, using the instructions in the link below.

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

pjmara, 2020-06-09:
Alright, switched back to windows. Yes it is on.

MartyG-RealSense, 2020-06-09:
When you installed RealSense on Windows, did you install the complete RealSense SDK please?  Doing so places launch icons for the RealSense Viewer and the Depth Quality Tool on your Windows desktop.

The complete SDK can be installed by visiting the link below, scrolling down to the file list headed '**Assets**', and left-clicking on the **Intel.RealSense.SDK-WIN10-2.35.0.1758.exe** link.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.35.0

[image]

pjmara, 2020-06-09:
That was what I had installed, but I had the version listed in the header of this post (2.33.1). Just updated to the most recent you showed above. Same behavior, USB CAM Overflow

MartyG-RealSense, 2020-06-10:
Thanks for your patience.  Given that you are experiencing the same errors in both Windows and Ubuntu, it seems more likely to be a hardware issue than a software one (SDK or OS).

Could you try re-upgrading the camera firmware driver to the latest one, please?  Sometimes a downgrade and then a re-upgrade has corrected errors.  You can do this in the Viewer by going to the 'More' option at the top of the Viewer's options side-panel and selecting 'Update to recommended firmware' from its menu.  You do not need to provide a firmware file.

pjmara, 2020-06-11:
Alright, cable is in, here is what I've learned, I think
Firmware updates do not function properly over USB 2, and this is actually documented to not function at the link in above comment. But the ide still allows for the user to attempt updates over USB 2, and I think it breaks something, I'm still not sure what. 
When I first received the new cable, I tried turning everything on. I was able to get the depth and camera streaming, but when I added the infrared sensors, I got the same overflow error. 
I updated the firmware over usb3, ran the test again, no problems. 
Happy to announce that it is working, but if updating firmware doesn't work over USB 2, the realsense viewer probably shouldn't prompt users to run an update with a USB 2 connection.

MartyG-RealSense, 2020-06-11:
I'm very glad that you succeeded in the end!

The linked article was written in September 2018, and since then Intel introduced a new cross-platform firmware system in which firmware update in the Viewer over USB 2 does work (or is meant to).  I haven't encountered problems with updating through the Viewer on USB2 myself.  I'm sorry that you had trouble with it.