realsenseai/librealsense #6560: export_to_ply sometimes save wrong point cloud

EmmanuelCos, 2020-06-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC |
| Language                            |  C++               | 

Hello,
I am using the rs-multicam with two cameras. At the end of the process, when the user leaves the app, I save the point clouds with export_to_ply. 
Most of the time it works good, but sometimes the point cloud have wrong scale, and color not aligned with depth (without changing anything to the code...). 

What I am doing differently from the standard program is that I do some retry for the poll_for_frames, and then when I have my frameset, I change several parameters (like temporal filter etc...).

Here is what I am doing to save the point clouds :

	//for (auto &&pipe : pipelines){
	for (auto &&pipe : pipelines) {

		///auto frames = pipe.wait_for_frames();
		rs2::frameset frames;
		while (!pipe.poll_for_frames(&frames)) {
			
		}
		//setParameters(frames, pipe.get_active_profile());
		const char* serialNumber = rs2::sensor_from_frame(frames)->get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
		std::string currentName;

		char* dirToSave = argv[1];
		std::stringstream ss;
		ss << dirToSave;
		std::string dirStr = ss.str();
		char* imageName = argv[2];
		std::stringstream ss2;
		ss2 << imageName;
		std::string imNameStr = ss2.str();

		if (strcmp(serialNumber, serialNumbers[0].c_str()) == 0) {
			currentName = dirStr + "/" + imageName + "RR.ply";
		}
		else if (strcmp(serialNumber, serialNumbers[1].c_str()) == 0) {
			currentName = dirStr + "/" + imageName + "RL.ply";
		}
		else {
			currentName = dirStr + "/" + imageName + "ERROR.ply";
		}

		//rs2::align align(RS2_STREAM_DEPTH);//RS2_STREAM_COLOR
		//auto aligned_frames = align.process(frames);
		auto color = frames.get_color_frame();
		rs2::pointcloud pc;
		// Tell pointcloud object to map to this color frame
		pc.map_to(color);
		auto depth = frames.get_depth_frame();
		// Generate the pointcloud and texture mappings
		rs2::points points;
		points = pc.calculate(depth);
		points.export_to_ply(currentName, color);
	}

MartyG-RealSense, 2020-06-10:
Hi @EmmanuelCos   I could not identify where in your program you are changing the temporal filter.  If you are doing it after alignment though then it is recommended that post-processing filters are applied **before** alignment to help avoid distortions such as aliasing (jagged lines).

EmmanuelCos, 2020-06-10:
Hello MartyG.
Yes I am doing it before alignement.
Here is my entire code :

    void setParameters(rs2::frameset frames, rs2::pipeline_profile &profile) {

	rs2::decimation_filter dec_filter;
	rs2::disparity_transform depth_disparity;
	rs2::spatial_filter spat_filter;
	rs2::temporal_filter temporal_filter(0.4, 20.0, 8);
	rs2::hole_filling_filter hole_fill_filter;
	frames = temporal_filter.process(frames);

	auto depth = frames.get_depth_frame();

	auto color = frames.get_color_frame();
	int widthC = color.get_width();
	std::cout << "widthC " << widthC << std::endl;

	rs2::device selected_device = profile.get_device();

	// JSON, short range preset 
	auto advanced_mode_dev = selected_device.as<rs400::advanced_mode>();
	std::string json_content = "{\"param-censususize\": 9,\n\"param-censusvsize\" : 3,\n\"param-disableraucolor\" : 0,\n\"param-disablesadcolor\" : 0,\n\"param-disablesadnormalize\" : 0,\n\"param-disablesloleftcolor\" : 0,\n\"param-disableslorightcolor\" : 0,\n\"param-lambdaad\" : 2000.0,\n\"param-lambdacensus\" : 39,\n\"param-leftrightthreshold\" : 52,\n\"param-maxscorethreshb\" : 2000,\n\"param-medianthreshold\" : 100,\n\"param-minscorethresha\" : 13,\n\"param-neighborthresh\" : 1,\n\"param-raumine\" : 6,\n\"param-rauminn\" : 1,\n\"param-rauminnssum\" : 1,\n\"param-raumins\" : 1,\n\"param-rauminw\" : 4,\n\"param-rauminwesum\" : 8,\n\"param-regioncolorthresholdb\" : 0.02404790738643174,\n\"param-regioncolorthresholdg\" : 0.05924149629687343,\n\"param-regioncolorthresholdr\" : 0.1693732549608611,\n\"param-regionshrinku\" : 4,\n\"param-regionshrinkv\" : 1,\n\"param-regionspatialthresholdu\" : 7,\n\"param-regionspatialthresholdv\" : 3,\n\"param-robbinsmonrodecrement\" : 11,\n\"param-robbinsmonroincrement\" : 1,\n\"param-rsmdiffthreshold\" : 4.707440954582961,\n\"param-rsmrauslodiffthreshold\" : 0.007420896027247603,\n\"param-rsmremovethreshold\" : 0.31771802714889097,\n\"param-scanlineedgetaub\" : 849,\n\"param-scanlineedgetaug\" : 515,\n\"param-scanlineedgetaur\" : 736,\n\"param-scanlinep1\" : 5,\n\"param-scanlinep1onediscon\" : 341,\n\"param-scanlinep1twodiscon\" : 248,\n\"param-scanlinep2\" : 56,\n\"param-scanlinep2onediscon\" : 28,\n\"param-scanlinep2twodiscon\" : 415,\n\"param-secondpeakdelta\" : 26,\n\"param-texturecountthresh\" : 0,\n\"param-texturedifferencethresh\" : 500,\n\"param-usersm\" : 1\n}";
	advanced_mode_dev.load_json(json_content);
	// JSON, short range preset

	auto depth_sensor = selected_device.first<rs2::depth_sensor>();

	depth_sensor.set_option(rs2_option::RS2_OPTION_VISUAL_PRESET, rs2_rs400_visual_preset::RS2_RS400_VISUAL_PRESET_DEFAULT); //RS2_RS400_VISUAL_PRESET_DEFAULT

	depth_sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0); // Enable emitter
																 //CUSTOM MANU
	depth_sensor.set_option(RS2_OPTION_GAIN, 16); // Enable emitter //default 16
	depth_sensor.set_option(RS2_OPTION_EXPOSURE, 5000); // Enable emitter //default 8500
	depth_sensor.set_option(RS2_OPTION_LASER_POWER, 30); // Enable emitter //default 150
	depth_sensor.set_option(RS2_OPTION_DEPTH_UNITS, 0.0003);//default 0.001
	//depth_sensor.set_option(RS2_OPTION_DEPTH_UNITS, 0.001);//default 0.001

	auto color_sensor = selected_device.query_sensors()[1];

	color_sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0); // Enable emitter
	color_sensor.set_option(RS2_OPTION_EXPOSURE, 5000.0);//2500. 10000 = max
	color_sensor.set_option(RS2_OPTION_GAMMA, 300.0);//300 //500
	color_sensor.set_option(RS2_OPTION_GAIN, 60.0);//default is 60//128.0 improved to see brighter

	//rs2::process_interface opti;

	// Configure filter parameters
	dec_filter.set_option(RS2_OPTION_FILTER_MAGNITUDE, 1);//By default 2

	spat_filter.set_option(RS2_OPTION_FILTER_MAGNITUDE, 2);//By default 2
	spat_filter.set_option(RS2_OPTION_FILTER_SMOOTH_ALPHA, 0.50f);
	spat_filter.set_option(RS2_OPTION_FILTER_SMOOTH_DELTA, 20);
	spat_filter.set_option(RS2_OPTION_HOLES_FILL, 2);

	int N = 30;

	rs400::advanced_mode advanced_device(selected_device);
	auto depth_table = advanced_device.get_depth_table();
	//depth_table.depthClampMax = 1100; // 1m30 if depth unit at 0.001// mettre entre 600 et 1500
	depth_table.depthClampMax = 4000; // 1m30 if depth unit at 0.001// mettre entre 600 et 1500
	depth_table.disparityShift = 50;//50

	advanced_device.set_depth_table(depth_table);
	//CUSTOM MANU - change la profondeur maximale - END

	STDepthControlGroup depth_control = advanced_device.get_depth_control();

	depth_control.deepSeaNeighborThreshold = 120;//150 mieux au niveau des bords
[... 146 lines cut ...]
		//rs2::align align(RS2_STREAM_DEPTH);//RS2_STREAM_COLOR
		//auto aligned_frames = align.process(frames);
		auto color = frames.get_color_frame();
		rs2::pointcloud pc;
		// Tell pointcloud object to map to this color frame
		pc.map_to(color);
		auto depth = frames.get_depth_frame();
		// Generate the pointcloud and texture mappings
		rs2::points points;
		points = pc.calculate(depth);
		points.export_to_ply(currentName, color);
	}

    return EXIT_SUCCESS;
    }
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception & e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}

MartyG-RealSense, 2020-06-10:
I read through your script carefully.  I note that you are using **poll_for_frames**, which is appropriate for a multicam setup.  You need to be careful about when to put the CPU to sleep though when using this instruction, so that CPU percent usage is not maxed out.

https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709

EmmanuelCos, 2020-06-10:
Thank you for the answer. Actually I don't put the CPU to sleep. How should I do that?

MartyG-RealSense, 2020-06-10:
A sleep instruction typically takes this form:

**std::this_thread::sleep_for(std::chrono::milliseconds(10));**

Where the number value in brackets is the number of milliseconds (10 in this example) that you want the CPU to sleep for.

EmmanuelCos, 2020-06-10:
Ok but should I systematically use sleep after pole_for_frame? How long should I set the CPU to sleep?

MartyG-RealSense, 2020-06-10:
I found a multicam example from Dorodnic that uses sleep with poll_for_frames, and uses a value of 1 millisecond:

https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412350887

MartyG-RealSense, 2020-06-18:
Hi @EmmanuelCos  Do you still require sssistance with this case please?   Thanks!

EmmanuelCos, 2020-06-18:
Hello MartyG,
Thank you for updating me on this. Actually I still have the problem from time to time, and still can't explain why.

EmmanuelCos, 2020-06-19:
Hello MartyG,

They advice to set depth options before streaming. So I will try this way.

I also noticed that they use enable_stream(), while I only use  enable_device. Do you know what is the difference exactly and what is the best practise?

Thank you,

Emmanuel

MartyG-RealSense, 2020-06-19:
My understanding is that enable.device is used to make queries to the camera hardware, such as retrieving its serial number, whilst enable.stream is used to define the streams if you are defining a custom stream configuration instead of using the camera's default configuration.  For example, to define a custom depth stream in C++:

   config.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);

EmmanuelCos, 2020-06-19:
I am not sure if it's still possible to do it this way nowadays because this line
auto depthSensor = dev.firstrs2::depth_sensor();
doesn't work. firstrs2 is not a member of dev.

MartyG-RealSense, 2020-06-19:
Looking at the example script provided by the RealSense team member, it looks as though they set up **dev** and define it as a **list** before calling it in auto depthSensor = dev.firstrs2::depth_sensor();

[image]

EmmanuelCos, 2020-06-19:
Yes I did that too, but it doesn't work for me because firstrs2 isn't a member of rs2::device.
I am wondering if it's because the case is dated from 2018 and maybe this code isn't up to date anymore?

MartyG-RealSense, 2020-06-19:
The example script I linked to is the only place I can find where the instruction is formatted  as dev.firstrs2::depth_sensor(); ... most of the other examples seems to format it as a variation on this line:

**auto sensor = profile.get_device().first<rs2::depth_sensor>();**

This is similar to the format in your own script, which was:

**auto depth_sensor = selected_device.first<rs2::depth_sensor>();**

The main difference that I see is that instead of using **profile** directly, you use the selected_device variable that is equal to profile.get_device()

Does it make any difference if you take out selected_device and put in the full device definition instead?

**auto depth_sensor = profile.get_device().first<rs2::depth_sensor>();**

I appreciate your patience of my programming knowledge limitations!

EmmanuelCos, 2020-06-20:
Ah I understand it's because the <> were delited when writting it here in github... 
The true line code is:
	auto depthSensor = dev.first\<rs2::depth_sensor>();
Sorry I should have noticed that.
So I manage to change the development, I will keep you in touch on the next days to tell if it's working or not.
Thank you for your help!

MartyG-RealSense, 2020-06-20:
Ah yes, I see.  I didn't notice either that the < > was being deleted from my comment.  Sorry about the confusion!

[image]

MartyG-RealSense, 2020-07-05:
Hi @EmmanuelCos   Do you still require assistance with this case please, or can it be closed?  Thanks!

closed 2020-07-18 by MartyG-RealSense