realsenseai/librealsense #6569: imshow nothing if I want to imshow only one image

Ted-EE, 2020-06-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |  Win10 32-bit  | 
| Kernel Version (Linux Only)    |  N/A                                         | 
| Platform                                 | PC |
| SDK Version                            | 2.0                           | 
| Language                            |  C++                           | 
| Segment			|  others                        | 

### Issue Description

Hi, I would like to imshow just **one** image, but when I delete while() loop, it only imshow nothing.

`
int getDepth() try
{

    rs2::colorizer color_map;
    color_map.set_option(RS2_OPTION_COLOR_SCHEME, 2); // white to black

    rs2::units_transform ut;
    rs2::pipeline pipe;

    rs2::config pipe_config;
    pipe_config.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);

    rs2::pipeline_profile profile = pipe.start(pipe_config);

    while (waitKey(1000)) {

        rs2::frameset data = pipe.wait_for_frames();

        rs2::frame depth_frame = data.get_depth_frame().apply_filter(color_map);

        rs2::depth_frame depth = data.get_depth_frame();
        //rs2::frame color = data.get_color_frame();

        const int depth_w = depth_frame.as<rs2::video_frame>().get_width();
        const int depth_h = depth_frame.as<rs2::video_frame>().get_height();

        Mat depth_image(Size(depth_w, depth_h), CV_8UC3, (void*)depth_frame.get_data(), Mat::AUTO_STEP);

        imshow("depth image", depth_image);
    }
}
return EXIT_SUCCESS;
}

catch (const rs2::error& e) {
    std::cout << "RealSense error calling" << e.get_failed_function() << "(" << e.get_failed_args() << "):\n"
        << e.what() << endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e) {
    std::cout << e.what() << endl;
    return EXIT_FAILURE;
}
`

The screenshot is shown as follows.
[image]

I will appreciate it if you give me some suggestion.

MartyG-RealSense, 2020-06-11:
There was a past case where imshow would not show the image.  They got it working by modifying the while loop in the way shown in the image below.  

[image]

https://github.com/IntelRealSense/librealsense/issues/4362#issuecomment-508551975

ev-mp, 2020-06-11:
@Ted-EE , there are couple of things to check:
1 You operate under the assumption that the depth frame has arrived:  
>rs2::frameset data = pipe.wait_for_frames();  
rs2::frame depth_frame = **data.get_depth_frame()**.apply_filter(color_map);

This actually has to be verified, similar to [6291](https://github.com/IntelRealSense/librealsense/issues/6291#issue-605461906)
```cpp
        frames = pipe.wait_for_frames();
        auto depth_frame = frames.get_depth_frame();
        auto color_frame = frames.get_color_frame();
        // Make sure that both depth and  color are present for calculations
        if (!depth_frame || !color_frame)
            continue;
        .....
```

So you'd still have to retry inloop till the proper frame arrives.
There are ways to avoid the loop, but those require switching to asynchronous polling APIs (callbacks).

2. If after applying (1) you'd still experience the empty frame then please provide a dump of that frame. You can look up the `rs-dave-to-disk` demo for reference

Ted-EE, 2020-06-11:
@MartyG-RealSense Thanks for your help. The problem is solved!

MartyG-RealSense, 2020-06-11:
@Ted-EE   Great to hear - thanks so much for the update!

MartyG-RealSense, 2020-06-18:
Case closed due to solution provided.

closed 2020-06-18 by MartyG-RealSense