realsenseai/librealsense #6571: Notification Callback RS2_NOTIFICATION_CATEGORY_POSE_RELOCALIZATION not working

PLMR-FR, 2020-06-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version |   Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  (4.15.0-99)                                         | 
| Platform                                 | PC |
| SDK Version                            |   2.34.0                           | 
| Language                            |  C++                           | 
| Segment			|  Robot                          | 

### Issue Description
Hello, I am trying to get the info when a relocalization occurs (or a jump in the position). It is necessary when you use a drone and you need to inform the autopilot to not consider this jump in the position.

So I run the ar-advanced example which contain the following part:

```
//Add relocalization callback
    tm_sensor.set_notifications_callback([&](const rs2::notification& n) {
        if (n.get_category() == RS2_NOTIFICATION_CATEGORY_POSE_RELOCALIZATION) {
            std::cout << "Relocalization Event Detected." << std::endl;
            // Get static node if available
            if (tm_sensor.get_static_node(virtual_object_guid, object_pose_in_world.translation, object_pose_in_world.rotation)) {
                std::cout << "Virtual object loaded:  " << object_pose_in_world.translation << std::endl;
                object_pose_in_world_initialized = true;
            }
        }
    });
```

When I hide the camera, shake it for 5 seconds in order to have a drift in the position and then remove my hands from the sensor to get a relocalization, the message for the relocalization never show up.

Is there another way to be informed about the relocalization ?

Thanks

thien94, 2020-06-12:
I am also facing this issue in Python. Specifically, the notification callback does not produce any updates when there are sudden changes in the pose data.
My guess is that the event is actually [pose jumping](https://github.com/IntelRealSense/librealsense/blob/development/doc/t265.md#are-there-any-t265-specific-options), but currently it is not listed in the [available events](https://github.com/IntelRealSense/librealsense/blob/development/include/librealsense2/h/rs_types.h). Hope this gets results soon.

PLMR-FR, 2020-08-09:
Hello,

Is there any plan to implement this kind of feature ?  Like a flag when you have a loop closure with a jump in the position ?

Alamoris, 2020-11-27:
Hello,

As far as I understand POSE_RELOCALIZATION notification is responsible for finding a match between the current position and the previously recorded map. Moreover, this event can be triggered only once.
At the same time, it is currently impossible to receive a notification that the current position has been relocated in the current map?

@MartyG-RealSense @dorodnic @ev-mp

alessandromorra, 2021-02-02:
Dear @RealSense-Customer-Engineering @RealSenseSupport @RealSenseCustomerSupport 
Can you comment on that? I am facing the same issue.
Thank you so much!

closed 2026-10-23 by MartyG-RealSense