realsenseai/librealsense #6573: Build Failing for 2.35.2 release

malapatiravi, 2020-06-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D400 | 
| Firmware Version                   |5.12.3 | 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    | 4.15.0-101-generic                                        | 
| Platform                                 | PC |
| SDK Version                            |2.35.2                          | 
| Language                            |  C++                         | 
| Segment			| Robot                        | 

### Issue Description
I am trying to compile librealsens SDK with the following compilation flags. The compilation breaks. 
`cmake ../ -DBUILD_EXAMPLES=true -DENFORCE_METADATA=false -DBUILD_PYTHON_BINDINGS=true -DFORCE_RSUSB_BACKEND=OFF -DBUILD_WITH_CUDA=true -DIMPORT_DEPTH_CAM_FW=false -DBUILD_WITH_TM2=false -DCMAKE_BUILD_TYPE=Release
`
Please find the screenshot. Thank you
[image]

malapatiravi, 2020-06-11:
@MartyG-RealSense Did anything changed from 2.34 to 2.35. Any kernel or compiler changes do i have to make? I am able to compile 2.34 and earlier without any problem.

MartyG-RealSense, 2020-06-11:
2.35.2 mostly introduces official L515 lidar depth camera support.  The major addition for other cameras is improvements to how multiple cameras are handled.

If you think there may be a kernel conflict, you could install with the backend = true flag if you have an internet connection, as it is not dependent on Linux versions or kernel versions and does not require patching.

To do so, put **-DFORCE_RSUSB_BACKEND=true** at the front of your CMake command in the librealsense build folder, like the example below:

cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
.

malapatiravi, 2020-06-11:
I tried all the above mentioned options. I see the same problem.

MartyG-RealSense, 2020-06-12:
There was a report that Librealsense 2.35.2 was failng to build on Jetson if **-DBUILD_WITH_CUDA=true** was used in the build statement.  Could you try building without the CUDA flag in the build statement please and see if it makes a difference?

malapatiravi, 2020-06-12:
Actually the issue is when DBUILD_WITH_CUDA=true is set. Is there any plan to fix it in near future?

MartyG-RealSense, 2020-06-12:
Re-reading the case, I noticed that it was not confirmed which computing device you are using in this case.  If it is not a Jetson, or another computer that has an Nvidia GPU, then CUDA would not work because CUDA is specific to Nvidia.

For non-Nvidia GPUs, there is an alternative means in the SDK for offloading processing from the CPU onto the GPU.  It is called "GLSL Processing Block", though this method may be ineffective on low-power devices.

https://github.com/IntelRealSense/librealsense/pull/3654

tispratik, 2020-06-12:
Received a response from Realsense support on this topic.

Jun 11, 2020, 5:47:09 PM PDT

Hi Pratik,

The error happens when build with CUDA. 
Set the -DBUILD_WITH_CUDA=false should allow the build to pass, for now... 

I will be looking into this error. Thank you for informing us.

MartyG-RealSense, 2020-06-12:
@tispratik   Thanks!

ev-mp, 2020-06-14:
@tispratik hello,
This is indeed a bug in the latest version.
To fix it navigate  to `src/proc/pointcloud.h` in the source tree and  add
```#include "synthetic-stream.h"```
next to
```#include "../include/librealsense2/hpp/rs_frame.hpp"```

malapatiravi, 2020-06-15:
It worked. Thanks @ev-mp

closed 2020-06-15 by malapatiravi