realsenseai/librealsense #6574: Announcement: Community contribution for using multiple cameras in RealSense Unity wrapper

MartyG-RealSense, 2020-06-12:
Hi everyone,

RealSense community member **JMcC16** has kindly share their method for using multiple cameras in the RealSense Unity wrapper. 

https://support.intelrealsense.com/hc/en-us/community/posts/360047049173

closed 2020-06-15 by MartyG-RealSense

mpottinger, 2020-06-15:
Thank you for posting this. I was surprised when I was unable to find any reviews of this camera yet, I thought for sure some people had gotten their hands on them and I'm really interesting in how it compares to the other cameras.

MartyG-RealSense, 2020-06-15:
The release of SDK 2.35.2 last week, the first version to officially support the L515, coincided with customers beginning to receive the camera, post GitHub questions and create YouTube videos of their day-one use.

There is also a video of an analysis performed with a pre-release L515 unit by a digital display company called lumoPLAY, who reviewed it highly favourably.

https://youtube.com/watch?v=J8W1gLi3Idk