realsenseai/librealsense #6575: T265 SLAM_ERROR Speed leading to unresponsible device

mindThomas, 2020-06-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.0.0-1052-oem-osp1                                         | 
| Platform                                 | PC  |
| SDK Version                            |  v2.35.2                          | 
| Language                            |  C++ & ROS                          | 
| Segment			|  Hand held                          | 

Using ROS Melodic I get the following output when starting the T265 node.
`[ INFO] [1591981920.259104175]: Initializing nodelet with 16 worker threads.
[ INFO] [1591981920.471577114]: RealSense ROS v2.2.13
[ INFO] [1591981920.471612092]: Running with LibRealSense v2.35.2
[ INFO] [1591981920.478615668]:  
[ INFO] [1591981920.486262725]: Device with serial number 943222110748 was found.

[ INFO] [1591981920.486320643]: Device with physical ID 2-1-7 was found.
[ INFO] [1591981920.486347414]: Device with name Intel RealSense T265 was found.
[ INFO] [1591981920.486957809]: Device with port number 2-1 was found.
[ INFO] [1591981920.489201928]: No calib_odom_file. No input odometry accepted.
[ INFO] [1591981920.489235312]: getParameters...
[ INFO] [1591981920.527738468]: setupDevice...
[ INFO] [1591981920.527781454]: JSON file is not provided
[ INFO] [1591981920.527809131]: ROS Node Namespace: camera
[ INFO] [1591981920.527828359]: Device Name: Intel RealSense T265
[ INFO] [1591981920.527844207]: Device Serial No: 943222110748
[ INFO] [1591981920.527868807]: Device physical port: 2-1-7
[ INFO] [1591981920.527889432]: Device FW version: 0.2.0.926
[ INFO] [1591981920.527907176]: Device Product ID: 0x0B37
[ INFO] [1591981920.527922001]: Enable PointCloud: Off
[ INFO] [1591981920.527935434]: Align Depth: Off
[ INFO] [1591981920.527948678]: Sync Mode: Off
[ INFO] [1591981920.527974979]: Device Sensors: 
[ INFO] [1591981920.528027997]: Tracking Module was found.
[ INFO] [1591981920.528080011]: (Depth, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1591981920.528106882]: (Color, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1591981920.528128741]: (Infrared, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1591981920.528142959]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1591981920.528156352]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1591981920.528183123]: num_filters: 0
[ INFO] [1591981920.528198344]: Setting Dynamic reconfig parameters.
[ INFO] [1591981920.540546060]: Done Setting Dynamic reconfig parameters.
[ INFO] [1591981920.541010314]: fisheye1 stream is enabled - width: 848, height: 800, fps: 30, Format: Y8
[ INFO] [1591981920.541311773]: fisheye2 stream is enabled - width: 848, height: 800, fps: 30, Format: Y8
[ INFO] [1591981920.541341333]: setupPublishers...
[ INFO] [1591981920.543211426]: Expected frequency for fisheye1 = 30.00000
[ INFO] [1591981920.569771259]: Expected frequency for fisheye2 = 30.00000
[ INFO] [1591981920.585444268]: setupStreams...
[ INFO] [1591981920.585500626]: insert Fisheye to Tracking Module
[ INFO] [1591981920.585526293]: insert Fisheye to Tracking Module
[ INFO] [1591981920.585549320]: insert Gyro to Tracking Module
[ INFO] [1591981920.585565542]: insert Accel to Tracking Module
[ INFO] [1591981920.585580747]: insert Pose to Tracking Module
[ INFO] [1591981920.619027792]: SELECTED BASE:Pose, 0
[ INFO] [1591981920.620535062]: RealSense Node Is Up!`

### Issue Description
It appears that the sensor can go into some kind of deadlock if the sensor is covered for a while, simulating a dark environment.
What happened was that the reported pose started to accelerate in a given direction after covering the sensor for a while. In most cases where the coverup only lasts for a second or so the internal SLAM algorithm was able to recover. But for the situations where I kept the sensor covered for >10 s it would end up going into some sort of failure mode reporting the following:
` 13/06 01:16:31,414 WARNING [140391512708864] (tm-device.cpp:1289) SLAM_ERROR Speed
[ERROR] [1591982196.599509329]: Ignoring transform for child_frame_id "camera_pose_frame" from authority "unknown_publisher" because of a nan value in the transform (-nan -nan nan) (-nan -nan nan nan)`

The only way to recover from this was to restart the node (i.e. restart the pipeline).

ROSbag can be provided at request.

dookei, 2021-11-17:
Hi @mindThomas , I am also facing the same issue. Once in a while. But causes the tracking to get corrupt and unusable.
I assume no one from intel will never address this. Too bad. It had potential.

learninging1, 2023-05-13:
hey, @mindThomas ,did you solve this problem finally. If you get the solution, could you leave a message to help me with the same problem? Thank you very much.

closed 2026-10-23 by MartyG-RealSense