realsenseai/librealsense #6576: 'odom' and 'map' transforms as two separate poses concurrently

mindThomas, 2020-06-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.0.0-1052-oem-osp1                                         | 
| Platform                                 | PC  |
| SDK Version                            |  v2.35.2                          | 
| Language                            |  C++ & ROS                          | 
| Segment			|  Hand held                          | 

### Issue Description
The T265 seems to be working well when either Pose jumping is enabled or disabled.
When enabled I experienced the expected jumps in the reported pose due to corrections of drift and alignment with the internal map. On the other hand without the Pose jumping enabled I get a temporally smooth trajectory which is good for controls which deals with only short windows of time.

Why do we have to choose? Why can't we get both the smooth odom pose as one pose and get the map corrections as another pose? Concatenating these two transforms would resemble the pose where "Pose jumping" is enabled, while only using the odom pose would resemble the pose where "Pose jumping" is disabled.
Ideally for robotic applications one would construct a tf-tree with a **map** frame at the top followed by an **odom** frame finally connecting to the **base_link** frame of the robot.

Are there any future plans to allow this feature with the T265?

RealSenseSupport, 2021-03-18:
Thank you for highlighting this feature. We have moved our focus to our next generation of products and consequently, we will not be addressing this in the T265.

closed 2021-03-18 by RealSenseSupport

mindThomas, 2021-03-19:
Huh? Is this how you handle questions and issues?
Let them sit for almost a year and then tell the customer that: "Sorry, your product is now discontinued"

What will be the "next generation of products" and when will that be available?
And until then, how would I go about to get the two transforms separately as requested in this ticket?