realsenseai/librealsense #6577: loop break condition for data from recorded file?

vams08, 2020-06-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
< Hello all,
 I am using recorded data and when I run hello-realsense example it is looping infinite times instead I want it to run upto last recorded frame only so can any one suggest how to handle this?>

MartyG-RealSense, 2020-06-13:
Hi @vams08   In Python, you can stop bag playback from looping by using **repeat_playback=false**

https://github.com/IntelRealSense/librealsense/issues/4344#issuecomment-507387892

If you are using C++ then the solution is not so obvious.  A RealSense user did write a script though that included "stop management" code.

https://github.com/IntelRealSense/librealsense/issues/1579#issuecomment-430163908

vams08, 2020-06-13:
Thank you @MartyG-RealSense , just now i got another way to do in c++.
rs2::frameset frames = p.wait_for_frames();
long long int a = frames.get_frame_number();
using these frame number as stamps we can close the loop.

closed 2020-06-13 by vams08

MartyG-RealSense, 2020-06-13:
Great!   Thanks for posting the code for the benefit of C++ users.   :)