realsenseai/librealsense #6579: 2.35.2 UnityPackage example scene errors with L515

xraxra, 2020-06-14:
Camera Model: L515
Firmware Version: 01.04.01.00
OS: Win 10 1909 18363.900 (PC)
SDK: 2.35.2
Language: Unity 2019.3.12f1

Imported the RealSense UnityPackage from latest 2.35.2 release, running the StartHere scene and clicking START on the Point Cloud Processing Blocks (Depth) results in console repeatedly logging the following error below.  I tested the device in RealSense Viewer 2.35.2 and updated firmware in there, all seems to be working outside of Unity, also made sure to close viewer before attempting to run the Unity example scene. 

The <0x01>,<0x02> in the out of range exception show as invalid characters in the unity log ( it just looks like [,] )

ExternalException: rs2_set_option(options:000001BE9D2BEF30, option:Filter Magnitude, value:0)
Rethrow as ArgumentException: Given value 0 is outside [<0x01>,<0x02>] range!
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <b87f0373bec9442cb6bcfff600427347>:0)
Intel.RealSense.OptionInternal.set_Value (System.Single value) (at <b87f0373bec9442cb6bcfff600427347>:0)
RsPointCloud.UpdateOptions (System.Boolean isComposite) (at Assets/RealSenseSDK2.0/Scripts/ProcessingBlocks/RsPointCloud.cs:64)
RsPointCloud.Process (Intel.RealSense.Frame frame, Intel.RealSense.FrameSource frameSource) (at Assets/RealSenseSDK2.0/Scripts/ProcessingBlocks/RsPointCloud.cs:57)
RsProcessingPipe.ProcessFrame (Intel.RealSense.Frame frame, Intel.RealSense.FrameSource src) (at Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs:99)
UnityEngine.Debug:LogException(Exception)
RsProcessingPipe:ProcessFrame(Frame, FrameSource) (at Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs:119)
Intel.RealSense.CustomProcessingBlock:ProcessingBlockCallback(IntPtr, IntPtr, IntPtr)
Intel.RealSense.NativeMethods:rs2_process_frame(IntPtr, IntPtr, Object&)
Intel.RealSense.CustomProcessingBlock:Process(Frame)
RsDevice:RaiseSampleEvent(Frame) (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:148)
RsDevice:WaitForFrames() (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:160)
System.Threading.ThreadHelper:ThreadStart()

RealSenseCustomerSupport, 2020-06-19:
----------------------------------------------

Hi xraxra,
Please try to apply this pull request to your Unity script.
https://github.com/IntelRealSense/librealsense/pull/6644
It should work.

shoegbcn, 2020-06-19:
Hi! I have the same problem. 
I just applied the pull, but I still get the same error. 
Any idea?

shoegbcn, 2020-06-19:
More info - mine is happening with Unity 2019.2.3f1, and the d435.
Log: 
2.0/Scripts/RsProcessingPipe.cs:119)
Intel.RealSense.CustomProcessingBlock:ProcessingBlockCallback(IntPtr, IntPtr, IntPtr)
Intel.RealSense.NativeMethods:rs2_process_frame(IntPtr, IntPtr, Object&)
Intel.RealSense.CustomProcessingBlock:Process(Frame)
RsDevice:RaiseSampleEvent(Frame) (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:148)
RsDevice:WaitForFrames() (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:160)
System.Threading.ThreadHelper:ThreadStart()

wodndb, 2020-06-22:
I solved this problem.

It's not enough to fix the enum "OcclusionRemoval" in RsPointCloud.cs for solving the problem.

You should also change option of the Processing Blocks which is attached to a profile of a RsPeocessingPipe prefab.

https://github.com/IntelRealSense/librealsense/issues/6647

RealSenseCustomerSupport, 2020-06-22:
----------------------------------------------

Yes, you have to update the script and related scriptable objects.

shoegbcn, 2020-06-25:
Hi again, 
changing the RsPointCloud.cs and the other options in the Processing Blocks the point cloud demo still doesn't work. 
I also get the same error + this other error: 

InvalidOperationException: Error while running DisparityTransform processing block, check the log for info
Intel.RealSense.ProcessingBlock.Process[T] (Intel.RealSense.Frame original) (at <cc4cf66ca4ac447089d4777657d39da9>:0)
Intel.RealSense.ProcessingBlock.Process (Intel.RealSense.Frame original) (at <cc4cf66ca4ac447089d4777657d39da9>:0)
RsDisparityTransform.Process (Intel.RealSense.Frame frame, Intel.RealSense.FrameSource frameSource) (at Assets/RealSenseSDK2.0/Scripts/ProcessingBlocks/RsDisparityTransform.cs:56)
RsProcessingPipe.ProcessFrame (Intel.RealSense.Frame frame, Intel.RealSense.FrameSource src) (at Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs:99)
UnityEngine.Debug:LogException(Exception)
RsProcessingPipe:ProcessFrame(Frame, FrameSource) (at Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs:119)
Intel.RealSense.CustomProcessingBlock:ProcessingBlockCallback(IntPtr, IntPtr, IntPtr)
Intel.RealSense.NativeMethods:rs2_process_frame(IntPtr, IntPtr, Object&)
Intel.RealSense.CustomProcessingBlock:Process(Frame)
RsDevice:RaiseSampleEvent(Frame) (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:148)
RsDevice:WaitForFrames() (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:160)
System.Threading.ThreadHelper:ThreadStart()

wodndb, 2020-06-26:
I find what is wrong.

If you start point cloud demo from StartHere scene, the RsPointCloud option of RsProcessingBlock is initialized to null. So error is occurred.

So It's not good to use StartHere scene now. I'm trying to find the solution.

Open scene file "PointCloudDepthandColor" or "PointCloudProcessingBlocks" and try to start the scene.

shoegbcn, 2020-06-26:
Hey @wodndb, thanks for the tip! It works via the separate scene files. 
I rechecked the PointCloudProcessingBlocks, but I can't make it work in the StartHere scene btw. 
Weird thing is I also have a problem with the cursor texture, that works perfectly in the separate scene.
Anyway, I can test things with this working.

RealSenseSupport, 2020-06-28:
Not all of Post Processing filters can work with L515.
In this version, please disable the filters if work with L515.
or try the filters in realsense-viewer first.

wodndb, 2020-06-29:
Maybe L515 can't support disparity map. Because disparity map is generated for stereo IR camera to calculate depth map. L515 has Lidar sensor. This sensor directly get depth value from sensor. Unfortunately I have no L515 camera because it's not for sale in my country. So I can't test. Try to turn off disparity map option.

shoegbcn, 2020-06-29:
@wodndb @RealSenseSupport oh, my camera is a d435

RealSenseSupport, 2020-10-26:
librealsense has been fixed this issue in 2.39.0.
Please try the latest release. Thx.

RealSenseSupport, 2020-11-05:
@xraxra As the issue fixed, close this accordingly. Please feel free to create new one if you have any other questions or issues. Thanks!

closed 2020-11-05 by RealSenseSupport