realsenseai/librealsense #6580: Human Pose estimation with RGBD IntelRealsense D435i

johnny-wang16, 2020-06-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

Hello there, I'd like to ask if you guys can point me to some resources regarding human pose estimation with both the RGB and depth data stream from intel depth camera. I've seen a couple papers that does this with just RGB images but not sure if there's already one with RGBD. Thank you very much.

MartyG-RealSense, 2020-06-14:
The best available commercial solution that supports pose estimation on RealSense specifically is the **Cubemos Skeleton Tracking SDK**.  It is available from the official RealSense store for 75 dollars and also has a free trial version.

https://www.intelrealsense.com/skeleton-tracking/

https://store.intelrealsense.com/buy-skeletal-tracking-sdk.html

Another commercial solution that supports the 400 Series cameras is **Nuitrack SDK**, which recently introduced a 7.99 a month subscription price option.

https://nuitrack.com/

For a free open source solution, you could investigate**OpenPose**.

https://github.com/gregorsamsa183/RealTime3DPoseTracker-OpenPose

Intel also published an advanced level seminar on using RealSense to train the camera to recognise and track joints.

https://youtube.com/watch?v=VSHDyUXSNqY

johnny-wang16, 2020-06-15:
Got it. Thank you very much for the information.

closed 2020-06-15 by johnny-wang16

johnny-wang16, 2020-06-15:
Hello @MartyG-RealSense  , just one last question, I'm wondering if Cubemos Skeleton Tracking SDK works for a single D435i camera.

MartyG-RealSense, 2020-06-15:
Yes, the Cubemos Skeleton Tracking SDK can work with a single camera and is D445i compatible.  

https://github.com/IntelRealSense/librealsense/issues/5995

SammaniSandumini, 2024-04-19:
Hello @MartyG-RealSense I'm working on a project using an Intel RealSense D435 camera to perform 3D pose estimation of a weightlifter. I've recorded the exercises using Intel RealSense SDK and want to extract x, y, z 3D coordinates in real-world coordinates from the recorded bag files. I'm using MediaPipe for pose estimation.

Can you provide any guidance on ,
1.How can I convert the X, Y, Z coordinates into real-world coordinates ?

2.Can I use the timestamps from the recorded bag file to determine the relevant time range for the exercise part?(The timestamp on depth stream and rgb stream seems slightly different)

3.Additionally, I have used two Intel RealSense D435 cameras for front view and side view, each connected to separate laptops for recordings. I've noticed differences in timestamps between the recordings from these cameras. Is it still possible to use timestamps to determine the same exercise from both cameras? If not, are there alternative methods or techniques recommended for synchronizing the data from multiple cameras?

MartyG-RealSense, 2024-04-19:
Hi @SammaniSandumini   Your pose project sounds similar to a MediaPipe one at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/23218415064467-Offline-processing-with-Intel-RS-D455-on-Windows

In regard to your questions:

1.  In that project, they used the SDK instruction `rs2_deproject_pixel_to_point` to convert 2D pixel coordinates to 3D world coordinates.  I believe that this would be the best approach for your project too.  

2.  Depth and RGB frames will be temporally offset, bounded by a period of one frame with the lowest FPS.  If `wait_for_frames()` is used in a script then the SDK tries to find the closest timestamp match between different stream types.

The different types of timestamp available are described at https://github.com/IntelRealSense/librealsense/issues/2188 - **frame_timestamp** would likely be the best type to use if you are able to retrieve it from the recorded bag file.  If you are using Python then https://github.com/IntelRealSense/librealsense/issues/9891 has an example of script code for retrieving frame_timestamp.

3.  You can syncronize two cameras together using **hardware sync**, which involves connecting the cameras together using wiring that attaches to a set of pins under a small removable cap on the top of the camera's casing.  An easy way to get the cap off is to insert a long thin rigid object such as a sewing pin into the small hole in the center of the cap.  A guide to hardware sync can be found at the link below.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration

Alternatively, metadata timestamps may be able to sync the cameras as you suggested.  https://github.com/IntelRealSense/librealsense/issues/2186 discusses this.

SammaniSandumini, 2024-04-20:
Thank you so much for the response @MartyG-RealSense . It was really helpful.

I have a problem about selecting the Frame Timestamp since there are 2 timestamps. I have paln to do is enter the exercise_start_timestamp and exercise_end_timestamp manually. So which timestamp should I use?
[image]

MartyG-RealSense, 2024-04-20:
You are very welcome.  I'm pleased that I could help.  :)

You should be able to use the same **frame_timestamp** timestamp type for both the exercise start and end if you have defined separate 'exercise_start_timestamp' and 'exercise_end_timestamp' variable names.  This is because by having separate variables, you can store different values (start and end) from frame_timestamp.  You should also be able to display the value of start and end by using a  Python **print** instruction on each separate variable name.

SammaniSandumini, 2024-04-20:
Hi @MartyG-RealSense , thanks for the response.
 
In my code, I've defined separate variables for exercise_start_timestamp and exercise_end_timestamp to specify the  time period of the exercise.I'm using these timestamps with the following conditional statement to process frames within the exercise duration:
`if exercise_start_timestamp <= frame_timestamp <= exercise_end_timestamp:
`
Here, frame_timestamp is retrieved from the bag file using:
`frame_timestamp = frames.get_frame_metadata(rs.frame_metadata_value.frame_timestamp) / 1000.0
`
My question is, which timestamp should I use for exercise_start_timestamp and exercise_end_timestamp? Can I use any type?

MartyG-RealSense, 2024-04-20:
You can use any timestamp.  The main difference between them is (a) whether the timestamp is generated by the camera's firmware driver or by the time clock on your computer, and (b) when the timestamp is generated.  For your particular application, I would recommend frame_timestamp if your installation of the librealsense SDK has hardware metadata support enabled, or time_of_arrival if metadata support is not enabled.

Metadata support should be enabled if (a) you have installed librealsense from packages, (b) if you built librealsense from source code with CMake and applied a kernel patch script to the kernel, or (c) built from source code and included the CMake build flag **-DFORCE_RSUSB_BACKEND=TRUE****
  
The most likely scenario where metadata support is not enabled and frame_timestamp would be unavailable would be if librealsense was built from source code, and you did not apply a kernel patch script or use the RSUSB = true build flag.

SammaniSandumini, 2024-04-21:
Thank you so much for the reply @MartyG-RealSense