realsenseai/librealsense #6584: Devices under /dev/video* do not seem to provide RGB formats

sp4rc, 2020-06-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Ubuntu 16.04.6 LTS  | 
| Kernel Version (Linux Only)    |  4.15.0-106-generic                                         | 
| Platform                                 | PC  |

### Issue Description

Somehow it seems that the provided video devices under `/dev/video*` do not provide an RGB stream. This means I can't e.g. use `Cheese` or `Zoom Conferencing` with this camera. However the `realsense-viewer` util shows all the streams correctly. Maybe there is something wrong with my udev rules? 

I have followed [this](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) link to install the kernel module and utils. The devices correctly show up:
```
$ v4l2-ctl --list-devices --verbose
Intel(R) RealSense(TM) Depth Ca (usb-0000:00:14.0-4):
        /dev/video0
        /dev/video1
        /dev/video2

$ modinfo uvcvideo | grep "version:"
version:        1.1.2.realsense-1.3.11
srcversion:     DDEE98EBF0181C7A35FC635
```
However when looking at the provided formats it seems there is no RGB stream supported... What am I doing wrong here?

```
$ v4l2-ctl --list-formats --device=/dev/video0
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'Z16 '
        Name        : 16-bit Depth

        Index       : 1
        Type        : Video Capture
        Pixel Format: 'Z16 '
        Name        : 16-bit Depth

$ v4l2-ctl --list-formats --device=/dev/video1
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'GREY'
        Name        : 8-bit Greyscale

        Index       : 1
        Type        : Video Capture
        Pixel Format: 'UYVY'
        Name        : UYVY 4:2:2

        Index       : 2
        Type        : Video Capture
        Pixel Format: 'GREY'
        Name        : 8-bit Greyscale

        Index       : 3
        Type        : Video Capture
        Pixel Format: 'Y8I '
        Name        : Interleaved 8-bit Greyscale

        Index       : 4
        Type        : Video Capture
        Pixel Format: 'Y12I'
        Name        : Interleaved 12-bit Greyscale

        Index       : 5
        Type        : Video Capture
        Pixel Format: 'W10 '
        Name        : 10-bit packed 8888[2222]

$ v4l2-ctl --list-formats --device=/dev/video2
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'YUYV'
        Name        : YUYV 4:2:2

        Index       : 1
        Type        : Video Capture
        Pixel Format: 'BYR2'
        Name        : 16-bit Bayer BGBG/GRGR
```

ev-mp, 2020-06-14:
@sp4rc , the listing is correct -  D415 enumerates as three UVC devices :
1 /dev/video0 - Depth sensor (Z16)
2 /dev/video1 - IR sensor (Y8/Y16)
3 /dev/video2 - RGB sensor (YUYV/BYR2)

To use the camera with `cheese` you need to tell the app to use the proper device.
Here are two snapshots from `cheese` - the left half uses RGB, the right-IR:
[image]

MartyG-RealSense, 2020-06-14:
@ev-mp   Thanks very much for your answer, I didn't have the setup to test this myself in Ubuntu right now.   I thought it might be as you illustrated though, since my experiences of accessing the camera via RealSense driver selection in video-equipped applications like Skype and Discord in Windows are similar.  Thanks again!

https://github.com/IntelRealSense/librealsense/issues/6209#issuecomment-615091643

sp4rc, 2020-06-14:
> @sp4rc , the listing is correct - D415 enumerates as three UVC devices :
> 1 /dev/video0 - Depth sensor (Z16)
> 2 /dev/video1 - IR sensor (Y8/Y16)
> 3 /dev/video2 - RGB sensor (YUYV/BYR2)
>
> To use the [...]

Thanks for replying. Unfortunately I can't get any output. When using `cheese` it simply crashes (Speicherfehler), even when supplying the device via `-d /dev/video2`... Also no luck using `camorama`. What else can I try?

ev-mp, 2020-06-14:
@sp4rc , according to the log `cheese` crashed, so I went to Ubuntu 16 to check and experienced the same issue with application crashes on start.
I kind of found a work-around - start `cheese` and then connect the camera to the PC. This way the application doesn't crash on start, but it then also doesn't generate live stream even though the three D415 sensors are listed in the app's `preferences` menu.

Therefore, since `cheese` runs properly in Ubuntu 18 the issue seem to be isolated to app-level, and that it was already rectified in bionic release. You can try to upgrade the app for xenial, but I don't know if that is feasible.

sp4rc, 2020-06-15:
> Therefore, since `cheese` runs properly in Ubuntu 18 the issue seem to be isolated to app-level, and that it was already rectified in bionic release. You can try to upgrade the app for xenial, but I don't know if that is feasible.

Thanks again. I will consider upgrading to the latest LTS version of Ubuntu.

closed 2020-06-20 by ev-mp