realsenseai/librealsense #6586: L515 Inter Cam Sync Broken

jordanharveyvfx, 2020-06-14:
| | |
|---|---|
|**librealsense**|2.35.2 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|00000000f0172144|
|**Firmware Version**|01.04.01.00|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|00000003a8a6256d|
|**Firmware Update Id**|00000003a8a6256d|
|**Name**|Intel RealSense L515|
|**Serial Number**|00000000f0171962|
|**Firmware Version**|01.04.01.00|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|00000003a95811e6|
|**Firmware Update Id**|00000003a95811e6|

It appears as though the inter cam sync mode is broken in the viewer. I've connected two L515's and check inter cam sync mode active on both before turning on the depth streaming. Both cameras report no streaming data. when the depth streaming is turned on.

When activating depth streaming with no inter cam sync, both cameras display their individual depth and RGB channels appropriately. Once you turn on inter cam sync, only one camera will display it's depth data in the 3D viewer. 

No cameras synchronize data in any scenario I've tried.

MartyG-RealSense, 2020-07-04:
Hi @jordanharveyvfx   I apologise for the delay in responding.  Are you still experiencing this problem with L515 hardware sync please, or can the case be closed?   Thanks!

dorodnic, 2020-08-05:
https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup

RealSenseSupport, 2020-08-14:
Hi @jordanharveyvfx 

If there is nothing else needed on this thread, we can go ahead and close it.  Let us know.

*If we don’t hear from you in 7 days, this issue will be closed.