realsenseai/librealsense #6588: [No device connected] L515 LiDAR Camera is not activate in Python

oronaminc, 2020-06-15:
I used " align-depth2color.py code " as the original basis for Python application development of D415, D435

(https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py)

However, in this case(L515 LiDAR), it is not executed.
Error Message is "RuntimeError: No device connected"

How should I solve this problem?
(In intel realsense viewer v2.35.2, L515 LiDAR camera is now well working)

ev-mp, 2020-06-15:
@oronaminc hello, you need to ensure that the python wrapper that you use is based on the latest v2.35.2 release. 
You haven't specified the OS/Distribution that you're using so I'm short of recommending the correction path - please verified the installation that you use and update.

oronaminc, 2020-06-17:
> @oronaminc hello, you need to ensure that the python wrapper that you use is based on the latest v2.35.2 release.
> You haven't specified the OS/Distribution that you're using so I'm short of recommending the correction path - please verified the installation that you use and update.

Thank you for solving this problem.
As I update firmware, it works well :>

RealSenseSupport, 2020-06-18:
@oronaminc is there anything else you need on this thread?

oronaminc, 2020-06-19:
> @oronaminc is there anything else you need on this thread?

@RealSenseSupport Everything is so good! Thank you for your interest.

closed 2020-06-19 by RealSenseSupport

lightov73, 2021-01-12:
I have the same problem an dthe solution provided above doesnt work for me/
I'm working with phyton 3.7, pyrealsense2    version   2.41.0.2666.
When running the above example I get RuntimeError: Couldn't resolve requests on the " pipeline.start(config)" line .
What is wrong?

markliou, 2021-01-28:
> I have the same problem an dthe solution provided above doesnt work for me/
> I'm working with phyton 3.7, pyrealsense2 version 2.41.0.2666.
> When running the above example I get RuntimeError: Couldn't resolve requests on the " pipeline.start(config)" line .
> What is wrong?

You can also find similar issue at:
https://github.com/IntelRealSense/librealsense/issues/6684
Hope this information is helpful.