realsenseai/librealsense #6591: CMake linking error

noamzaks, 2020-06-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.951 | 
| Operating System & Version |  Windows 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.35.2 |
| Language                            | C++ | 
| Segment			|  Robot | 

### Issue Description
My librealsense is installed at C:\rs (it is from the Intel.RealSense.SDK-WIN10-2.35.2.1897.exe installer, simply moved the folder from the Program Files x86)
I am using MinGW g++ and CMake with CLion, and my project is stored at C:\T265.

When trying to build the following
```cpp
#include "../rs/include/librealsense2/rs.hpp"
int main(int argc, char * argv[])
{
    rs2::pipeline p;
    return 0;
}
```
with the CMakeLists:
```
cmake_minimum_required(VERSION 3.16)
project(RealSense_T265)
set(CMAKE_CXX_STANDARD 14)
add_executable(RealSense_T265 main.cpp)
```

I get the following error (when running):
```
====================[ Build | T265 | Debug ]====================================
"C:\Program Files\JetBrains\CLion 2020.1.2\bin\cmake\win\bin\cmake.exe" --build C:\T265\cmake-build-debug --target T265 -- -j 4
Scanning dependencies of target T265
[ 50%] Building CXX object CMakeFiles/T265.dir/main.cpp.obj
[100%] Linking CXX executable T265.exe
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: CMakeFiles\T265.dir/objects.a(main.cpp.obj): in function `ZN3rs25errorC2EP9rs2_error':
c:/rs/include/librealsense2/hpp/rs_types.hpp:90: undefined reference to `rs2_get_error_message'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: c:/rs/include/librealsense2/hpp/rs_types.hpp:92: undefined reference to `rs2_get_failed_function'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: c:/rs/include/librealsense2/hpp/rs_types.hpp:92: undefined reference to `rs2_get_failed_function'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: c:/rs/include/librealsense2/hpp/rs_types.hpp:93: undefined reference to `rs2_get_failed_args'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: c:/rs/include/librealsense2/hpp/rs_types.hpp:93: undefined reference to `rs2_get_failed_args'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: c:/rs/include/librealsense2/hpp/rs_types.hpp:94: undefined reference to `rs2_get_librealsense_exception_type'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: c:/rs/include/librealsense2/hpp/rs_types.hpp:95: undefined reference to `rs2_free_error'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: CMakeFiles\T265.dir/objects.a(main.cpp.obj): in function `ZN3rs25errorC1EP9rs2_error':
c:/rs/include/librealsense2/hpp/rs_types.hpp:90: undefined reference to `rs2_get_error_message'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: c:/rs/include/librealsense2/hpp/rs_types.hpp:92: undefined reference to `rs2_get_failed_function'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: c:/rs/include/librealsense2/hpp/rs_types.hpp:92: undefined reference to `rs2_get_failed_function'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: c:/rs/include/librealsense2/hpp/rs_types.hpp:93: undefined reference to `rs2_get_failed_args'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: c:/rs/include/librealsense2/hpp/rs_types.hpp:93: undefined reference to `rs2_get_failed_args'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: c:/rs/include/librealsense2/hpp/rs_types.hpp:94: undefined reference to `rs2_get_librealsense_exception_type'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: c:/rs/include/librealsense2/hpp/rs_types.hpp:95: undefined reference to `rs2_free_error'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: CMakeFiles\T265.dir/objects.a(main.cpp.obj): in function `ZN3rs25error6handleEP9rs2_error':
c:/rs/include/librealsense2/hpp/rs_types.hpp:141: undefined reference to `rs2_get_librealsense_exception_type'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: CMakeFiles\T265.dir/objects.a(main.cpp.obj): in function `ZN3rs27contextC1Ev':
c:/rs/include/librealsense2/hpp/rs_context.hpp:102: undefined reference to `rs2_create_context'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: c:/rs/include/librealsense2/hpp/rs_context.hpp:102: undefined reference to `rs2_delete_context'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: CMakeFiles\T265.dir/objects.a(main.cpp.obj): in function `ZN3rs28pipelineC1ENS_7contextE':
c:/rs/include/librealsense2/hpp/rs_pipeline.hpp:375: undefined reference to `rs2_create_pipeline'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: c:/rs/include/librealsense2/hpp/rs_pipeline.hpp:375: undefined reference to `rs2_delete_pipeline'
collect2.exe: error: ld returned 1 exit status
mingw32-make.exe[3]: *** [T265.exe] Error 1
mingw32-make.exe[2]: *** [CMakeFiles/T265.dir/all] Error 2
mingw32-make.exe[1]: *** [CMakeFiles/T265.dir/rule] Error 2
mingw32-make.exe: *** [T265] Error 2
CMakeFiles\T265.dir\build.make:84: recipe for target 'T265.exe' failed
CMakeFiles\Makefile2:74: recipe for target 'CMakeFiles/T265.dir/all' failed
CMakeFiles\Makefile2:81: recipe for target 'CMakeFiles/T265.dir/rule' failed
Makefile:117: recipe for target 'T265' failed
```

Any suggestions?

ev-mp, 2020-06-15:
@noamzaks , apparently the CMake lacks the references to librealsense:
>cmake_minimum_required(VERSION 3.16)
project(RealSense_T265)
set(CMAKE_CXX_STANDARD 14)
**find_package(realsense2)**
**target_link_libraries(${PROJECT_NAME} PRIVATE realsense2::realsense2)**
add_executable(RealSense_T265 main.cpp)

#4277, #4388.

noamzaks, 2020-06-16:
@ev-mp thank you very much!

closed 2020-06-16 by noamzaks