realsenseai/librealsense #6596: Ubutnu 18.04, RealSense V2.33.1, Pi4, T265 not recognizing in bootup

awanthas, 2020-06-15:
I am doing a T265 project with Pi4 and Ubuntu 18.04, which I need to use ROS (hence I am using libRealSense V2.33.1). The T265 is connected to a USB 3 port and the problem is every time I need to hot unplug and plug it back to make it recognized in realsense-viewer (or any other application). It does not recognize at boot time.
PS. I have tried with ubuntu 20.04 (which does not have ROS support yet) in Pi 4 and ubuntu 16.04 (in a Pi 3-B) and they both recognize the plugged T265 in boot time without any issue.
Any help is greatly appreciated. Thanks

m4xr1sk, 2020-06-15:
the first of a long set of problems
I don't know how they can sell a product like this
literally 200 bucks throw in the trash
I've got the same issue, solved by using an amazon basic hub
there's no way to make it works directly connected to the rpi4
then after a 30 minutes run of rs-pose or everything else it start to say "Unrecoverable! error native function rs2_pipeline_wait_for_frames: Frame didn't arrive within 5000"
good luck ...

csr-kick, 2020-06-17:
The workaround for this is to powercycle the USB chip when the rpi 4 boots up.

install uhubctl then I put in /etc/rc.local to power cycle usb hub:
uhubctl -a cycle -l 1 -p 1-4
uhubctl -a cycle -l 2 -p 1-4

But yea, this device is incredibly poorly supported. We can't even get anyone from intel to comment on these issues never mind a fix.

Ddonsy, 2020-06-17:
Thanks a lot guys will try that.

RealSenseSupport, 2021-03-18:
Thank you for highlighting this issue. We have moved our focus to our next generation of products and consequently, we will not be addressing this in the T265. There is a known FW doesn't load at times due to boot up USB power sequencing. It is worth reviewing the specification update as outlined on the following page: https://dev.intelrealsense.com/docs/tracking-camera-t265

closed 2021-03-18 by RealSenseSupport