realsenseai/librealsense #6597: Segmentation Fault: 11 - IMU sensor on MacOs Mojave

orla11, 2020-06-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   MacOS Mojave  | 
| Kernel Version (Linux Only)    |  10.14.6                                         | 
| Platform                                 | Mac  |
| SDK Version                            |  latest                         | 
| Language                            |  C++                           | 
| Segment			|  others                           | 

### Issue Description
I am having issues running librealsense examples. Most of them exit with a `segmentation fault: 11`

Digging deeper I came up with the hypothesis that the issue might be related to the IMU sensor. When running `./realsense-viewer` the viewer start but i get this errors in console:

```
 15/06 17:29:51,843 WARNING [0x700002a6f000] (types.cpp:49) Accel Sensitivity:hwmon command 0x4f failed.
Error type: No data to return (-21).
 15/06 17:29:51,846 WARNING [0x700002a6f000] (ds5-motion.cpp:463) IMU Calibration is not available, see the previous message
```

Furthermore, if I try to turn the motion module on, the program crash and exit with `segmentation fault: 11`

I tried to calibrate the imu sensor as an alternative way but when I run `python rs-imu-calibration.py` that is the outcome:

```
waiting for realsense device...
  Device PID:  0B3A
  Device name:  Intel RealSense D435I
  Serial number:  947522071113
  Product Line:  D400
  Firmware version:  05.12.05.00
Start interactive mode:
Segmentation fault: 11
```
I also noticed that sometimes after opening and closing realsense-viewer, when I run some examples the segmentation fault error does not happen anymore but the program get stuck without any further message like (considering the previous example):

```
waiting for realsense device...
  Device PID:  0B3A
  Device name:  Intel RealSense D435I
  Serial number:  947522071113
  Product Line:  D400
  Firmware version:  05.12.05.00
Start interactive mode:
# Here the execution is stuck like the realsense camera is not responding
```

I tried everything but I cannot get my head around this problem. Can someone help? Is it a compatibility problem?

MartyG-RealSense, 2020-06-16:
Hi @orla11  The D435i is not currently supported on MacOS, unfortunately.  I do not have a time estimate available for when it will be supported on Mac.

[image]

orla11, 2020-06-16:
Ok, then I guess I'll temporarily switch to Windows 10. Thanks.

MartyG-RealSense, 2020-06-25:
Closed due to no further comments received.

closed 2020-06-25 by MartyG-RealSense

MartyG-RealSense, 2021-02-15:
Hi again.  The new librealsense version **2.42.0** has introduced IMU support for MacOS.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.42.0