realsenseai/librealsense #6598: RPI4, rs-pose and "Frame didn't arrive within 5000"

m4xr1sk, 2020-06-15:
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.951 | 
| Operating System & Version |  Raspbian GNU/Linux 10 (buster) | 
| Kernel Version (Linux Only)    |  Linux raspberrypi 4.19.97-v7l+ | 
| Platform                                 | Raspberry Pi 4 |
| SDK Version                           |  v2.35.2 | 
| Language                               |  C | 
| Segment			         |  Desktop test | 

Running rs-pose, after some minutes it stops saying:
Unrecoverable! error native function rs2_pipeline_wait_for_frames: Frame didn't arrive within 5000

A similar issue appears using ROS, in this case it stops publishing the pose data and the node stays up without any error

csr-kick, 2020-06-15:
Yea a few of us have run into this problem, and it severely limits the usefulness of the sensor to have it stop functioning every few seconds. We have been looking for a solution or workaround, but don't have any good ones yet. Unfortunately, judging by the number open issues here, Intel seems to no longer be supporting this device.

m4xr1sk, 2020-06-17:
crazy and unbelievable from a company like Intel. 
started some test in september but stopped after few weeks because to many bugs. 
now, after about 10 months, I have the same problems and the camera works only for few minutes.
ok, then what? 
I ask to Intel if I have to use this camera to stop my papers in the desk.
Also to the Intel team, next time I could pay 20 bucks more but I really would like to have a working product, not something under "virtual" "never finishing" development

yputra, 2020-06-18:
yeah, I was working with the same setup like the end of last years, spent weeks trying to find the possible solution, as well as waiting for the new firmware, hoping they could somehow fix it, but they never did, so I ended up abandoning that project

csr-kick, 2020-06-18:
What is really shocking is how responsive @MartyG-RealSense is to every other issue while ignoring anything to do with the T265. Concerning for this whole line of products that Intel is not invested in their success.

yputra, 2020-06-18:
in the beginning it was quite responsive on T265 though, and it works with NUC and some other platform, but I notice it has problems on Raspberry, Mac, and some other uncommon platform. And maybe they are working on some other product (maybe the new tracker which is working well?), and they dont bother to keep working on the old one with the uncommon platform

m4xr1sk, 2020-06-18:
Probabibly the raspberry is not the right device for this camera. I can't put a nuc on my system cause to big and expensive. If they can suggest us the right compact device to work with this camera then it would be enought for me

closed 2026-10-23 by MartyG-RealSense