realsenseai/librealsense #6599: RealSense Viewer crashes at start

robertxwu, 2020-06-15:
| Required Info                          |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 and D435 | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  NA                                         | 
| Platform                                 | PC |
| SDK Version                            |  v2.35.2                       | 
| Language                                |  NA                          | 
| Segment			          |  NA                      | 

RealSense Viewer kept crashing at start, while the Depth Quality Tool worked fine.

The Windows Event says:

Windows cannot access the file  for one of the following reasons: there is a problem with the network connection, the disk that the file is stored on, or the storage drivers installed on this computer; or the disk is missing. Windows closed the program Intel(R) RealSense(TM) Viewer because of this error.

Program: Intel(R) RealSense(TM) Viewer
File: 

The error value is listed in the Additional Data section.
User Action
1. Open the file again. This situation might be a temporary problem that corrects itself when the program runs again.
2. If the file still cannot be accessed and
	- It is on the network, your network administrator should verify that there is not a problem with the network and that the server can be contacted.
	- It is on a removable disk, for example, a floppy disk or CD-ROM, verify that the disk is fully inserted into the computer.
3. Check and repair the file system by running CHKDSK. To run CHKDSK, click Start, click Run, type CMD, and then click OK. At the command prompt, type CHKDSK /F, and then press ENTER.
4. If the problem persists, restore the file from a backup copy.
5. Determine whether other files on the same disk can be opened. If not, the disk might be damaged. If it is a hard disk, contact your administrator or computer hardware vendor for further assistance.

Additional Data
Error value: 00000000
Disk type: 0

All your assistance would be greatly appreciated

Thanks

robertxwu, 2020-06-15:
FYI: the Viewer, directly built from source code, worked fine.

MartyG-RealSense, 2020-06-16:
I have not seen other reports of this problem occurring yet, so it may be a one-off glitch.

If you would like the convenience of launching your source-built Viewer executable from the Windows desktop like normal, I recommend going to the .exe file, right-clicking on it and selecting the 'Create shortcut' menu option, and clicking Yes when asked if you would like to place the shortcut on the desktop.

[image]

[image]

Then you can rename the shortcut to 'RealSense SDK Viewer'.

[image]

robertxwu, 2020-06-18:
@MartyG-RealSense thank you for the suggestions. I will take the workaround for now.

MartyG-RealSense, 2020-06-18:
Okay, thank you very much.   I will close this case.  Please feel free to create a new issue whenever you have another question.

closed 2020-06-18 by MartyG-RealSense

wldevries, 2020-09-15:
I have the exact same problem with SDK 2.0 version 2.36.0.2034. It will crash whether my D435 camera is connected or not.

Replacing the shortcut or launching the viewer exe directly does not help at all.

MartyG-RealSense, 2020-09-15:
Hi @wldevries   If downloading the separate standalone version of the Viewer (which can correct the incompatibility with the Viewer's OpenVINO function on SDK 2.36.0 onward on old computers) then there are a couple of other actions that you can try:

-  The Viewer and the Depth Quality Tool are dependent on OpenGL graphics.  Make sure that your OpenGL graphics drivers are up to date by updating the graphics drivers on your computer.

-  If you are using Windows, open your user folder (e.g **c: > Users > your username**) and see whether there is a folder called **Documents** with a capital D there.  If there is not then create a folder in that location with the name **Documents** (again, with a capital D)

[image]

wldevries, 2020-09-15:
Thanks for the advice. The directory `Documents` already existed in my user directory, although seriously it should not depend on US hardcoded names, but use Windows API to get the correct path. My GeForce drivers are also completely up-to-date so that's not the problem either.

I'm now downloading the latest SDK release from Github instead of the older one I downloaded from the Intel website. 🤞

wldevries, 2020-09-15:
I now get the following error message

> Program terminated due to an unrecoverable SEH exception:
Illegal CPU instruction!
Possibly newer CPU architecture is required

Note that I have been using earlier versions of the viewer just fine on this same PC.

MartyG-RealSense, 2020-09-15:
This is related to an issue that has been present since SDK 2.36.0 when OpenVINO face detection was introduced into the Viewer.  Since then, computers that have a processor chip older than the Intel Core 6th Generation (Skylake) from around 2016 can experience the error that you received.  The solution can be to download the standalone .exe version of the Viewer from the SDK **Releases** page, as this does not contain the OpenVINO functions.

https://github.com/IntelRealSense/librealsense/issues/7080#issuecomment-671807292