realsenseai/librealsense #6601: Realsense-Viewer can't detect L515

nawara72, 2020-06-16:
Hi

I just received my L515 today.  The realsense-viewer can't detect it, neither can rs-sensor-control. I am running on windows and my device manager shows the L515 under cameras as both Depth and RGB camera.

Is their a separate sdk for the L515, I assumed I just use the librealsense. What are quick tests/tools I can use quickly on L515?

Thanks

RealSenseCustomerSupport, 2020-06-16:
----------------------------------------------

Hi nawara72,
did you use the latest viewer which is  version 2.35.2?
only newer version of viewer 2.35.x support the L515

nawara72, 2020-06-16:
I was using 2.35.0, great i will try the new version 2.35.2, thanks for all your help

RealSenseSupport, 2020-06-18:
Hi nawara72,
Could realsense viewer v 2.35.2 help to fix your  issue?
thanks

nawara72, 2020-06-18:
Yes just compiled the new version now and worked immediately, and can detect and view the L515 using the viewer. Brilliant  and Thanks

closed 2020-06-18 by RealSenseCustomerSupport