realsenseai/librealsense #6604: Alignment performance of L515 is worse D400 series.

dnjfk2003, 2020-06-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515| 
| Firmware Version                   | 1.4.1| 
| Operating System & Version |  Ubuntu 18.04| 
| Kernel Version (Linux Only)    | | 
| Platform                                 | PC |
| SDK Version                            |  2.35.2 | 
| Language                            |  python | 
| Segment			|  others  | 

I've got a L515 today and tested with an alignment code.(align depth_stream to color_stream)
But I could know that alignment performance is worse than other realsense camera such as D415, D435.

For instance, when I get distance between the camera and a small piece object, I first check where the small object is located in color_array as pixel index.
With that pixel index, I get depth value in depth_array(aligned), but the value is not for the object but for other object near my object.

When I used D400 series camera, they aligned color and depth almost same.
Does IntelRealSense already know this problem, if yes do you have any plan to improve the align performance of L515 later?

thank you.

dnjfk2003, 2020-06-17:
For further expanation...
let me compare D415 with L515
next two images are taken with D415(color and depth)
[image]
[image]

when I get a part of chair index, color and depth, both of them have similar x,y value

but in case of L515, color and depth have different x,y value(5~10 pixels..)
[image]
[image]

can I calibrate some parameters for alignment?

RealSenseSupport, 2020-06-19:
@dnjfk2003 - Thanks for the notification of this issue.  We will have a look.  Are you using RSViewer to get the images you provided?  Or are you using some other application?  What setting are you using for the L515, resolutions for both depth and RGB, etc?  Are you using a preset (Max Range, Low Ambient, No Ambient, Short Range)?

dnjfk2003, 2020-06-20:
@RealSenseSupport I used python with my code which is modified a little bit from align example code.
I set color resolution is 1280*720 and origin depth's is 1024*768 and depth frames are aligned to color frames.
To see depth image like rsViewer, I used pyplot.imshow with parameter vmin, vmax.

L515 align two frames well in the center of frames but outside is bad.
Hope realsense team will solve this problem soon.
Thank you.

cdb0y511, 2020-06-26:
@dnjfk2003 Hi, I wonder if it gets better when you use the same resolution, like 1024 for both.
I have D435, but I always use the same resolution when depth align to RGB. Evan with the same resolution it still have several pixels misaligned due to the different FOV. It gets worse on the border .
Does the misaligned bothers you? Like,it  increases reprojected error when you work on slam or something else. If you have any idea pls let me know.

RealSenseSupport, 2020-06-26:
Hi @dnjfk2003 

Thanks for the information on your setup and configuration.  Can you also provide images of the IR stream with pixel location information like you have for the depth and RGB images?

dnjfk2003, 2020-06-26:
> @dnjfk2003 Hi, I wonder if it gets better when you use the same resolution, like 1024 for both.
> I have D435, but I always use the same resolution when depth align to RGB. Evan with the same [...]
Thank you for the advice. but there is no option of 1024 * 768 for color streaming. L515 has only two option for color, one is HD, the other is full HD. It's very important for my application to have well aligned frames. If I know how I can make it better, I will let you know.

dnjfk2003, 2020-06-26:
@RealSenseSupport 
could you please tell me how to get IR stream image(or array) from python code?
I can't use L515 camera currently, but I will check later(5~6 hours later from now).

can I get ir stream with code like
"aligned_ir_frame = aligned_frames.get_ir_frame()" ?

When I got color&depth frame, i used next code.
-------------------------------------------
frames = pipeline.wait_for_frames()
aligned_frames = align.process(frames)
aligned_depth_frame = aligned_frames.get_depth_frame()
color_frame = aligned_frames.get_color_frame()
------------------------------------------

RealSenseSupport, 2020-07-02:
Hi @dnjfk2003 

It would be easier debug if you were able to move to using the RealSense Viewer for debugging.  That way it takes Python out and the potential for not being able to view all streams.

dnjfk2003, 2020-07-03:
@RealSenseSupport 
I can get color, depth and ir frames using realsense-viewer how can I get those frames with aligning them?
Is there any option in rsViewer?

RealSenseSupport, 2020-07-07:
There are quite a bit of advanced settings and things to explore within the RealSense Viewer.  You can have a look at the 3D view once you start streaming of IR, RGB and Depth, it's in the right hand corner of the viewer screen

[image]

Then you can texture map over the depth the RGB, IR or Depth

dnjfk2003, 2020-07-08:
@RealSenseSupport I'm not sure I explained here what I want to know.

What I'm doing is that is a distance information between the camera and an object. For that the first thing I do is, find the object location in RGB image. Then I get the distance from DEPTH array.
Of course, I should know the pixel address in the depth array and I get this address from RGB array.

For that, I'm using "pyrealsense2.align(pyrealsense2.stream.color)" function. But D400 series cameras give good align result but L515 doesn't.(You can see the result above images I uploaded before)

Anyway.. I took a capture from realsense viewer and I just used 2D for explaining my situation better.
here it is
[image]
Let's say I want to find the distance of the points (242, 604) in color array.
To get the distance,of course, I shouldn't use depth value of (242, 604) in depth array, but (149, 350).
So that's why I'm using alignment code and hope L515 alignment performance become better..

Hope I explained well for you to understand what I'm saying.

cdb0y511, 2020-07-10:
> explained
Hi, you can use : rs-enumerate-devices -c to list all Intrinsic and Extrinsic.
The SDK align filter just use these to force align one to another image. You can write your own align filter if you like.
In my opinion , if you want use the default align filter.  You should choose what you need most, align to depth provide accurate depth estimation to infrared camera, but the color always few pixels misaligned. And vice versa .

RealSenseSupport, 2020-07-10:
Hi @dnjfk2003 
Thanks for the explanation.  Now I do understand what you're looking for.  My apologies.  @cdb0y511 has a good outline in the comment above on using the align capability of the SDK.  Couple of links to the align examples:  https://github.com/IntelRealSense/librealsense/tree/v2.36.0/examples/align and https://github.com/IntelRealSense/librealsense/tree/v2.36.0/examples/align-advanced

The alignment isn't going to be exact, as stated it will be a few pixels difference.

dnjfk2003, 2020-07-11:
@cbd0y511 thank you for letting me know the command. I didn't know that I can check camera parameters from that. I have another question how I can use the matrix? I can see rotation matrix with size 3X3.
Is it something like
DEPTH_PIXEL = MATRIX_COLOR_TO_DEPTH * COLOR_PIXEL?
could you please let me know usage of those matrix?

dnjfk2003, 2020-07-11:
@RealSenseSupport thank you. I've checked those examples document.
As you said, now I know there can be a few pixels difference when using alignment but L515 is worse than D415(or 435).
Can L515 be improved later? Thanks.

cdb0y511, 2020-07-11:
Hi, @dnjfk2003 . I am not an expert on this specified  image processing domain. So maybe my answer  will not be corrected.
In my opinion, the align is like Image Panorama Stitching (normally no one align two identical picture together ).
OpenCV already has a lot of  Image Panorama Stitching algorithms.
like, https://www.pyimagesearch.com/2018/12/17/image-stitching-with-opencv-and-python/.
https://docs.opencv.org/master/d9/dab/tutorial_homography.html 
If you  have homography matrix , you can applied the alignment easily. Most  algorithms focus on calculate the accurate homography matrix. Since you can get the extrinsic and intrinsic from sensor, this will not be a problem.
But I still have no ideal how to deal with the different FOV. Maybe Intel have some tricks dont let us know.

dnjfk2003, 2020-07-12:
hi @cdb0y511. I am learning image processing domain as well. Thank you for the good information of homography matrix. I've read some material of that and it is very difficult to understand well with just one reading. But I feel I could get some inspiration from that. If I know new things. I'll share to you.
You said, you'd like to know how to handle the different FOV... but I'm not sure we have to consider that?
(Because I just think that we can make some equations for alignment with matrix like
POINT_depth = MATRIX * POINT_color if we know their resolution... I just suppose)

RealSenseSupport, 2020-09-11:
Hi @dnjfk2003 

Is there anything else needed on this one?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-18 by RealSenseSupport

wangsff, 2020-10-12:
hi @dnjfk2003 , I have met the same align problem between color and depth map. And I write a issue about it. #7541 
Do you any idea about it? May I discuss to you about this question?
Thanks!

5trobl, 2021-07-30:
> If you have homography matrix , you can apply the alignment easily.

For a general 3D scene you cannot apply homography if you want to be any accurate. Homography is about planar relationships within a common 3D perspective projection relationship. Here the scene is neither planar nor the punctual origin of the projection. So you must go 3D either using the `rs.align()` function (if it's accurate and the camera/lidar calibrations are as well) or doing the 3D reprojections yourself.