realsenseai/librealsense #6605: Firmware update failed L515

svebert, 2020-06-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (???) | 
| Operating System & Version |   {Win 10   | 
| Platform                                 | PC |
| SDK Version                            | n/a                         | 

### Issue Description
After recommended firmware update with realsense viewer 2.35.2 the newly shipped L515 does not reconnect. The camera is "dead" now. 
What can I do?

With the D400 DFU tool, i cannot enumerate or recover the camera.

ev-mp, 2020-06-16:
@svebert hello, I understand that you've used Viewer tool v2.35.2 to upgrade the FW.
Please check and share the Device Manager with view of camera and USB devices branches.

W.r.t to D400 DFU tool - are you using the tool [from the same release](https://github.com/IntelRealSense/librealsense/releases/download/v2.35.2/Intel.RealSense.FW.Update.exe) ?

svebert, 2020-06-16:
[image]

[image]

No, I did use the DFU tool from an older release. Where can I download the DFU tool from the newest release? Do I need to download the SDK?

ev-mp, 2020-06-16:
@svebert hello, the camera is actually recognized, can you also share the Dev. Mgr view of USB devices - should include L515 entry as well.
As for the error - seem a connection/initialization-related. Did you try to switch USB3 port and/or tighten the cable ?

svebert, 2020-06-16:
[image]

[image]

Hi!
Thank you for the fast response.
I used another USB3-Port and another USB3 cable. --> Same problem

ev-mp, 2020-06-16:
@svebert , can you share the output of `rs-enumerate-devices` ? 
additionally - if you build from source you can enable and post the Librealsense log for review

RealSenseSupport, 2020-06-16:
Just to round out the DFU or FW update part of the discussion, as part of the new releases of librealsense which includes the 2.35.2 version there is a FW update executable that should be used to update the FW:  https://github.com/IntelRealSense/librealsense/releases/tag/v2.35.2; https://github.com/IntelRealSense/librealsense/releases/download/v2.35.2/Intel.RealSense.FW.Update.exe

svebert, 2020-06-17:
for unknown reason I turned on the pc this morning and reconnected the L515. The only thing I have done, was running the Intel.RealSense.FW.Update.exe with -l (listing connected devices). 
L515 did show up immediatly. Intel RealSense Viewer works fine, now with L515.
I am very sorry for reporting this issue... No idea, why the camera works, now.

closed 2020-06-17 by svebert