realsenseai/librealsense #6607: turn off emitter

xiebei, 2020-06-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (05.12.05.00) | 
| Operating System & Version |   {Win10}  |                            | 
| SDK Version                            |  {2.35.2 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Body Scan }                          | 

Is there a faster way to turn off the emitter?

I use six D415 cameras to get accurate 3D models of human body. I have the method of splicing models, but there is infrared interference between cameras.

So I want to turn on / off the infrared emitter one by one to avoid infrared interference. I tested that it takes hundreds of milliseconds to turn off the infrared transmitter of a camera. It's too slow for me. The human body may shake. Is there a fast way?

sensor.set_option(RS2_OPTION_EMITTER_ENABLED, enable ? 1 : 0);

MartyG-RealSense, 2020-06-16:
Multiple RealSense 400 Series cameras should not interfere with each other (SR30x cameras do experience interference with each other though).  Could you give details please of why you think your problem with multiple D415 is due to infrared interference?  Thanks!

xiebei, 2020-06-16:
Thanks for your response.

When I turn on only one camera, like this
[image]

When I turn on the opposite camera, like this

[image]

ev-mp, 2020-06-16:
@xiebei hello,
> I tested that it takes hundreds of milliseconds to turn off the infrared transmitter of a camera

The time interval of data propagation for standard controls is within 1-3 frames, which is generally lees than 100 msec. Can you elaborate what method did you use to establish this figure?

xiebei, 2020-06-16:
@ev-mp Yes, about 100 ms at 30 fps,  about 350ms at 6fps
My laptop can only run to 6fps

100ms is also a little time-consuming for me，
6 camera turn on emitter - > capture - > turn off emitter one by one

MartyG-RealSense, 2020-06-16:
@xiebei   Regarding your kindly provided images, it does look as though there is a light source on the other side of the mannequin that coincides with where the image is breaking up at the armpit.

[image]

It makes me wonder if it is not so much a case of interference, but rather that the light source projecting through the armpit is blinding the affected camera to the armpit area and making it hard to read depth information from.  I note the green glow floating under the armpit that coincides with the position of the light-source on the other side of the mannequin.

xiebei, 2020-06-17:
@MartyG-RealSense OK, I'll try
By the way，I also bought the l515 camera，should arrive early next month，will the L515 be any different in that respect，will l515 should interfere with each other？

MartyG-RealSense, 2020-06-17:
The link below, under the section headed 'Body Scanning', explains how Intel demonstrated multicam full-surround body scanning with nine L515.  They avoided interference by hardware syncing the cameras.

https://www.intelrealsense.com/lidar-camera-l515-at-ces-2020/

xiebei, 2020-06-17:
OK，Thank you very much.

closed 2020-06-17 by xiebei