realsenseai/librealsense #6608: Eporting to PLY in Android

wkyoon, 2020-06-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415| 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Android V9 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Android  |
| SDK Version                            |  SDK 2.0                         | 
| Language                            |  JAVA                         | 
| Segment			| Smartphone                          | 

### Issue Description
Android example does not exist.
Has anyone export ply data from android?

There is a points class, but there is no way to export ply data.

MartyG-RealSense, 2020-06-16:
I researched the subject thoroughly but could not find any references for how to do so.  the most recent information I could find about the question of exporting a ply from Android is in the link below.

https://github.com/IntelRealSense/librealsense/issues/4874

wkyoon, 2020-07-05:
i found a way how to export ply data from android.

I made it by getting a hint from the 3D view in the java example.

MartyG-RealSense, 2020-07-05:
Thanks so much for the update!   Is it okay to close this case now if you are satisfied with the solution that you found?

closed 2020-07-18 by MartyG-RealSense