realsenseai/librealsense #6610: Announcement: Introducing the RealSense D455 Depth Camera

MartyG-RealSense, 2020-06-16:
Hi everyone,

The official RealSense store has now opened pre-orders for a new member of the RealSense 400 Series depth camera family: the **RealSense D455**, which increases range and accuracy over the D435 by a factor of two.

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d455.html

The official Intel Newsroom press release about the D455 can be found here:

https://newsroom.intel.com/news/intel-realsense-depth-camera-d455-twice-range-better-performance/

devshank3, 2020-06-16:
Wow quick  surprise !!
Any sample data available ?
and when is the shipping date for preorders ?

sam598, 2020-06-16:
Awesome! So many well thought out improvements, I can wait for the preorder to arrive and to work with these devices moving forward.

The language in the white paper seems to say this, but just to confirm explicitly; can the Global Shutter RGB camera in the D455 also be controlled through the new genlock mode?

And is the D455 maximum framerate in genlock mode also limited to half the target frame rate (e.g. 30fps is limited to 15fps and 90fps is limited to 45fps)?

MartyG-RealSense, 2020-06-16:
Hi @devshank3   Glad you're excited for the D455!

The official RealSense store's listing for the D455 states that pre-orders will be shipped in the week of July 20 2020.

There is not a D455 sample file available yet, though when one is available it should be posted on the sample data page.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

Regarding an IMU-less model: Intel test a wide range of camera configurations, and do not discuss future possible products.

MartyG-RealSense, 2020-06-16:
hi @sam598   The opening paragraph of Intel's white paper document on external synchronization states that the feature is "for all Intel RealSense depth cameras that are based on Global Shutter imagers".  This would implicitly include the D455.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras

@dorodnic  Could you provide advice please about the question below:

**Is the D455 maximum framerate in genlock mode also limited to half the target frame rate (e.g. 30fps is limited to 15fps and 90fps is limited to 45fps)?**

dorodnic, 2020-06-16:
I'm not sure, but @agrunnet is likely to know

agrunnet, 2020-06-16:
Correct. Same feature. Same limitation.

dheera, 2020-06-16:
Oh finally the depth and RGB have the same field of view! This was the biggest, almost mind-blowlingly idiotic design flaw in the D435 but it is finally fixed!

MartyG-RealSense, 2020-06-23:
The official CAD model of the D455 has been added to the RealSense CAD file archive.

https://dev.intelrealsense.com/docs/cad-files

closed 2020-07-01 by RealSenseCustomerSupport

VisionaryMind, 2020-08-28:
We just received this camera yesterday after a long pre-order wait and was expecting everything Intel's promotional material conveys and more. Unfortunately, it has been a huge disappointment. Using the Realsense Viewer program, this unit only achieves a fluid frame rate of 30fps. Taking it up to 90fps or higher will create video freezes and eventually crash a computer (and we have tested it on several, including a 32-core CPU, 64Gb RAM, GeForce NVIDIA RTX 2070 video card, and high-speed USB 3 ... so it's not the computer).

We were hoping to use it for robotics and 3D scanning applications but once it is connected to such a device (we used a Samsung A30S since it is portable), the frame rate drops down to 15fps and resolution to 640x480 -- and even at that stepped down quality, it cannot track depth and RGB images in a synchronized manner. That's the least of the problems, unfortunately --- as it also cannot properly relay depth for nearly 80% of the surroundings. There are the expected black shadows behind objects not 5 meters away due to occlusion but there are also huge black spots ** on ** objects or, in some cases, entirely covering them. And yes -- we have have performed on-chip calibration. Here is a sample extraction with the camera on a tripod (in a well-lit room during the day) [https://imgur.com/9oA3YQQ](url)

So it marginally works tethered to a powerful computer, but using it as Intel markets it? I am highly doubtful. The Realsense SDK 2.0 has eliminated many useful features the original SDK contained and it does not yet fully support D455 --- clearly, given these preliminary tests. I would never, in a million years, attach this to a robot that requires proper obstacle avoidance or even space sensing. It could be dangerous! With this level of performance and quality, how can it be used for anything whatsoever? Is the purpose of the camera merely for hobbyist tinkering?

We will run some additional tests -- but with the amount of stereo IR warping, point cloud distortion, and inability to synchronize high-resolution RGB, we will likely be shipping this straight back. I have reached out to support, and they have gone silent. I presume they are aware of these problems. Hopefully, this comment will remain here so as to provide a more balanced view. Intel is welcome to refute anything I have said here, within reason.

RealSenseSupport, 2020-08-28:
Thank you for the feedback.  We would like to learn more about your configuration and experience.  We will be in contact with you.

Thank you from the RealSense™ Customer Support Team.

michaelvorobiev, 2021-01-31:
D455 is the first camera that consists all global shutter cameras (including RGB sensor). Does GenLock option work for all cameras (including RGB) or only for Depth Module sensors?

MartyG-RealSense, 2021-01-31:
Hi @michaelvorobiev   The discussion in the link below addresses the subject of color sync with genlock on D455.

https://github.com/IntelRealSense/librealsense/issues/7502

As far as I am aware, it is a feature that is still being worked on at the time of writing this.

https://github.com/IntelRealSense/librealsense/issues/7502#issuecomment-724784638

matanster, 2021-02-03:
@VisionaryMind I wonder whether you still find the D455 less robust than previous models, or have these aspects ironed out by now. A supposed inability to track color and depth in a synchronized manner might sound prohibitive to any application to me, as I would kind of assume that to always work in order to enable any depth enabled application.

VisionaryMind, 2021-02-03:
@matanster One of my colleagues in Sweden had a conversation with the Realsense director of engineering along with at last one other Intel technical lead. They point blank told her that these cameras are not suited to photo-realistic re-construction of 3D environments, which was our original goal. The language used, if I recall was "such a goal would be possible with a great investment of resources". This sentiment is echoed throughout the forums globally, and I frequently encounter agreement that the data coming out of the Realsense cameras, at last for now, is too noisy.

That being said, I have seen very good SLAM results using a D455 + T265 paired, and if accuracy of pointcloud / 3D mesh is not a concern, this is one of the areas where Realsense actually shines. They may also be used, to a degree, for body tracking but again, the noise (which is not manageable with configuration adjustment) confounds results. Even the Azure Kinect, which has shown 10x less noise and cleaner depth boundaries than its Realsense contemporaries, nonetheless suffers from extreme latency and data overload --- similar to all things Microsoft in general. Overly bloated and inefficient data payloads.

Nonetheless, we have found K4A (Azure Kinect) to yield better 3D pointclouds and textured meshes with far less noise but --- nonetheless --- it still falls quite short of the higher-end LiDARs and / or photogrammetry. All depth cameras do. So it all depends on your context. Realsense depth cameras are fast but noisy and inaccurate, well-suited to SLAM but not viable in contexts where either photo-realism or cm-accuracy is a hard mandate. I would also venture to say clean, consistent-quality volumetric capture would also be problematic due to this shortcoming.

matanster, 2021-02-04:
Thanks for the wide exposition @VisionaryMind, really helpful. I wonder what noise to expect with body tracking, as my main scenario is fine hand gesture recognition. The colorized depth official photo images looked fairly clean to me IIRC . Is it anything that a convolution/interpolation with the neighboring pixels won't solve?