realsenseai/librealsense #6611: Apt server is broken, gives 403 Forbidden

TheButlah, 2020-06-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Platform                                 | PC  |
| SDK Version                           | 2.35.0 | 
| Language                            | C++ | 
| Segment			| Robot | 

### Issue Description
I cannot properly `sudo apt install librealsense2` anymore. The apt server appears broken and returns an HTTP 403 Error code.

Script:
```bash
# Add Intel server to list of repos and register its public key
sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE \
    || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u

# Use the newly added intel server to install librealsense2-dev
RS_VERSION=2.35.0-0~realsense0.2750
sudo apt-get update && sudo apt-get install -y librealsense2=$RS_VERSION librealsense2-dev=$RS_VERSION
```
Output:
```
Warning: apt-key output should not be parsed (stdout is not a terminal)
Executing: /tmp/apt-key-gpghome.gWwk6gIGtn/gpg.1.sh --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: keyserver receive failed: Cannot assign requested address
Warning: apt-key output should not be parsed (stdout is not a terminal)
Executing: /tmp/apt-key-gpghome.3DEa2LPXG1/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: public key ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" imported
gpg: Total number processed: 1
gpg:               imported: 1
Get:2 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]
Hit:3 http://archive.ubuntu.com/ubuntu bionic InRelease
Get:4 http://archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]
Err:5 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease
  403  Forbidden [IP: 52.218.110.98 80]
Get:6 http://security.ubuntu.com/ubuntu bionic-security/main amd64 Packages [960 kB]
Get:7 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]
Get:8 http://archive.ubuntu.com/ubuntu bionic-updates/universe amd64 Packages [1,392 kB]
Get:1 https://apt.llvm.org/bionic llvm-toolchain-bionic InRelease [4,232 B]
Get:9 https://apt.llvm.org/bionic llvm-toolchain-bionic/main amd64 Packages [10.6 kB]
Get:10 http://security.ubuntu.com/ubuntu bionic-security/universe amd64 Packages [861 kB]
Get:11 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 Packages [9,092 B]
Get:12 http://security.ubuntu.com/ubuntu bionic-security/restricted amd64 Packages [66.4 kB]
Get:13 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages [1,253 kB]
Get:14 http://archive.ubuntu.com/ubuntu bionic-updates/restricted amd64 Packages [80.7 kB]
Get:15 http://archive.ubuntu.com/ubuntu bionic-updates/multiverse amd64 Packages [20.3 kB]
Reading package lists...
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/bionic/InRelease  403  Forbidden [IP: 52.218.110.98 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease' is not signed.
```

TheButlah, 2020-06-16:
For context, this has broken our whole build system. Done of our docker images will build now

ev-mp, 2020-06-16:
@TheButlah , thank you for the report.
We're looking into this issue and will update accordingly.

ev-mp, 2020-06-16:
@TheButlah , the server's accessibility is restored now.
Sorry for the inconvenience.

closed 2020-06-16 by ev-mp

adamkrawczyk, 2020-09-16:
I think the same problem occurs now.

arunlee77, 2020-09-16:
Same issue seen now.

beduffy, 2021-07-13:
Same problem here.

haraldsteinlechner, 2021-07-17:
i just ran into the same problem - i changed the source as shown here - https://github.com/IntelRealSense/librealsense/issues/9105#issuecomment-852076013

JaeYoungLee79, 2022-02-21:
So is ti fixed?