realsenseai/librealsense #6612: Matlab wrapper: depth_frame() class issue.

suki71, 2020-06-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   |5.12.05.00 | 
| Operating System & Version |  Windows 10  | 
| Platform                                 | PC |
| SDK Version                            |  v2.35.2  | 
| Language                            | Matlab | 

### Issue Description
Hi, I am currently using the using the Matlab wrapper for the Intel Realsense D435i camera on Windows 10 and am trying to get distance measurements from the camera to a specific point. To do this, I know I have to call the get_distance(this, x, y) function in the depth_frame class. To create a depth_frame object, I am putting in the following line of code:
depth_frame = realsense.depth_frame();

When I run this, I get the following error:
[image]

I read in another issue post that Matlab should be creating the handle parameter by itself, but I've tried on multiple computers and am getting the same result. In this context, what exactly is "handle" and how can I pass this parameter into my code to be able to successfully run it?

I would like to apologize in advance if this has already been asked, I went through the forum already and was not able to find a solution to this problem. Thank you in advance.

MartyG-RealSense, 2020-06-17:
Hi @suki71   There was a case of a RealSense D435i user who was using the RealSense MATLAB wrapper for Windows 10 and its example script called **depth_example.m**.  Their aim was to use it as a basis for reading the depth distance from the camera.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/depth_example.m

They too were trying to use get_distance to calculate the distance.  Their version of the statement that they tried was:

**depth.get_distance(x,y) %x,y**

This represented the coordinates of the point that they wanted to get the distance for.

They were advised in the final comment of the discussion that it should be possible to calculate the distance using MATLAB's own stereo vision routines, with links provided to those routines.

https://uk.mathworks.com/matlabcentral/answers/466039-distance-of-a-line-by-camera-points

suki71, 2020-06-17:
Hi @MartyG-RealSense, thank you for providing me with this resource. I will try to apply this line of code to my program and will update with it working or not.

MartyG-RealSense, 2020-07-05:
Hi @suki71   Do you still require assistance with this case, please?   Thanks!

suki71, 2020-07-05:
My issue has been resolved. Thank you!

closed 2020-07-05 by suki71